MicrochipTech/cryptoauthlib #117: PKCS11 Yocto Setup

agilesai1294, 2019-11-01:
Hi,

I am using Yocto2.1 to build my image onto the ATSAMA5D2 board. I am using Yocto2.1 Krogoth. I am having issues initializing ATECC608A and generating CSR using the private key. Please help me resolve this issue.

root@:~# p11tool --version
p11tool 3.4.9
Copyright (C) 2000-2016 Free Software Foundation, and others, all rights reserved.
This is free software. It is licensed for use, modification and
redistribution under the terms of the GNU General Public License,
version 3 or later <http://gnu.org/licenses/gpl.html>

root@:~# cat /etc/cryptoauthlib/cryptoauthlib.conf
# Cryptoauthlib Configuration File
 
filestore = /var/lib/cryptoauthlib

root@:~# cat /var/lib/cryptoauthlib/0.conf
# Reserved Configuration for a device
# The objects in this file will be created and marked as undeletable
# These are processed in order. Configuration parameters must be comma
# delimited and may not contain spaces
 
interface = i2c,0xC0
freeslots = 1,2,3
 
# Slot 0 is the primary private key
object = private,device,0
 
# Slot 10 is the certificate data for the device's public key
#object = certificate,device,10
 
# Slot 12 is the intermedate/signer certificate data
#object = certificate,signer,12
 
# Slot 15 is a public key
object = public,root,15

root@:~# cat /usr/share/p11-kit/modules/cryptoauthlib.module
module: /usr/lib/libcryptoauth.so
critical: yes
trust-policy: yes
managed: yes
log-calls: no

root@:~# p11tool --list-all
Object 0:
        URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=946EEF92A0E7;token=0123EE;object=device;type=private
        Type: Private key
        Label: device
        Flags: CKA_PRIVATE; CKA_SENSITIVE;
        ID:
 
Object 1:
        URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=946EEF92A0E7;token=0123EE;object=device;type=public
        Type: Public key
        Label: device
        ID:
 
Object 2:
        URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=946EEF92A0E7;token=0123EE;object=root;type=public
        Type: Public key
        Label: root
        ID:

root@:~# p11tool --initialize "pkcs11:serial=946EEF92A0E7" --label test
Enter Security Officer's PIN:
Enter new User's PIN:
Token '0123EE' with URL 'pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=946EEF92A0E7;token=0123EE' requires security officer PIN
Enter PIN:
No PIN given.

root@:~# openssl req -engine pkcs11 -key "pkcs11:token=0123EE;object=device;type=private" -keyform engine -new -out new_device.csr -subj "/CN=NEW CSR EXAMPLE"
engine "pkcs11" set.
Found slot without user PIN
PKCS11_get_private_key returned NULL
cannot load Private Key from engine
3070076112:error:26096080:engine routines:ENGINE_load_private_key:failed loading private key:eng_pkey.c:124:
unable to load Private Key

root@:~# ls -l /usr/lib/ | grep libp11
lrwxrwxrwx  1 root root      19 Oct 21  2019 libp11-kit.so.0 -> libp11-kit.so.0.3.0
-rwxr-xr-x  1 root root  996284 Oct 21  2019 libp11-kit.so.0.3.0
lrwxrwxrwx  1 root root      15 Oct  6 11:35 libp11.so.2 -> libp11.so.2.5.2
-rwxr-xr-x  1 root root   44268 Oct 31  2019 libp11.so.2.5.2
lrwxrwxrwx  1 root root      19 Oct 21  2019 p11-kit-proxy.so -> libp11-kit.so.0.3.0

Thanks

bryan-hunt, 2019-11-01:
[libp11 versions less than 0.4.10](https://github.com/OpenSC/libp11/releases/tag/libp11-0.4.10) do not have support EC keys with OpenSSL 1.1.x

You'll have to change your yocto configuration to require libp11 v0.4.10

agilesai1294, 2019-11-01:
@bryan-hunt I used https://layers.openembedded.org/layerindex/recipe/32120/ to build my libp11. But it is mentioned that yocto krogoth supports 0.3.0. Does that mean we can't use libp11 with yocto 2.1?

bryan-hunt, 2019-11-01:
The Yocto build process is outside my area of expertise. Looking at the above it is clear that it included 0.3.0 rather than the 0.4.10 even if you had replaced the recipe. I'm not sure what kind of API changes there might between 0.3.0 and 0.4.10 to say if 0.4.10 is truly not compatible with the other parts of Yocto 2.1

agilesai1294, 2019-11-04:
@bryan-hunt openssl version is 1.0.2h and found the following text as a part of configure.ac file from lbp11 https://github.com/OpenSC/libp11/blob/master/configure.ac

case "`$PKG_CONFIG --modversion --silence-errors libcrypto || \
        $PKG_CONFIG --modversion openssl`" in
        1.1.*) # Predicted engines directory prefix for OpenSSL 1.1.x
            LIBP11_LT_OLDEST="3"
            debian_ssl_prefix="openssl-1.1.0";;
      

>  1.0.*) # Engines directory prefix for OpenSSL 1.0.x
>             LIBP11_LT_OLDEST="2"
>             debian_ssl_prefix="openssl-1.0.0";;**

        *) # Engines directory prefix for OpenSSL 0.9.x
            LIBP11_LT_OLDEST="2"
            debian_ssl_prefix="ssl";;
esac

Is it why my p11 version is changing to libp11.so.2.5.2

agilesai1294, 2019-11-06:
@bryan-hunt Is there any minimum version of openssl required? I am using openssl 1.0.2h

agilesai1294, 2019-11-07:
@bryan-hunt 
Can you please look into the log that is generated while requesting CSR?

root@linux:/usr/lib/engines# openssl req -engine pkcs11 -key "pkcs11:token=0123EE;object=device;type=private" -keyform engine -new -out new_device.csr -subj "/CN=NEW CSR EXAMPLE"
C_Initialize
  IN: pInitArgs = NULL
C_Initialize = CKR_OK
C_GetSlotList
  IN: tokenPresent = CK_FALSE
  IN: pulCount = 0xBEB9BFD4 = 3065496424
 OUT: pSlotList = (1) NO-VALUES
C_GetSlotList = CKR_OK
C_GetSlotList
  IN: tokenPresent = CK_FALSE
  IN: pulCount = 0xBEB9BFD4 = 1
 OUT: pSlotList = (1) [ SL0 ]
C_GetSlotList = CKR_OK
engine "pkcs11" set.
C_GetSlotList
  IN: tokenPresent = CK_FALSE
  IN: pulCount = 0xBEB9BE0C = 3065490348
 OUT: pSlotList = (1) NO-VALUES
C_GetSlotList = CKR_OK
C_GetSlotList
  IN: tokenPresent = CK_FALSE
  IN: pulCount = 0xBEB9BE0C = 1
 OUT: pSlotList = (1) [ SL0 ]
C_GetSlotList = CKR_OK
C_GetSlotList
  IN: tokenPresent = CK_FALSE
  IN: pulCount = 0xBEB9BE0C = 0
 OUT: pSlotList = (1) NO-VALUES
C_GetSlotList = CKR_OK
C_GetSlotList
  IN: tokenPresent = CK_FALSE
  IN: pulCount = 0xBEB9BE0C = 1
 OUT: pSlotList = (1) [ SL0 ]
C_GetSlotList = CKR_OK
C_GetSlotInfo
  IN: slotID = SL0
 OUT: pInfo = {
        slotDescription: "0_3_0"
        manufacturerID: "Microchip Technology Inc"
        flags: 5 = CKF_TOKEN_PRESENT | CKF_HW_SLOT
        hardwareVersion: 0.2
        firmwareVersion: 255.255
      }
C_GetSlotInfo = CKR_OK
C_GetTokenInfo
  IN: slotID = SL0
 OUT: pInfo = {
        label: "0123EE"
        manufacturerID: "Microchip Technology Inc"
        model: "ATECC608A"
        serialNumber: "946EEF92A0E7"
        flags: 1025 = CKF_RNG | CKF_TOKEN_INITIALIZED
        ulMaxSessionCount: 1
        ulSessionCount: 0
        ulMaxRwSessionCount: 1
        ulRwSessionCount: 0
        ulMaxPinLen: 0
        ulMinPinLen: 0
        ulTotalPublicMemory: 1
        ulFreePublicMemory: 1
        ulTotalPrivateMemory: 1
        ulFreePrivateMemory: 1
        ulFreePrivateMemory: 1
        hardwareVersion: 0.2
        firmwareVersion: 255.255
        utcTime:
      }
C_GetTokenInfo = CKR_OK
Found slot without user PIN
PKCS11_get_private_key returned NULL
cannot load Private Key from engine
3069396176:error:26096080:engine routines:ENGINE_load_private_key:failed loading private key:eng_pkey.c:124:
unable to load Private Key
C_Finalize
  IN: pReserved = NULL
C_Finalize = CKR_OK

bryan-hunt, 2019-11-07:
It's clearly a problem with the version of OpenSSL and libp11. It will take a decent amount of research to figure out what is wrong with that combination. If you can use a moder release of yocto (like warrior) that would be better. Otherwise I can't give a timeline as to when I'll be able to dig into OpenSSL and libp11 sources. You can try to ask the libp11 project why EC keys with 1.0.2h are not working.

agilesai1294, 2019-11-26:
@bryan-hunt was able to solve this issue with Yocto 2.1. Thanks for your support.

closed 2019-11-26 by agilesai1294