MicrochipTech/cryptoauthlib #118: PKCS11 Initialization Error

agilesai1294, 2019-11-05:
Hi,

root@sama5d2-xplained-sd:~# i2cdetect -y 0
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: 50 -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: 60 -- -- -- 64 -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

root@sama5d2-xplained-sd:~# cat /etc/cryptoauthlib/cryptoauthlib.conf
# Cryptoauthlib Configuration File

filestore = /var/lib/cryptoauthlib

root@sama5d2-xplained-sd:~# cat /var/lib/cryptoauthlib/0.conf
0.conf   0.conf~
root@sama5d2-xplained-sd:~# cat /var/lib/cryptoauthlib/0.conf
0.conf   0.conf~
root@sama5d2-xplained-sd:~# cat /var/lib/cryptoauthlib/0.conf
# Reserved Configuration for a device
# The objects in this file will be created and marked as undeletable
# These are processed in order. Configuration parameters must be comma
# delimited and may not contain spaces

interface = i2c,0xC0
freeslots = 1,2,3

# Slot 0 is the primary private key
object = private,device,0

# Slot 10 is the certificate data for the device's public key
#object = certificate,device,10

# Slot 12 is the intermedate/signer certificate data
#object = certificate,signer,12

# Slot 15 is a public key
object = public,root,15

root@sama5d2-xplained-sd:~# cat /etc/pkcs11/pkcs11.conf
# This setting controls whether to load user configuration from the
# ~/.config/pkcs11 directory. Possible values:
#    none: No user configuration
#    merge: Merge the user config over the system configuration (default)
#    only: Only user configuration, ignore system configuration
user-config: merge

root@sama5d2-xplained-sd:~# cat /usr/share/p11-kit/modules/cryptoauthlib.module
module: /usr/lib/libcryptoauth.so
critical: yes
trust-policy: yes
managed: yes
log-calls: no

root@sama5d2-xplained-sd:~# p11tool --list-all
pkcs11_init: PKCS #11 initialization error.
warning: no token URL was provided for this operation; the available tokens are:

I dont see anything on p11tool --list-all

bryan-hunt, 2019-11-05:
Do you have [p11-kit](https://github.com/MicrochipTech/cryptoauthlib/wiki/PKCS11-Linux-Setup#using-p11-kit-proxy) set up?

bryan-hunt, 2019-11-05:
In your previous ticket #117 it was working. What did you change?

closed 2019-11-05 by agilesai1294