MicrochipTech/cryptoauthlib #119: Can't initialize Slot 0 private key

howey, 2019-11-12:
Hi,

I can't initialize the private key at slot 0. No public key is created in the device after the --initialize step:

`# p11tool --provider=/usr/lib/libcryptoauth.so --list-all
Object 0:
        URL: pkcs11:model=ATECC508A;manufacturer=Microchip%20Technology%20Inc;serial=4FFE4B2C3251;token=0123EE;object=device;type=private
        Type: Private key
        Label: device
        Flags: CKA_PRIVATE; CKA_SENSITIVE; 
        ID: 

# p11tool --provider=/usr/lib/libcryptoauth.so --initialize "pkcs11:token=0123EE
;type=private" --label test
Enter Security Officer's PIN: 
Initializing token... 
Error in pkcs11_init:1182: PKCS #11 error.
# p11tool --provider=/usr/lib/libcryptoauth.so --list-all
Object 0:
        URL: pkcs11:model=ATECC508A;manufacturer=Microchip%20Technology%20Inc;serial=4FFE4B2C3251;token=0123EE;object=device;type=private
        Type: Private key
        Label: device
        Flags: CKA_PRIVATE; CKA_SENSITIVE; 
        ID: 
`
If I create a new key pair with the private key in slot 1, it works. Only slot 0 doesn't work.

Versions:
libcryptoauth: head of pkcs11 a0007d2f..
p11tool: 3.5.19
linux: 4.9.40

Please let me know any other information you need from me. Thanks.

bryan-hunt, 2019-11-12:
Initialize does not create a key pair - it initializes the token. This is actually a false report of a failure - it in fact completed successfully.

It is a duplicate report of #95 however.

closed 2019-11-12 by bryan-hunt

howey, 2019-11-12:
Bryan,

Thanks for the very fast response.

I thought a public key should show up with --list-all after a successful --initialize. If the token is initialized then shouldn't --export-pubkey work?

p11tool --provider=/usr/lib/libcryptoauth.so --export-pubkey "pkcs11:token=012
3EE;object=device;type=private"
warning: --login was not specified and it may be required for this operation.
warning: no --outfile was specified and the public key will be printed on screen.
Error in pkcs11_export_pubkey:1131: The requested data were not available.

bryan-hunt, 2019-11-12:
After resetting the device you should see something yes.

howey, 2019-11-12:
After power cycling the board I still don't see a public key with --list-all. And --export-pubkey doesn't work either.

What I'm really trying to do is generate a CSR with openssl. But my understanding is that --export-pubkey has to work before openssl can use it.

bryan-hunt, 2019-11-12:
What is the history of the device and the configuration?

If possible can you run the info.py script from https://github.com/MicrochipTech/cryptoauthtools/tree/master/python

howey, 2019-11-13:
It's a new chip, the only thing I've done is lock the config zone.

The board doesn't have Python installed but I called the same methods in the info.py script from a C program, this should give you the same information:

Device Part: 00005000
Serial number: 01234FFE4B2C3251EE
Configuration zone: 01234FFE000050004B2C3251EEC04100C0005500832087208F20C48F8F8F8F8F9F8FAF8F0000000000000000000000000000AF8FFFFFFFFF00000000FFFFFFFF00000000FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF00005500FFFF0000000000003300330033001C001C001C001C001C003C003C003C003C003C003C003C001C00
Config Zone is locked
Data Zone is unlocked
Public key: 00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000

howey, 2019-11-13:
Calling atcab_genkey(0) gets me working. After that call I can see a public key in p11tool --list-all, and p11tool --export-pubkey works too.

So the problem is that I had to call atcab_genkey, calling p11tool --initialize does not work. It's a workaround but I would prefer to use the p11tool command in my workflow.

bryan-hunt, 2019-11-13:
The problem the workflow was broken by locking the config zone manually. The initialize command is explicitly for one thing - write a configuration, generate keys, and lock the config and data zones. If you wish to manually set the configuration and provision the device that is fine - it just needs to be done completely and then you can use pkcs11 for the rest of the tasks (and of course inform the pkcs11 library of your configuration in the configuration files). 

Also the device is not in an operational state without the datazone being locked (hence the empty public key). The config zone lock makes the configuration immutable. The datazone lock (a bit of a misnomer) activates the configuration and enforces it's rules. When the datazone is unlocked you can freely write data into the slots - you may not read any data, nor use any keys.

The configuration cited above is not a particularly useful one (for example slots 5,6,7 & 15 are completely unusable). Slot 4 needs to be provisioned before the datazone lock and then it can never be changed again - it's only purpose is to be able to read slot 3. The only way to write into slot 3 is to perform an ECDH with slot 2.

howey, 2019-11-13:
I remember with a previous version (21f9f2, maybe) I had to lock the config zone manually in order for the --initialize to work. But maybe this is no longer the case.

I'll get another new chip, try --initialize before locking manually, and let you know if I still have an issue. Thanks.