MicrochipTech/cryptoauthlib #120: Encrypt/Decrypt File

agilesai1294, 2019-11-25:
Hi,

Is it possible for me to use cryptoauthlib to encrypt or decrypt the file? Lets say i have hello.txt and i want to encrypt/decrypt the file using the key stored in slot 3.

bryan-hunt, 2019-11-25:
Yes, but the question is pretty generic - how do you want to accomplish it?

* The ATECC608A has an AES feature which can encrypt and decrypt data sent to the device over I2C using a key stored in a slot. However the data is transmitted over I2C in the clear. It is also very slow.

* You could read a key out and do what you want with it using an encrypted read

* You could perform an HKDF on the key in the slot, the answer of which would be transferred back to the host using the IO protection secret.

agilesai1294, 2019-11-26:
@bryan-hunt The configuration for slot 3 is no read no write. So I have to send the file via i2c to encrypt it. My question is can I send the **whole** file over the i2c bus and encrypt it with the key that is there in slot 3 or do I need to send 16 bytes of contents each time over i2c to encrypt it?

CarlMitchellKT, 2019-11-26:
You'll probably want to use authenticated encryption. Cryptoauthlib makes that reasonably easy, if slow (due to I2C bus speed).

Encryption starts by calling `atcab_aes_gcm_init_rand()` and storing the returned IV, then you (optionally) send any data you want to be authenticated but not encrypted with `atcab_aes_gcm_aad_update()`, then send 16 bytes of plaintext at a time with `atcab_aes_gcm_encrypt_update()`, and finally `atcab_aes_gcm_encrypt_finish()` to finish and get the tag. 

Decryption uses the relevant decryption function: `atcab_aes_gcm_init()` passing in the IV output by init_rand, `atcab_aes_gcm_aad_update()` for the additional authenticated data, `atcab_aes_gcm_decrypt_update()` for the ciphertext, and `atcab_aes_gcm_decrypt_finish()` to finish and verify the tag.

Important: if the tag fails to verify, the plaintext MUST be discarded without any further processing. Failure to do so violates the security guarantees of AES-GCM. 

Important: IVs must never be re-used for another encryption under the same key. That should be impossible if using `atcab_aes_gcm_init_rand()` for encryption, but it's possible to use `atcab_aes_gcm_init()` there and thus break all security.

It's possible to use unauthenticated encryption (no tag) via CTR mode. The IV concern is still present. The disadvantage of unauthenticated encryption is that it's only secure against a passive attacker, if the attacker can modify the ciphertext there are a number of attacks possible. Since GCM is supported by this library it's the best option from a security perspective.

bryan-hunt, 2019-11-26:
I really must emphasize that using the AES feature of the 608 to encrypt and decrypt a file when you have a Linux based host processor is not recommended. All the data is sent over I2C in the clear. So two questions a) what is the desired purpose of your encryption (is the data sensitive?) and b) what are your performance requirements.

Using HKDF to produce your keying material for your encryption/decryption will yield far better performance and security.

agilesai1294, 2019-11-26:
@bryan-hunt I have several keys that I want to store for my use cases. I cant store all the keys in ATECC608A. My other option is to encrypt the keys with ATECC608A and store it in the local storage of the processor and whenever I need to use those keys I can decrypt it with the key and get the plain text. I am planning on using the IO protection key for encrypted i2c channel between the processor and secure element. 

Can you tell me how I can use cryptoauthlib to achieve that functionality?

bryan-hunt, 2019-11-26:
HKDF will allow you to perform the operation on a key (and other data) to produce key material for this use case. The result of the HKDF operation can be transferred using the IO protection secret. AES output is always in the clear. The IO protection secret can’t be used with the AES commands

agilesai1294, 2019-11-26:
@bryan-hunt Can you tell me the procedure on how I can use HKDF with cryptoauthlib and ATECC608A for implementing this functionality?

bryan-hunt, 2019-11-26:
> @bryan-hunt Can you tell me the procedure on how I can use HKDF with cryptoauthlib and ATECC608A for implementing this functionality?

To get application design support you need to request this from your FAE. You can also make a request through the [microchip support portal](https://www.microchip.com/mymicrochip/)

mbark-blm, 2019-12-11:
@bryan-hunt  @agilesai1294  can anyone help me to use this command please atcab_aes_encrypt(key_id, key_block, plaintext, ciphertext),I don't know what i have to put in key_id and key_block parameters!!?

closed 2020-06-15 by bryan-hunt