MicrochipTech/cryptoauthlib #121: Running atcab_random() example returns non random number

kapteynpeter, 2019-11-26:
I am using an SAME51J20A with Atmel Start and ATECC608 using the I2C bus. I'm running the following code:

`uint8_t random_number[32] = {0};`
  `atcab_init(&cfg_ateccx08a_i2c_default);`
  `atcab_random(&random_number); // get a random number from the chip`

The` atcab_init` function returns successful, but after the `atcab_random` call, `random_number` is set to 
`[0xff,0xff,0x00,0x00,0xff,0xff,0x00,0x00,... repeating]`

I probed the bus with a scope and the `[...0xff, 0x00...]` data is what's actually being returned from the chip. 

There is successful I2C communication going on between the MCU and the ATECC chip, so I assume it is some sort of setup issue. What am I missing here?

bryan-hunt, 2019-11-26:
The bytes you are observing are the pattern the device returns when you haven’t configured it. See section 3.3.2 of the [datasheet](http://ww1.microchip.com/downloads/en/DeviceDoc/20005927A.pdf).

You will have to configure (and lock) the config zone before the RNG will output random numbers rather than the diagnostic code.

(Duplicate #85 )

closed 2019-11-26 by bryan-hunt

sangamkrish, 2025-01-22:
Hello @pkapt 

could  you please provide the code for above that you have mentioned .
When I tried to do it 
 status = atcab_init(&cfg_atecc608);
    if (status != ATCA_SUCCESS) {
        printf("Device initialization failed\n");
        return 1;
    }

I am getting Device initialization  failed.

pkapt, 2025-01-24:
It’s been 5 years, I’m sure I don’t have the code anymore haha. Thanks for
responding.

sangamkrish, 2025-01-25:
Any solution for me so that I can proceed in this project.