MicrochipTech/cryptoauthlib #123: crypt and decrypt with ATECC608A

mbark-blm, 2019-12-11:
can anyone help me to use this command please atcab_aes_encrypt(key_id, key_block, plaintext, ciphertext),I don't know what i have to put in key_id and key_block parameters!!?

agilesai1294, 2019-12-13:
@mbark-blm I believe key_id is the key location and key_block is the index of the block to use for the actual key.
/** \brief Perform an AES-128 encrypt operation with a key in the device.
 *
 * \param[in]  key_id      Key location. Can either be a slot number or
 *                         ATCA_TEMPKEY_KEYID for TempKey.
 * \param[in]  key_block   Index of the 16-byte block to use within the key
 *                         location for the actual key.
 * \param[in]  plaintext   Input plaintext to be encrypted (16 bytes).
 * \param[out] ciphertext  Output ciphertext is returned here (16 bytes).
 *
 * \return ATCA_SUCCESS on success, otherwise an error code.

mbark-blm, 2019-12-15:
@agilesai1294 
Thank you for your reply,
but I have another problem concerning the function to use to store the key in a slot

closed 2022-11-23 by github-actions[bot]