MicrochipTech/cryptoauthlib #124: Encrypt using atcab_aes_encrypt

agilesai1294, 2019-12-13:
I wanted to do AES Encryption and Decryption using ATECC608A. I have a test AES key and is stored in slot 9.  Configuration of my slot 9 is pasted below.
[image]

I have also attached the snippet of code that i am doing to do encryption and decryption using the key stored in slot 9
    plain_data_9  = os.urandom(16)
    cipher_text_9 = bytearray(16)
    decrypted_text_9 = bytearray(16)
    print('\nEncrypting Data using Slot 9')
    assert atcab_aes_encrypt(9,0,plain_data_9,cipher_text_9)
    print('    Encrypted data:')
    print(pretty_print_hex(cipher_text_9, indent='        '))
    `print('\nDecrypting` Data using Slot 9')
    assert atcab_aes_decrypt(9,0,cipher_text_9,decrypted_text_9)
    print('    Decrypted data:')
    print(pretty_print_hex(decrypted_text_9, indent='        '))`

The output of the program:
Write command:
    Writing data to slot 9
    Write Success
    Generated data for slot 9:
        2A 3B 14 C8 D4 39 F7 B4 31 76 CA 9F 11 FC 67 67

Encrypting Data using Slot 9
    Encrypted data:
        00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Decrypting Data using Slot 9
    Decrypted data:
        00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

Done

Can you explain me why my Encrypted Data and Decrypted are showing 00?

Thanks

agilesai1294, 2019-12-13:
@bryan-hunt Can you explain why aes command fails?

mbark-blm, 2019-12-15:
firstly, it's useless in using assert without putting a condition:
assert atcab_aes_decrypt (9.0, cipher_text_9, decrypted_text_9) = 0x00
if it displays an exception, you must change the slot used 
you can try slot 5

agilesai1294, 2019-12-16:
The data zone is not locked. After locking Datazone aes_encrypt and aes_decrypt works.

closed 2019-12-16 by agilesai1294