MicrochipTech/cryptoauthlib #126: atcab_read_enc issue

agilesai1294, 2019-12-17:
Hi, I have a problem running the python example read_write.py code. I have a slot 5 which has configuration of encrypted read and encrypted write. When I try to do encrypted read it gives me an error of 0xF4 which means chip was in a state where it could not execute the command, response status byte indicates command execution error. Can you help me resolve this issue?

agilesai1294, 2019-12-20:
Changing the read key slot from 15 to 6 worked.

closed 2019-12-20 by agilesai1294

dogusural, 2020-09-01:
Hey agilesai1294 , 

Can you tell me where you found the GUI to configure the atecc device ? 

Best

bryan-hunt, 2020-09-01:
@dogusural For product support questions you should submit a product support request through the [Microchip Support portal](https://microchipsupport.force.com/s/). We are unable to provide the same level of product usage support via github as we do through the support portal.

For details on the Support Portal or how to use it please see [How to Submit a Case](https://microchipsupport.force.com/s/article/How-to-submit-a-case)