MicrochipTech/cryptoauthlib #128: python info.py wit hid not working

agilesai1294, 2020-01-09:
Hi,

I have tried out a sample code to test ATECC608A.
from cryptoauthlib import *
def info(iface='i2c', device='ecc', **kwargs):
    ATCA_SUCCESS = 0x00
    load_cryptoauthlib()
    cfg = cfg_ateccx08a_kithid_default()
    cfg.iface_type = ATCAIfaceType.ATCA_I2C_IFACE
    cfg.devtype=ATCADeviceType.ATECC608A
    cfg.cfg.atcahid.dev_interface=ATCAKitType.ATCA_KIT_I2C_IFACE
    cfg.cfg.atcahid.dev_identity=0xC0
    print('0x{:08X}'.format(cfg.cfg.atcahid.dev_identity))
    atcab_init(cfg)
    info = bytearray(4)
    assert atcab_info(info) == ATCA_SUCCESS
    print('\nDevice Part:')
    print('    ' + get_device_name(info))
   
    serial_number = bytearray(9)
    print('0x{:08X}'.format(atcab_read_serial_number(serial_number)))
    atcab_release()

The output of the following program is:
[image]

Can you tell me what am i doing wrong?

bryan-hunt, 2020-01-09:
It looks like you didn't compile HID support when you built the library.

agilesai1294, 2020-01-09:
@bryan-hunt I am using the following code on Jupyter notebook and i installed cryptoauthlib using the following command: pip install cryptoauthlib>=20191112
Let me know if I had to do something to get this working

agilesai1294, 2020-01-10:
@bryan-hunt I was able to resolve that. Now when I try to initialize I get this error.

[image]

Cryptoauthlib failed with error 0xE5( ATCA_RX_CRC_ERROR)

closed 2020-01-17 by agilesai1294