MicrochipTech/cryptoauthlib #130: libp11 claims pkcs11 ECDH function is not supported

drewwestrick, 2020-01-14:
I'm trying to use openssl and the pkcs11 engine to derive a shared secret. Currently, I can use a private key at the inkey and the public key on the ATECC608A to create a shared secret. When I attempt to flip the roles however I get an error shown below. I'm not sure if I need to modify the conf file to allow the private key to be used for the derive function or if there is another issue at play here.

`Key derivation failed
3069980688:error:82067054:PKCS#11 module:pkcs11_ecdh_derive:Function not supported:p11_ec.c:585:`

### Current key configuration. From the default template provided in the Wiki
> Object 0:
        URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=A1CA3745A533;token=0123EE;object=device;type=private
        Type: Private key
        Label: device
        Flags: CKA_PRIVATE; CKA_SENSITIVE;
        ID:
Object 1:
        URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=A1CA3745A533;token=0123EE;object=device;type=public
        Type: Public key
        Label: device
        ID:
Object 2:
        URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=A1CA3745A533;token=0123EE;object=root;type=public
        Type: Public key
        Label: root
        ID:

### **Command with PCKS11 as the peerkey**
`openssl pkeyutl -inkey drew_priv.pem -engine pkcs11 -peerform engine -derive -peerkey "pkcs11:token=0123EE;object=device" -hexdump`

This commands succeeds showing the shared secret.

### **Command with PCKS11 as the inkey**
`openssl pkeyutl -engine pkcs11 -keyform engine -derive -inkey "pkcs11:token=0123EE;object=device;type=private" -peerform PEM -peerkey drew_pub.pem -hexdump`

This command fails with the error above.

bryan-hunt, 2020-01-14:
This is likely a problem with the library configuration or the device configuration. Do you have the configuration zone data for the part and the library #defines that were used? The command definitely does work (to be fair I tried it without the -peerform option since the default for openssl is pem).

drewwestrick, 2020-01-15:
The only code in the library I changed was the i2c interface number. Do you have a good way for me to view / produce the configuration zone data? Also does anything need to be added or changed in the 0.config file? For instance, you can specify the term signer in one of the commented out lines. Is there something similar for the derive function? I could not find any good documentation on what other parameters can be added or changed in that config file.

bryan-hunt, 2020-01-15:
The configuration is fairly limited and what you see in the example configuration is what is supported. I'm wondering if there might be a versioning thing. Obviously libp11 seems to be up to date - the ECDH support was a relatively recent edition. In your build system (yocto or buildroot I assume) what hash are you pulling in and building?

drewwestrick, 2020-01-15:
@bryan-hunt I'm actually building directly from the git repo. I'm on the latest pkcs11 branch at commit a0007d2f6c42fddab5dca1575e0f404788829ddc. I believe the Wiki had me pulling from the pcks11 branch but it looks like that hasn't seen updates since July. I will go ahead and try and build the latest on the master branch.

bryan-hunt, 2020-01-15:
Okay I see the issue. Hold tight for a second

bryan-hunt, 2020-01-15:
@drewwestrick Update with the latest 2df0eb1 and let me know if everything builds and runs properly for you.

drewwestrick, 2020-01-15:
@bryan-hunt I went ahead and checked out 2df0eb1. I rebuilt the entire project and installed it. I'm now getting a **pkcs11_ecdh_derive:Function failed** error.

bryan-hunt, 2020-01-16:
@drewwestrick Okay the default configuration that is provided for experimenting with the part sets the ECDH mode for slot 0 to a legacy mode. So in your case to experiment with this if you change the primary private key to slot 2 and the ECDH command will work.

Generally moving forward we recommend tuning the configuration for your specific use cases and an FAE can definitely help you there.

drewwestrick, 2020-01-17:
@bryan-hunt Thanks for taking the time to help troubleshoot this issue. I will give your suggestion a try in the morning. I agree and have already reached out to my local Microchip resources to get some application help. If this works I should be able to present a rough POC to my client on the value this chip provides in securing their IP. I will work with an FAE if things proceed further. I will let you know how the testing goes.

drewwestrick, 2020-01-17:
@bryan-hunt Moving to slot 2 did the trip. Thanks again for all the help on this. I'm now able to use the derive function on the private key stored in the ATEC device. I'll go head and close this issue.

closed 2020-01-17 by drewwestrick