MicrochipTech/cryptoauthlib #133: Writing private / public keys / certificates with p11tool fails

jens-totemic, 2020-01-30:
I'm trying to load my own set of private key / public key and root certificate into a ATECC608A using `p11tool`. I'm using this configuration:

```
# Reserved Configuration for a device
# The objects in this file will be created and marked as undeletable
# These are processed in order. Configuration parameters must be comma
# delimited and may not contain spaces

# use i2c bus 0
interface = i2c,0xC0,0
freeslots = 1,2,3

# Slot 0 is the primary private key
object = private,device,0

# Slot 10 is the certificate data for the device's public key
#object = certificate,device,10

# Slot 12 is the intermedate/signer certificate data
#object = certificate,signer,12

# Slot 15 is a public key
object = public,root,15
```

This is what I see before trying to add any keys/certs:
```
p11tool --list-all --provider=/usr/lib/libcryptoauth.so
Object 0:
	URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=AD0F8356CCA2;token=0123F2;object=device;type=private
Token '0123F2' with URL 'pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=AD0F8356CCA2;token=0123F2' requires user PIN
Enter PIN: 
	Type: Private key (EC/ECDSA-SECP256R1)
	Label: device
	Flags: CKA_PRIVATE; CKA_SENSITIVE; 
	ID: 

Object 1:
	URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=AD0F8356CCA2;token=0123F2;object=device;type=public
	Type: Public key (EC/ECDSA-SECP256R1)
	Label: device
	ID: 

Object 2:
	URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=AD0F8356CCA2;token=0123F2;object=root;type=public
	Type: Public key (EC/ECDSA-SECP256R1)
	Label: root
	ID: 
```

1. Writing the private key fails with this:
```
p11tool --login --write "pkcs11:token=0123F2" --load-privkey private.key --label "test" --provider /usr/lib/libcryptoauth.so 
13518:13518:C_Initialize:144:
13518:13518:C_Initialize:145:CKR_OK(0)
13518:13518:C_GetInfo:162:
13518:13518:C_GetInfo:163:CKR_OK(0)
13518:13518:C_GetSlotList:187:
13518:13518:C_GetSlotList:188:CKR_OK(0)
13518:13518:C_GetTokenInfo:205:
13518:13518:pkcs11_token_get_info:372:Token Info: 1
13518:13518:pkcs11_token_get_info:405:Token Locked
13518:13518:C_GetTokenInfo:206:CKR_OK(0)
13518:13518:C_GetSlotInfo:196:
13518:13518:C_GetSlotInfo:197:CKR_OK(0)
13518:13518:C_OpenSession:274:
13518:13518:C_OpenSession:275:CKR_OK(0)
13518:13518:C_GetSessionInfo:301:
13518:13518:C_GetSessionInfo:302:CKR_OK(0)
Token '0123F2' with URL 'pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=AD0F8356CCA2;token=0123F2' requires user PIN
[... 443 lines cut ...]
13534:13534:C_GetSlotInfo:196:
13534:13534:C_GetSlotInfo:197:CKR_OK(0)
13534:13534:C_OpenSession:274:
13534:13534:C_OpenSession:275:CKR_OK(0)
13534:13534:C_GetSessionInfo:301:
13534:13534:C_GetSessionInfo:302:CKR_OK(0)
13534:13534:C_Login:343:
13534:13534:C_Login:344:CKR_OK(0)
13534:13534:C_CreateObject:367:
13534:13534:C_CreateObject:368:CKR_GENERAL_ERROR(5)
13534:13534:C_CloseSession:283:
13534:13534:C_CloseSession:284:CKR_OK(0)
Error writing certificate: PKCS #11 error.
```

I'm using the lastest version of the pkcs11 branch (commit 2df0eb145c7241263d07577394dd12f8b1e783f0)

```
p11tool --version
p11tool 3.6.7
```

A couple of questions:
- is there anything that I could be missing in the call to p11tool?
- why am I asked for a user pin? I have never set one. I only had to set a SO pin when calling `p11tool --initialize`

bryan-hunt, 2020-01-30:
Writing private keys into a device such as the ATECC608A is consider bad cryptographic hygiene and we do not recommend it. While the pkcs11 implementation does have the ability to perform the correct operations it is dependent on having the device programmed the correct configuration (private key writes enabled for the slot and an encryption key configured).

Have you discussed your use cases with an FAE yet? 

You can also submit a product support request through the mySupport portal on [microchip.com](http://www.microchip.com/support/hottopics.aspx). We are unable to provide the same level of product usage support via github as we do through the support portal.

For details on the mySupport Portal or how to use it please see the [instructions for using the mySupport portal](https://microchip.secure.force.com/microchipknowledge/articles/en_US/FAQ/How-to-submit-a-case)

jens-totemic, 2020-01-30:
Hi @bryan-hunt Thanks for the quick reply. Two questions regarding that:
- we are receiving the certificate, public and private key from AWS IoT and would need to write them into the device. We don't have an FAE, but this feels like a typical case. Do you have recommendations around how to handle this case? 
- even if writing of the private key is not recommended, at least writing the public key and certificate should work. Why do those operations fail?
- you mentioned enabling slots for writing private keys and configuring an encryption key. Could you share what changes we have to make to the configuration to make this possible?

bryan-hunt, 2020-01-30:
> * we are receiving the certificate, public and private key from AWS IoT and would need to write them into the device. We don't have an FAE, but this feels like a typical case. Do you have recommendations around how to handle this case?

AWS has several methods that are cryptographically safer than the above which is why it is not a recommended practice:
* Using a CSR (for example see the [wiki](https://github.com/MicrochipTech/cryptoauthlib/wiki/Greengrass-HSI#setup-the-greengrass-certificate)
* Using JITR/JITP (see [JITR/JITP](https://aws.amazon.com/blogs/iot/just-in-time-registration-of-device-certificates-on-aws-iot/), the [Zero touch kit](https://github.com/MicrochipTech/aws-iot-zero-touch-secure-provisioning-kit), and the [Microchip Trust Platform & Design Suite using Trust Flex](https://www.microchip.com/design-centers/security-ics/trust-platform/trustflex)
* Using the Trust & Go products and uploading the manifests. See the [Microchip Trust Platform & Design Suite](https://www.microchip.com/design-centers/security-ics/trust-platform/trust-go)

> * even if writing of the private key is not recommended, at least writing the public key and certificate should work. Why do those operations fail?

A public key test fails because you have not defined a slot in which to store the key. The label provided "test" must align with a slot with the same name in the configuration. The certificate write fails because in order to store the certificate in the device you must provide meta data about the certificate formatting to the library. The only way to do that is to use the [PKCS11_USE_STATIC_CONFIG](https://github.com/MicrochipTech/cryptoauthlib/wiki/PKCS11-Configuration-Reference) option. 

> * you mentioned enabling slots for writing private keys and configuring an encryption key. Could you share what changes we have to make to the configuration to make this possible?

Again that is outside the scope of the product support found here. Please see the [Microchip Trust Platform Design Suite](https://www.microchip.com/design-centers/security-ics/trust-platform/trust-go) for detailed use case documentation and examples.

jens-totemic, 2020-01-31:
Thanks @bryan-hunt, we will try out these alternative options. Regarding storing the public key:
I have added this line above the slot 15 definition:
```
object = public,test,13
```

Then I ran this again:
```
p11tool --login --write "pkcs11:token=0123F2" --load-pubkey public.crt --label "test" --provider /usr/lib/libcryptoauth.so 
Enter PIN: 
Error in pkcs11_write:1278: The request is invalid.
```

- Is there anything else I need to do to write the public key?
- Do you know if the request for the PIN is an issue?

jens-totemic, 2020-02-07:
I was able to write the public key after using the patch from https://github.com/MicrochipTech/cryptoauthlib/issues/135

I'm trying to write the public certificate and I was wondering if you can give more specifics regarding the use of `PKCS11_USE_STATIC_CONFIG`? Without using that setting, `pkcs11_object_create` calls `pkcs11_cert_x509_write` which eventually fails because `cert_def` is not set. I can see there's a call to `pkcs11_config_cert` inside `pkcs11_object_create` which does nothing - would that be the place to add configuration information for the certificate?

esguti, 2020-06-15:
@bryan-hunt Could you help with the code to write a certificate?

bryan-hunt, 2020-06-15:
The PKCS11 interface for the library is to allow access to the a provisioned part via from a system that is able to access objects via PKCS11. The first step is to get your provisioning correct and then use the PKCS11 interface to access those objects. This means the first step has to be one of these two options:

a) Use the trust platform development suite (likely the trust flex devices). See [Microchip Trust Platform - Trust Flex](https://www.microchip.com/design-centers/security-ics/trust-platform/trustflex)
b) Work with your FAE to develop your use cases and provisioning

flongo82, 2020-12-30:
Hi,
if I initialize the cryptochip with p11tool  --initialize shouldn't I be able to write the certificate and the CA certificate in the crypto? 

After setting up the PKCS#11 environment, I created a configuration file as follows:

```
interface = i2c,0xc0,1
device = ATECC508A
freeslots = 1,2,3
object = private,device,0
object = certificate,device,10
object = certificate,signer,12
object = public,root,15
```

and then I lauched:  

```
p11tool --initialize "pkcs11:serial=23F64EB645089FEE" --label TLS
```

If I list the resources:

```
p11tool --list-all pkcs11:model=ATECC508A;manufacturer=Microchip%20Technology%20Inc;serial=23F64EB645089FEE;token=00ABC
```

I get: 

```
Object 0:
        URL: pkcs11:model=ATECC508A;manufacturer=Microchip%20Technology%20Inc;serial=23F64EB645089FEE;token=00ABC;object=device;type=private
        Type: Private key
        Label: device
        Flags: CKA_PRIVATE; CKA_SENSITIVE; 
        ID: 

Object 1:
        URL: pkcs11:model=ATECC508A;manufacturer=Microchip%20Technology%20Inc;serial=23F64EB645089FEE;token=00ABC;object=device;type=public
        Type: Public key
        Label: device
        ID: 

Object 2:
        URL: pkcs11:model=ATECC508A;manufacturer=Microchip%20Technology%20Inc;serial=23F64EB645089FEE;token=00ABC;object=device;type=cert
        Type: X.509 Certificate
        Label: device
        ID: 

Object 3:
        URL: pkcs11:model=ATECC508A;manufacturer=Microchip%20Technology%20Inc;serial=23F64EB645089FEE;token=00ABC;object=signer;type=cert
        Type: X.509 Certificate
        Label: signer
        Flags: CKA_PRIVATE; 
        ID: 
```
Which seems coherent to me. 

Then I created a self-signed CA, generated the CSR and signed the corresponding certificate: 

```
mkdir ca && cd ca
openssl genrsa -des3 -out rootCA.key 4096
openssl req -x509 -new -nodes -key rootCA.key -sha256 -days 1024 -out rootCA.crt

cd .. && mkdir device && cd device
openssl req -engine pkcs11 -key "pkcs11:token=00ABC;object=device;type=private" -keyform engine -new -out device.csr -subj "/CN=device"
openssl x509 -req -in device.csr -CA ../ca/rootCA.crt -CAkey ../ca/rootCA.key -CAcreateserial -out device.crt -days 500 -sha256
```

Now I want to write the certificate in the crypto: 

```
p11tool --write "pkcs11:model=ATECC508A;manufacturer=Microchip%20Technology%20Inc;serial=23F64EB645089FEE;token=00ABC;object=device;type=cert" --load-certificate device.crt --label "device"
```
but I get: 

```
Error writing certificate: PKCS #11 error.
```

What I am doing wrong?

bryan-hunt, 2020-12-30:
@flongo82 The ATECCx08 devices are not general purpose "smartcard" chips. In order to handle certificates the library needs to be informed of the certificate definition which involves compiling it into the the library. In other words what you're attempting to do is simply not supported by the library. The pkcs11 library does support certificates that can preloaded in the ATECC608A-TNGTLS devices which you can specify in the configuration file using the `device = ATECC608A-TNGTLS' option and removing all other object configuration.

I would suggest starting with the [Microchip Trust Platform Development Suite](https://www.microchip.com/design-centers/security-ics/trust-platform) and developing your use cases there and then when your configuration and use cases are ready moving to pkcs11.

flongo82, 2020-12-31:
Hi @bryan-hunt,
thank you very much for the quick reply. My use case for the time being is client authentication with a Mosquitto MQTT broker. I'm having some issues on that side too as you can see here: [https://github.com/eclipse/mosquitto/issues/1986](url) but that is another story. :-) 

Right now I do not actually need to store either the device certificate or the CA certificate in the cryptochip as in the Mosquitto client configuration I should be able to provide them from the filesystem while the private key is the only thing that should be taken from the crytochip. This is basically what happens also in the Greengrass use case, for example. However, I'm trying to understand in details everything and this is the reason I was doing these tests. 

I think I got the general idea from your answer but I have some specific questions if you would be so kind to help me. 

1. What does exactly the command "p11tool --initialize" do? 
 
As far as I understood it should perform both the configuration and the provisioning of the cryptochip accordingly to the cryptoauthlib configuration file but from your answer I understand that maybe this is not the case. The reason why I was trying to use p11tool to configure the chip is that is should be "easier" (there is a human-readable configuration file) and automatable.    
 
2. What is the exact cryptochip configuration that is used while executing "p11tool --initialize" in my case (with my cryptoauthlib configuration file)? 

As far as I understood it should be a standard TLS configuration with private and public keys, the certificate of the device and the certificate of the signer (which in my case should be the CA). Is there any documentation about the cryptochip configuration that is used in this case? 

3. Is there any documentation about the syntax of the configuration file of the cryptoauthlib? 

What is not clear to me is the semantic of the "freeslots" directive and why the same label is used in several "object" directives. 

4. Could you please provide a pointer to the ATECC608A-TNGTLS cryptochip configuration? 
 
If I understand properly this is a TLS configuration that should be more flexible but I would like to understand exactly how the cryptochip is configured in that case so that I know how to invoke the corresponding p11tool commands. 

I will surely have a look at the Microchip Trust Platform Development Suite because my final goal is understanding how to configure the cryptochip myself (either through p11tool or in any other more correct way).

bryan-hunt, 2020-12-31:
The p11tool --initialize step calls the C_TokenInitialize routine. For cryptoauthlib that is a basic example configuration. This should never be used since the use case for the cryptochip should be established before the product is ever produced. The configuration of these devices is very detailed and you need to reference the datasheet. The configuration file for pkcs11 configures the pkcs11 objects (what device interface to use and what slot handles the crypto operations for that object).

So to reiterate - you need to work on configuration and use cases first. Your best resource for that is the Trust Platform Development Suite.

You can download the datasheet for the ATECC608A-TNGTLS part from [microchip.com](https://www.microchip.com/wwwproducts/en/ATECC608A-TNGTLS)

user-redans, 2023-03-03:
Hi @bryan-hunt,
what's your suggestion on using embedded-devices with ATECC608 in the following scenario:
- there's a linux server/VM which can be configured under root by us to manage a list of embedded-devices;
- there might be a number of embedded-devices-ATECC608 which connects (via openssl/CURL, custom-built ssl-client/server app) to the server and gets "authenticated" and exchange sensitive data. 
- if the server has also an ATECC608A chip, how does it check the public-keys of all embedded-devices in-chip? I mean to avoid using files like "authorized_keys" for ssh.

We want a safe mechanism for authenticating/communication of these embedded-devices with the server.  
I'd like your opinion on how to manage private/public keys of server/devices ?