MicrochipTech/cryptoauthlib #136: Python OSError undefined symbol: hid_open on Linux armv7

renesy, 2020-02-12:
I'm using a Raspberry Pi (armv7) and install cryptoauthlib via pip3, Python 3.7.

After upgrading to version 20200205 I'm getting this error when attempting to import cryptoauthlib:
`from cryptoauthlib import load_cryptoauthlib`

`/lib/python3.7/site-packages/cryptoauthlib/libcryptoauth.so: undefined symbol: hid_open`

The device I'm using is ATECC608A

bryan-hunt, 2020-02-12:
Looking at piwheels.org it appears that the wheel for python 3.7 is not building which means pip had to build it on your pi. I'm going to run some tests myself to determine what might be breaking in the python 3.7 build on the pi.

Can you attach your logs from pip?

Uninstall first
```
> pip3 uninstall cryptoauthlib
```

Reinstall without a cache
```
pip3 install --no-cache cryptoauthlib
```

Thanks

renesy, 2020-02-12:
> Reinstall without a cache
> 
> ```
> pip3 install --no-cache cryptoauthlib
> ```
> 
> Thanks

```
% pip3 install --no-cache cryptoauthlib
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting cryptoauthlib
  Downloading cryptoauthlib-20200205.tar.gz (274 kB)
     |████████████████████████████████| 274 kB 1.8 MB/s 
Building wheels for collected packages: cryptoauthlib
  Building wheel for cryptoauthlib (setup.py) ... done
  Created wheel for cryptoauthlib: filename=cryptoauthlib-20200205-cp37-cp37m-linux_armv7l.whl size=101163 sha256=633c58587c7c52ee0d1c08b4069e9c2c181349dd12990dfa80a6dd6a8544c335
  Stored in directory: /tmp/pip-ephem-wheel-cache-fbo5k486/wheels/9f/5d/29/7f0d60426b6ece768a0e8c3dc0c77568ded942eab7cc8fa01f
Successfully built cryptoauthlib
Installing collected packages: cryptoauthlib
Successfully installed cryptoauthlib-20200205
pip3 install --no-cache cryptoauthlib  29.37s user 5.13s system 95% cpu 36.000 total
```

closed 2020-06-15 by bryan-hunt

bernardoaraujor, 2020-12-24:
same problem here.
reinstalling with `--no-cache` has no effect.

bryan-hunt, 2020-12-24:
@bernardoaraujor I assume you are cross compiling? If you are and your target does not have usb libraries that match your build environment then you can disable support by setting the environment variable CRYPTOAUTHLIB_NOUSB wherever you are performing the build - alternatively you can fix this by ensuring your build environment properly sets up CMake library search paths to match the target rather than the system.

bernardoaraujor, 2020-12-24:
hi @bryan-hunt thanks for the fast response.
I'm not cross-compiling. I'm under Debian for armv7l.

bernardoaraujor, 2020-12-24:
I uninstalled and did:
```
CRYPTOAUTHLIB_NOUSB=1 pip3 install --no-cache cryptoauthlib
```

now the error is:
```
$ python3 info.py
Traceback (most recent call last):
  File "info.py", line 96, in <module>
    info(args.iface, args.device, **parse_interface_params(args.params))
  File "info.py", line 35, in info
    cfg = eval('cfg_at{}a_{}_default()'.format(atca_names_map.get(device), atca_names_map.get(iface)))
  File "<string>", line 1, in <module>
  File "/home/usbarmory/.local/lib/python3.7/site-packages/cryptoauthlib/iface.py", line 144, in cfg_ateccx08a_kithid_default
    return ATCAIfaceCfg.in_dll(get_cryptoauthlib(), 'cfg_ateccx08a_kithid_default')
ValueError: /home/usbarmory/.local/lib/python3.7/site-packages/cryptoauthlib/libcryptoauth.so: undefined symbol: cfg_ateccx08a_kithid_default
```

bryan-hunt, 2020-12-24:
@bernardoaraujor Okay I've made a note - we have the next version of the library coming out shortly and this bug should be quick to fix.

bernardoaraujor, 2021-01-24:
hi @bryan-hunt any updates on the new release?