MicrochipTech/cryptoauthlib #137: Exiting with error code 04(DECODING_ERROR), (unexpected tag value, (where it checks for ‘0x03')

AdityaHPatwardhan, 2020-02-18:
- I have been using `ATECC608A` module , on `ESPWROOM32SE`, What I want to do is 
generate key, generate CSR from `ATECC` chip, sign the `CSR` on host and provide cert (in der ) and cert_def ( generated by using `cert2certdef.py` ) , to the chip and store the device cert on the chip. I want to do all of this in one go.

- I am able to get CSR and sign it on host. but when I provide cert definition to the chip along with the cert in der format, It is failing with error code 04 ( DECODING_ERROR) , it is failing where it checks the tag value `0x03`. ( In this case I am not providing cert definition at compile time, rather sending it as a string from host to ESP32 and decoding it on ESP32 itself and putting the values in variable `atcacert_def_t cert_def`, 

- When I take the same cert and now provide cert def and cert( der format) at compile time, It is succeeding. 
Is there any thing I am missing here.?
It would be really helpful if I get reply ASAP.
Thank You.

AdityaHPatwardhan, 2020-02-18:
@bryan-hunt, Also is it compulsory to keep Signer Name or Device Name as `something FFFF` , for `cert2certdef.py` to work.

AdityaHPatwardhan, 2020-03-20:
my code looks with which I start the TLS connection looks kinda like this

```
    if(0 != (ret = atcab_init(&cfg_ateccx08a_i2c_default))) {
        ESP_LOGE(TAG, " failed\n  ! atcab_init returned %02x", ret);
        return NETWORK_X509_ROOT_CRT_PARSE_ERROR;
        /* For now keeping some random error code,
         * ToDo- replace it with appropriate error code */
    }

    if ((ret = atca_mbedtls_pk_init(&(tlsDataParams->pkey),0)) != 0) {
        ESP_LOGE(TAG, "Failed to atca_mbedtls_pk_init, returned %02x ", ret);
        return NETWORK_X509_ROOT_CRT_PARSE_ERROR; 
        /* For now keeping some random error code,
         * ToDo- replace it with appropriate error code */
    }

    if ((ret = atca_mbedtls_cert_add(&(tlsDataParams->clicert), &g_cert_def_2_device)) != 0) {
        ESP_LOGE(TAG, "Failed to atca_mbedtls_cert_add device, returned %02x", ret);
        return NETWORK_X509_DEVICE_CRT_PARSE_ERROR;
    }
    if ((ret = atca_mbedtls_cert_add(&(tlsDataParams->clicert), &g_cert_def_1_signer)) != 0) {
        ESP_LOGE(TAG, "Failed to atca_mbedtls_cert_add signer, returned %02x", ret);
        return NETWORK_X509_DEVICE_CRT_PARSE_ERROR;
    }
    if(0 != (ret = mbedtls_ssl_conf_own_cert(
            &(tlsDataParams->conf), &(tlsDataParams->clicert), &(tlsDataParams->pkey))))
    {
        ESP_LOGE(TAG, " failed\n ! mbedtls_ssl_conf_own_cert returned %d\r\n", ret);
        return NETWORK_X509_DEVICE_CRT_PARSE_ERROR;
    }

```

I have watched the `TLS` handshake go ( with wire-shark) , It goes well till the `change-cipher-message` is sent from the client, after that server just sends peer-close notify and TLS handshake fails, I have checked that the hardware signing and verification is OK by verifying  the hardware signed data with mbedTLS software implementation.

I even have registered the certificate with AWS so it is working and I have checked it AWS connections using mbedTLS ( without using ATECC ).The connection succeeds as expected.

@bryan-hunt, @BenUdall-Microchip  Can you please take a look and see if I am missing something.

anani112, 2020-09-20:
@AdityaHPatwardhan did you find what was going on with this issue? I might be going through the same issue

AdityaHPatwardhan, 2020-09-20:
@anani112  Yeah, I did find the problem,  the problem in my case was with date and time of the certificate , When I used to create the certificate on host machine and then store it in the ATECC, as there are some constraints on storing time of certificate in the ATECC so the date and time of my cert were messed up when it got sent to the aws ( from ATECC) at the time of authentication, 
So after I created the certificate on host which conformed to the ATECC standards, I did not face that problem again.
You can have a look at the [link](https://github.com/espressif/esp-cryptoauthlib/blob/c3d3a69021cfec3236ca2c0b63be4048ec6643a4/esp_cryptoauth_utility/helper_scripts/cert_sign.py#L40) for more details. The repo also creates the cert compatible with the ATECC on host.

and thanks for reminding me that I have to close this issue :p.

closed 2020-09-20 by AdityaHPatwardhan

anani112, 2020-09-20:
thanks for getting back to me about this and for the suggestions !