MicrochipTech/cryptoauthlib #139: ECC608 with USB kit is not detected

vgottardi2, 2020-02-20:
The device is not found because the code checks dev_type (ECC108) read from the board against device_match (ECC508) set in configuration
https://github.com/MicrochipTech/cryptoauthlib/blob/102d7fcfa2cc1671fddb814bf8079756f70f7109/lib/hal/kit_protocol.c#L166

`Enumerate HID device(s)
Kit USB Device Node: 0002:0006:00
  Manufacturer A (C)
  PID 8978
  VID 1003

HID layer: Write: board:device(00)
HID layer: Read: **ECC108** TWI 00(C0)
HID layer: Write: board:device(01)
HID layer: Read: C5()
kit_init() Failed`

bryan-hunt, 2020-02-20:
I believe this is a problem with older firmware in the kit that is improperly identifying the 608 as a 108 combined with an update to the protocol driver in cryptoauthlib that is attempting to pick exactly the device that is specified. There should be an update to the firmware in the kit that will properly identify the target.

Which kit are you using?

vgottardi2, 2020-02-20:
It's an AT88 REV-2.0

bryan-hunt, 2020-02-20:
@KalyanCManukonda Can you provide the AT88CK101 update instructions and updated firmware url?

vgottardi2, 2020-02-20:
AT88Microbase to be precise - http://ww1.microchip.com/downloads/en/devicedoc/doc8723.pdf
Quite old

closed 2022-05-01 by vgottardi2