MicrochipTech/cryptoauthlib #140: p11tool : user PIN

FabienLehoussel, 2020-02-27:
I try to use ATECC608A on Sama5d3 Ethernet Development kit.

I compiled with buildroot : 
 - last pkcs11 cryptoauthlib version (2df0eb145c7241263d07577394dd12f8b1e783f0)
 - libp11 0.4.10
 - gnutls 3.6 with tools
 - p11-kit 0.23.16.1

```
root@buildroot:~# uname -a
Linux buildroot 4.19.56-linux4sam-6.1 #1 Wed Feb 26 14:52:43 CET 2020 armv7l GNU/Linux
root@buildroot:~#

root@buildroot:~# ll /usr/lib/libcryptoauth.so
-rwxr-xr-x    1 root     root       91.9K Feb 27  2020 /usr/lib/libcryptoauth.so
root@buildroot:~#
``` 

### I2C

Chip is detected (bus 2)

```
root@buildroot:~# i2cdetect -y 2
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: 60 -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
root@buildroot:~#
```
### Cryptoauthlib conf

```
root@buildroot:~# cat /etc/cryptoauthlib/cryptoauthlib.conf
# Cryptoauthlib Configuration File
filestore = /var/lib/cryptoauthlib
root@buildroot:~#
```

### slot conf

```
root@buildroot:~# cat /var/lib/cryptoauthlib/0.conf
# Reserved Configuration for a device
# The objects in this file will be created and marked as undeletable
# These are processed in order. Configuration parameters must be comma
# delimited and may not contain spaces

# use i2c bus 2 address 0x60
interface = i2c,0xC0,2
freeslots = 1,2,3

# Slot 0 is the primary private key
object = private,device,0

# Slot 10 is the certificate data for the device's public key
#object = certificate,device,10

# Slot 12 is the intermedate/signer certificate data
#object = certificate,signer,12

# Slot 15 is a public key
object = public,root,15
root@buildroot:~#
```

### pkcs11 conf 
```
root@buildroot:~# cat /etc/pkcs11/pkcs11.conf
# This setting controls whether to load user configuration from the
# ~/.config/pkcs11 directory. Possible values:
#    none: No user configuration
#    merge: Merge the user config over the system configuration (default)
#    only: Only user configuration, ignore system configuration
user-config: merge
root@buildroot:~#
```

### Cryptoauthlib module

```
root@buildroot:~# cat /usr/share/p11-kit/modules/cryptoauthlib.module
module: /usr/lib/libcryptoauth.so
critical: yes
trust-policy: yes
managed: yes
log-calls: no
root@buildroot:~#
```

### PIN problem 

```
root@buildroot:~# p11-kit list-modules
p11-kit-trust: p11-kit-trust.so
    library-description: PKCS#11 Kit Trust Module
    library-manufacturer: PKCS#11 Kit
    library-version: 0.23
cryptoauthlib: /usr/lib/libcryptoauth.so
    library-description: Cryptoauthlib PKCS11 Interface
    library-manufacturer: Microchip Technology Inc
    library-version: 3.2
    token: 0123EE
        manufacturer: Microchip Technology Inc
        model: ATECC508A
        serial-number: XXXXXXXXXXXX
        firmware-version: 255.255
        flags:
               rng
               login-required
root@buildroot:~#
```

````
root@buildroot:~# p11tool -d 99 --list-all
Setting log level to 99
|<2>| p11: Initializing module: p11-kit-trust
|<2>| p11: Initializing module: cryptoauthlib
|<3>| ASSERT: pkcs11.c[compat_load]:894
|<2>| p11: No login requested.
|<2>| p11: No login requested.
|<3>| ASSERT: pkcs11.c[find_token_num_cb]:2334
|<3>| ASSERT: pkcs11.c[_gnutls_pkcs11_token_get_url]:2418
|<2>| p11: No login requested.
|<2>| p11: No login requested.
|<3>| ASSERT: pkcs11.c[pkcs11_import_object]:2142
|<3>| ASSERT: pkcs11.c[pkcs11_import_object]:2142
|<3>| ASSERT: pkcs11.c[pkcs11_read_pubkey]:1893
|<3>| ASSERT: pkcs11.c[pkcs11_obj_import_pubkey]:1942
|<3>| ASSERT: pkcs11.c[pkcs11_import_object]:2165
|<3>| ASSERT: pkcs11.c[find_multi_objs_cb]:3321
|<3>| ASSERT: pkcs11.c[pkcs11_import_object]:2142
|<3>| ASSERT: pkcs11.c[pkcs11_read_pubkey]:1893
|<3>| ASSERT: pkcs11.c[pkcs11_obj_import_pubkey]:1942
|<3>| ASSERT: pkcs11.c[pkcs11_import_object]:2165
|<3>| ASSERT: pkcs11.c[find_multi_objs_cb]:3321
|<3>| ASSERT: pkcs11.c[find_multi_objs_cb]:3107
Object 0:
        URL: pkcs11:model=ATECC508A;manufacturer=Microchip%20Technology%20Inc;serial=XXXXXXXXXXXX;token=0123EE;object=device;type=private
Token '0123EE' with URL 'pkcs11:model=ATECC508A;manufacturer=Microchip%20Technology%20Inc;serial=XXXXXXXXXXXX;token=0123EE' requires user PIN
Enter PIN:
No PIN given.
root@buildroot:~#
```
```
root@buildroot:~# p11tool -d 99 --initialize "pkcs11:serial=XXXXXXXXXXXX" --labe
l test
Setting log level to 99
|<2>| p11: Initializing module: p11-kit-trust
|<2>| p11: Initializing module: cryptoauthlib
|<3>| ASSERT: pkcs11.c[compat_load]:894
Enter Security Officer's PIN:
Initializing token... |<3>| ASSERT: pkcs11_write.c[gnutls_pkcs11_token_init]:1237
|<2>| p11: The operation failed

Error in pkcs11_init:1453: PKCS #11 error.
root@buildroot:~#
````

What is the pin ?????
How can i use private/public key ?

twendtland, 2020-04-09:
hey, have you tried "1234" as the pin? Your last command doesn't work for me either, giving me the same result, but I've been able to use 

`p11tool --login --generate-privkey ecdsa --curve secp256r1 "pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=XXXXXXXXXXXX;token=0123EE;object=device;type=public"`

successfully. 

However, apart from that I'm very much struggling with the usage of the ATECx0, both with OpenSSL and with p11tool. Anyhow, the pin should definitely be documented somewhere, but I feel like the documentation is not very complete.

dspearl-amazon, 2020-04-10:
I am also being prompted for a pin unexpectedly with similar config, 1234 is not working

FabienLehoussel, 2020-05-11:
Solution ?

vishalSpintly, 2021-12-28:
I have been facing the same problem with my test of the ATECcc608 connected over I2C to a Linux host.
Following the instructions on installation and configuration as provided, I am able to see the ATECcc608 device using the p11-kit tools.
However, I am facing the same issue of PIN being requested.
Anyone have the solution or pointers to what needs to be done? Clearly the documentation is missing some steps that need to be carried out prior to those described.

All help appreciated.

closed 2022-11-22 by github-actions[bot]

bryan-hunt, 2022-12-06:
Unless you specifically configured it that way then it isn't required and the problem is that you're using outdated libraries.