MicrochipTech/cryptoauthlib #141: PKCS#11 token initialization error

Anabel-Dev, 2020-03-04:
Greetings of the day!

We are using ATECC608A chip to encrypt a folder via pkcs11-helper. At some point, we are trying to initialize the token but it remains uninitialized and we also dont know the user pin. Only the token info and the slot info are available. Can you help us initialize the token?

$ pkcs11-tool --module=/usr/lib/libcryptoauth.so -L
Available slots:
Slot 0 (0x0): 0_3_0
  token state:   uninitialized

$ pkcs11-tool --module=/usr/lib/libcryptoauth.so -I
Cryptoki version 2.40
Manufacturer     Microchip Technology Inc v
Library          Cryptoauthlib PKCS11 Interface (ver 3.2)
Using slot 0 with a present token (0x0)

Anabel-Dev, 2020-03-20:
Hello
To initialize the token you need to lock the data zone in the cryptochip(ATECC608A). Then initialize the token using pkcs11-tool. If you are getting the CANT_LOCK error, then set the flag.

closed 2020-03-20 by Anabel-Dev

SimoneBongini, 2024-07-05:
hello @Anabel-Dev, how can I lock the data zone in the cryptochip?

Anabel-Dev, 2024-07-05:
Hi Simone,

Once you lock the data zone, there's no unlock command or go back.

Please check with the cryptochip technical team for more info

Thanks,
Anabel