MicrochipTech/cryptoauthlib #142: Bad  atcacert_def_t declaration

JAHTKELD, 2020-03-07:
Hello every body.
I'm working with ATECC608A chip and I find that the  **atcacert_def_t** declaration has the **cert_template** and ***ca_cert_def** are   _const_
You have embed the **cert_template**  the **ca_cert_def** in your code and compile for each board that you have with the ATECC608A. If you has hundreds of devices in the market is imposible to update  them with a new firmware by OTA. I think that all data you need to work with the ATECC608A must be inside of ATECC608A not in the firmware. I'm looking to rewrite this library but my knowledge is limited.  I think that the date of  **cert_template** is not important (you get them from ATECC608A) but I need to change the size and also i need to change the **STDCERT_SIGNATURE count** . it seems simple but i have tried to change this struct to variable and typecast to const and allways I get the error that their are const.
I think that the microchip developpers doesn`t thought in a real solution whit this library

bryan-hunt, 2020-03-07:
Microchip has numerous resources one can use. One of the best is to contact your local FAE - there are hundreds around the globe trained to work with you in your region. You can find the contact information for you local sales via the [online sales directory](https://www.microchip.com/salesdirectory) who will work with you through the entire product development lifecycle.

You can also submit a product support request through the mySupport portal on [microchip.com](http://www.microchip.com/support/hottopics.aspx). Which will be answered by our team of FAEs.

For details on the mySupport Portal or how to use it please see the [instructions for using the mySupport portal](https://microchip.secure.force.com/microchipknowledge/articles/en_US/FAQ/How-to-submit-a-case)

> I think that the microchip developpers doesn`t thought in a real solution whit this library

I will assume that anybody will reading this will understand that this is both inappropriate and unprofessional. 

> You have to embed the cert_template the ca_cert_def in your code and compile for each board that you have with the ATECC608A. If you has hundreds of devices in the market is imposible to update them with a new firmware by OTA.

The ATECC608A is widely used by hundreds of customers for thousands of products. The above statement is merely a lack of understanding of how the APIs work to fulfill cryptographic and PKI use cases.

> I think that all data you need to work with the ATECC608A must be inside of ATECC608A not in the firmware. 

The ATECC608A is fundamentally a cryptographic processor with secure key storage. The amount of available space inside the device is limited and is insufficient except for the most trivial of X509 certificates. Even when using a trivial certificate for one key it leaves no room for other keys or an intermediate certificate which is required for most PKI.

In order to address the needs of customers that do require X509 based PKI we have provide a method of extracting important (dynamic) information from a certificate and storing it in the device and then correspondingly reconstruct those certificates from the stored information. When you look at an X509 certificate most of it does not change. Thus the atcacert functions are designed to utilize a template (which supports most X509 features) which is then embedded in the firmware. The explanation for the format can be found in the [Application Note](http://ww1.microchip.com/downloads/en/AppNotes/Atmel-8974-CryptoAuth-ATECC-Compressed-Certificate-Definition-ApplicationNote.pdf). The features of the atcacert module are quite extensive as seen in the [API documentation](https://microchiptech.github.io/cryptoauthlib/a01992.html). 

This is just one possible solution. A customer is always free to write their own solution - but to this point no customer has informed us that the provide solution does not satisfy their needs. 

> I'm looking to rewrite this library but my knowledge is limited. I think that the data of cert_template is not important (you get them from ATECC608A) but I need to change the size and also I need to change the STDCERT_SIGNATURE count . it seems simple but I have tried to change this struct to variable and typecast to const and allways I get the error that their are const.

There are numerous examples for how to interact with the atcacert API. I suggest that perhaps the [trust platform design suite](https://microchipdeveloper.com/authentication:trust-platform) is the proper place to start.

If using preprovisioned devices such Trust & Go the [tng api](https://microchiptech.github.io/cryptoauthlib/a01999.html) is the proper way to retrieve the reconstructed certificates for those that do not have experiance in working with the atcacert api.

closed 2020-03-07 by bryan-hunt

JAHTKELD, 2020-03-07:
Hello:
I have been developing electronic products with microprocessors, hardware and software for more than 25 years. I expected a product to be able to integrate into my designs without having to delve into tls communications. I have tested the ATECC508A and its kit and to use it with Azure it is not easy since you have to validate the certificate. Now I am using the ATECC608A, I also have the ATECC608A-TNGTLS but it does not work for me since it takes months with "coming soon" for Azure, and I do not understand the difference between the use of thumbprint and certificates when integrating it into the communication tls. I have been trying to solve my secure connection with Azure for 3 months and contacting microchip.
I do not try to offend anyone, but I agree with colleagues, more and more hardware and software solutions are sold with the idea that they are quick to apply, but as it is something different from the examples and manuals they offer you, you find many problems and you have to spend many hours solving library problems.
Thanks for answering,

CarlMitchellKT, 2020-03-09:
I agree that this is difficult, it was one of the harder parts of the process for me. But most of the difficulty isn't (IMO) the fault of this library or its design, it's the fault of the (ridiculously arcane) X.509 spec. OpenSSL is far, far more painful to use for this than Microchip's tools, sadly I ended up using both.

Everything is offsets in bytes from the start of the file. The cert must be in DER form. https://www.cs.auckland.ac.nz/~pgut001/#standards is helpful, both the dumpasn1.c program (or the JS decoder linked from there) for finding offsets and the X.509 file. Most of the offsets are to the actual data, after the ASN.1 length header, but at least one (the signature) is to the header value. 

Quickly checking the signature offset is to the start of the BIT STRING header for it, but everything else is to the start of the respective data block (IE after the headers). 

Microchip provided us with a package of information when we signed an NDA to have them provision these devices in their facility. That included several Python scripts which could generate example certificates and their c/h files respectively. They're very helpful, and it would be nice if Microchip included that with this library.

JAHTKELD, 2020-03-13:
CarlMitchellKT,  thanks for you information