MicrochipTech/cryptoauthlib #144: Segmentation fault with pkcs11 branch when using HID interface

flongo82, 2020-03-14:
Hi everyone,
I'm on a Ubuntu 19.10 and I'm using an LXD container with same distribution to perform some tests with the CryptoAuthentication SOIC XPRO Starter Kit on an ATECC508a chip. I wanted to test the PKCS#11 interface and I performed the following: 

 - cloned the latest version of the lip11 repo, compiled and installed it
 - cloned the latest version of the pkcs11 branch of this repo, compiled and installed it
 - performed all the other steps described in the wiki page to configure the cryptoauthlib PKCS11 library

Of course, considering that I'm using the CryptoAuthentication SOIC XPRO Starter Kit, I changed the interface to "hid" in /var/lib/cryptoauthlib/0.conf. I'm using p11-kit-proxy and the p11-tool. 

As soon as I launch any p11-tool command (even a simple --list-tokens), I get a segmentation fault. No other output even with --debug=999. 

I believe this issue is somehow linked to [this](https://github.com/MicrochipTech/cryptoauthtools/issues/16#issue-511851564) one considering that the HID interface does not work with versions of cryptoauthlib > 20190517. But, if I'm not wrong that version is earlier than the pkcs11 branch.

flongo82, 2020-03-14:
If you want to reproduce, here are the steps I followed: 

From the host machine:
`lxc launch ubuntu:19.10 cryptochip`
`lxc config device add cryptochip starterkit usb vendorid=03eb productid=2312`
`lxc exec cryptochip -- sudo --login --user ubuntu`

From inside the container:
`sudo apt install -y autoconf libtool-bin pkg-config libssl-dev`
`git clone https://github.com/OpenSC/libp11.git`
`cd libp11/`
`./bootstrap && ./configure && make && sudo make install`
`cd ..`
`sudo apt-get install -y cmake libudev-dev`
`git clone --single-branch -b pkcs11 https://github.com/MicrochipTech/cryptoauthlib`
`cd cryptoauthlib`
`cmake . && make && sudo make install`
`cd /var/lib/cryptoauthlib`
`cp slot.conf.tmpl 0.conf`
`sed -i 's/i2c,0xB0/hid/g' 0.conf`
`sudo apt-get install -y p11-kit`
`sudo mkdir /etc/pkcs11`
`cat << EOF | sudo tee /etc/pkcs11/pkcs11.conf`
`# This setting controls whether to load user configuration from the`
`# ~/.config/pkcs11 directory. Possible values:`
`#    none: No user configuration`
`#    merge: Merge the user config over the system configuration (default)`
`#    only: Only user configuration, ignore system configuration`
`user-config: merge`
`EOF`
`cat << EOF | sudo tee /usr/share/p11-kit/modules/cryptoauthlib.module`
`module: /usr/lib/libcryptoauth.so`
`critical: yes`
`trust-policy: yes`
`managed: yes`
`log-calls: no`
`EOF`
`sudo apt-get install -y gnutls-bin`
`sudo p11tool --list-tokens --debug=999`

bryan-hunt, 2020-03-14:
Try adding `-DATCA_HAL_KIT_HID=on` to your cmake invocation

flongo82, 2020-03-15:
When I added `-DATCA_HAL_KIT_HID=on` option to the cmake invocation it gave me the following error: 

`FATAL_ERROR,Missing Build Dependencies for USB - install libusb-1.0-0-dev or libudev-dev`

This is strange because libudev-dev was already installed: 

`$ sudo apt-get install libudev-dev`
`Reading package lists... Done`
`Building dependency tree`
`Reading state information... Done`
`libudev-dev is already the newest version (242-7ubuntu3.7).`
`0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.`

By the way, I installed also libusb-1.0-0-dev (I also tried with installing ONLY libusb-1.0-0-dev) but launching

`sudo p11tool --list-tokens --debug=999`

gives me always Segmentation fault. 

I also tried with valgrind if it could be useful (as root because I did not manage to launch it with sudo): 

```
# valgrind p11tool --list-tokens --debug=999
==14088== Memcheck, a memory error detector
==14088== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==14088== Using Valgrind-3.15.0 and LibVEX; rerun with -h for copyright info
==14088== Command: p11tool --list-tokens --debug=999
==14088==
Setting log level to 999
==14088== Invalid read of size 8
==14088==    at 0x544EF80: deleteATCADevice (atca_device.c:100)
==14088==    by 0x5454163: atcab_release (atca_basic.c:147)
==14088==    by 0x5465293: pkcs11_slot_init (pkcs11_slot.c:207)
==14088==    by 0x5462B76: pkcs11_init (pkcs11_init.c:249)
==14088==    by 0x4C7D133: ??? (in /usr/lib/x86_64-linux-gnu/libp11-kit.so.0.3.0)
==14088==    by 0x4C7D2F2: ??? (in /usr/lib/x86_64-linux-gnu/libp11-kit.so.0.3.0)
==14088==    by 0x4C7FE6F: p11_kit_modules_initialize (in /usr/lib/x86_64-linux-gnu/libp11-kit.so.0.3.0)
==14088==    by 0x4C80138: p11_kit_modules_load_and_initialize (in /usr/lib/x86_64-linux-gnu/libp11-kit.so.0.3.0)
==14088==    by 0x48D8EA9: ??? (in /usr/lib/x86_64-linux-gnu/libgnutls.so.30.25.0)
==14088==    by 0x48DA916: gnutls_pkcs11_init (in /usr/lib/x86_64-linux-gnu/libgnutls.so.30.25.0)
==14088==    by 0x116BE2: ??? (in /usr/bin/p11tool)
==14088==    by 0x4A701E2: (below main) (libc-start.c:308)
==14088==  Address 0x8 is not stack'd, malloc'd or (recently) free'd
==14088== 
==14088== 
==14088== Process terminating with default action of signal 11 (SIGSEGV)
==14088==  Access not within mapped region at address 0x8
==14088==    at 0x544EF80: deleteATCADevice (atca_device.c:100)
==14088==    by 0x5454163: atcab_release (atca_basic.c:147)
==14088==    by 0x5465293: pkcs11_slot_init (pkcs11_slot.c:207)
==14088==    by 0x5462B76: pkcs11_init (pkcs11_init.c:249)
==14088==    by 0x4C7D133: ??? (in /usr/lib/x86_64-linux-gnu/libp11-kit.so.0.3.0)
==14088==    by 0x4C7D2F2: ??? (in /usr/lib/x86_64-linux-gnu/libp11-kit.so.0.3.0)
==14088==    by 0x4C7FE6F: p11_kit_modules_initialize (in /usr/lib/x86_64-linux-gnu/libp11-kit.so.0.3.0)
==14088==    by 0x4C80138: p11_kit_modules_load_and_initialize (in /usr/lib/x86_64-linux-gnu/libp11-kit.so.0.3.0)
==14088==    by 0x48D8EA9: ??? (in /usr/lib/x86_64-linux-gnu/libgnutls.so.30.25.0)
==14088==    by 0x48DA916: gnutls_pkcs11_init (in /usr/lib/x86_64-linux-gnu/libgnutls.so.30.25.0)
==14088==    by 0x116BE2: ??? (in /usr/bin/p11tool)
==14088==    by 0x4A701E2: (below main) (libc-start.c:308)
==14088==  If you believe this happened as a result of a stack
==14088==  overflow in your program's main thread (unlikely but
==14088==  possible), you can try to increase the size of the
==14088==  main thread stack using the --main-stacksize= flag.
==14088==  The main thread stack size used in this run was 8388608.
==14088== 
==14088== HEAP SUMMARY:
==14088==     in use at exit: 313,891 bytes in 1,325 blocks
==14088==   total heap usage: 5,106 allocs, 3,781 frees, 1,491,021 bytes allocated
==14088== 
==14088== LEAK SUMMARY:
==14088==    definitely lost: 0 bytes in 0 blocks
==14088==    indirectly lost: 0 bytes in 0 blocks
==14088==      possibly lost: 8,091 bytes in 25 blocks
==14088==    still reachable: 305,800 bytes in 1,300 blocks
==14088==         suppressed: 0 bytes in 0 blocks
==14088== Rerun with --leak-check=full to see details of leaked memory
==14088== 
==14088== For lists of detected and suppressed errors, rerun with: -s
==14088== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
```

bryan-hunt, 2020-03-15:
You’re getting the segfault because you are building without a USB interface but telling the library to use usb when invoke it. You must have it built to include the HID HAL. Since you’re are trying to invoke this from a container have you verified that the image also contains the USB libraries as well as the headers for them?

flongo82, 2020-03-15:
What should I check exactly? What do you mean by USB libraries and headers? 

By the way, I'm not sure that running inside a container could be the problem because if I give the following: 

From the host machine:
```
lxc launch ubuntu:19.10 cryptochip
lxc config device add cryptochip starterkit usb vendorid=03eb productid=2312
lxc exec cryptochip -- sudo --login --user ubuntu
```

From inside container:
```
git clone https://github.com/MicrochipTech/cryptoauthtools.git
sudo apt install python-pip cmake libusb-1.0-0-dev libhidapi-dev
sudo pip install cryptoauthlib==20190517 --no-cache-dir
```

Then, I manage to run the info.py example in cryptoauthtools from inside the container and I get the following: 

```
Device Part:
    ATECC508A

Serial number: 
    01 23 B2 61 9B 97 B6 2F EE

Configuration Zone:
    01 23 B2 61 00 00 50 00 9B 97 B6 2F EE C0 41 00
    C0 00 55 00 83 20 87 20 8F 20 C4 8F 8F 8F 8F 8F
    9F 8F AF 8F 00 00 00 00 00 00 00 00 00 00 00 00
    00 00 AF 8F FF FF FF FF 00 00 00 00 FF FF FF FF
    00 00 00 00 FF FF FF FF FF FF FF FF FF FF FF FF
    FF FF FF FF 00 00 55 55 FF FF 00 00 00 00 00 00
    33 00 33 00 33 00 1C 00 1C 00 1C 00 1C 00 1C 00
    3C 00 3C 00 3C 00 3C 00 3C 00 3C 00 3C 00 1C 00

Check Device Locks
    Config Zone is unlocked
    Data Zone is unlocked

Done
```

which seems working,  i.e. the USB interface is seen inside the container. 

Unfortunately, if I'm not wrong cryptoauthlib version 20190517 does not have the PKCS11 library.

flongo82, 2020-03-16:
Just to be 100% sure, I tested everything on the host machine (so not inside a container). It always gives Segmentation fault.

bryan-hunt, 2020-03-16:
Did you compile on your host machine with `-DATCA_HAL_KIT_HID=on` ?

flongo82, 2020-03-16:
Yep. 

I run the following script:

```
sudo apt install -y autoconf libtool-bin pkg-config libssl-dev
git clone https://github.com/OpenSC/libp11.git
cd libp11/
./bootstrap && ./configure && make && sudo make install
cd ..
sudo apt-get install -y cmake libudev-dev
git clone --single-branch -b pkcs11 https://github.com/MicrochipTech/cryptoauthlib
cd cryptoauthlib
cmake -DATCA_HAL_KIT_HID=on . && make && sudo make install
cd /var/lib/cryptoauthlib
cp slot.conf.tmpl 0.conf
sed -i 's/i2c,0xB0/hid/g' 0.conf
sudo apt-get install -y p11-kit
sudo mkdir /etc/pkcs11
cat << EOF | sudo tee /etc/pkcs11/pkcs11.conf
# This setting controls whether to load user configuration from the
# ~/.config/pkcs11 directory. Possible values:
# none: No user configuration
# merge: Merge the user config over the system configuration (default)
# only: Only user configuration, ignore system configuration
user-config: merge
EOF
cat << EOF | sudo tee /usr/share/p11-kit/modules/cryptoauthlib.module
module: /usr/lib/libcryptoauth.so
critical: yes
trust-policy: yes
managed: yes
log-calls: no
EOF
sudo apt-get install -y gnutls-bin
sudo p11tool --list-tokens --debug=999
```

and obtained the following output (you do not have the complete apt log here because packages where already installed but you can see versions): 

```
Reading package lists...
Building dependency tree...
Reading state information...
autoconf is already the newest version (2.69-11).
libtool-bin is already the newest version (2.4.6-10).
pkg-config is already the newest version (0.29.1-0ubuntu2).
libssl-dev is already the newest version (1.1.1b-1ubuntu2.4).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Cloning into 'libp11'...
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force -I m4
autoreconf: configure.ac: tracing
autoreconf: running: libtoolize --copy --force
libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, '.'.
libtoolize: copying file './ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: copying file 'm4/libtool.m4'
libtoolize: copying file 'm4/ltoptions.m4'
libtoolize: copying file 'm4/ltsugar.m4'
libtoolize: copying file 'm4/ltversion.m4'
libtoolize: copying file 'm4/lt~obsolete.m4'
libtoolize: Remember to add 'LT_INIT' to configure.ac.
autoreconf: running: /usr/bin/autoconf --force
autoreconf: running: /usr/bin/autoheader --force
autoreconf: running: automake --add-missing --copy --force-missing
configure.ac:27: installing './compile'
configure.ac:26: installing './config.guess'
configure.ac:26: installing './config.sub'
[... 450 lines cut ...]
-- Installing: /usr/lib/libcryptoauth.so
Reading package lists...
Building dependency tree...
Reading state information...
p11-kit is already the newest version (0.23.15-2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
mkdir: cannot create directory ‘/etc/pkcs11’: File exists
# This setting controls whether to load user configuration from the
# ~/.config/pkcs11 directory. Possible values:
# none: No user configuration
# merge: Merge the user config over the system configuration (default)
# only: Only user configuration, ignore system configuration
user-config: merge
module: /usr/lib/libcryptoauth.so
critical: yes
trust-policy: yes
managed: yes
log-calls: no
Reading package lists...
Building dependency tree...
Reading state information...
gnutls-bin is already the newest version (3.6.5-2ubuntu1.1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
./cryptochip.sh: line 31:  6062 Segmentation fault      sudo p11tool --list-tokens --debug=999
```

Medivh0416, 2020-04-07:
Hi @flongo82, did you get it resloved?

I am getting same error as you.

flongo82, 2020-04-07:
Hi, no the situation is the same on my side. This is really annoying.

Medivh0416, 2020-04-07:
Have you tried update this file https://github.com/MicrochipTech/cryptoauthlib/blob/pkcs11/lib/CMakeLists.txt#L5, also update the udev rule https://github.com/MicrochipTech/cryptoauthlib/wiki/Udev-Rules?

I am able to get crash away by updating this.

flongo82, 2020-04-08:
Did you modified the line as below: 

`
option(ATCA_HAL_KIT_HID "Include the HID HAL Driver", ON)
`

with an ON? If I'm not wrong, I already tried as also suggested by @bryan-hunt. 

I'm already running everything with sudo if not as root. By the way, I will give it a try. 

Thanks

flongo82, 2020-04-08:
I tried again and I can confirm that I did try that already when suggested by @bryan-hunt and that it does not solve the issue. 

Either there are some differences in terms (for example) of installed dependencies between my system and your system or maybe you are not using the pcks11 branch. 

Can you share the version of the relevant dependencies? You can find mine above in the earlier posts.

Medivh0416, 2020-04-08:
From my config, I set 

```
# Various Options for Build
option(ATCA_HAL_KIT_HID "Include the HID HAL Driver" ON)
option(ATCA_HAL_KIT_CDC "Include the CDC HAL Driver (Legacy)" OFF)
option(ATCA_HAL_I2C "Include the I2C Hal Driver - Linux & MCU only" OFF)
option(ATCA_HAL_CUSTOM "Include support for Custom/Plug-in Hal Driver" OFF)
option(ATCA_PRINTF "Enable Debug print statements in library" ON)
option(ATCA_PKCS11 "Build PKCS11 Library" ON)
option(ATCA_MBEDTLS "Integrate with mbedtls")
option(ATCA_BUILD_SHARED_LIBS "Build CryptoAuthLib as shared library" ON)
option(PKCS11_DEBUG_ENABLE "Enable PKCS11 Debug Output" ON)
```

I ran into a hang when run the command

ChrisDSwiss, 2020-06-01:
Hi. I ran into the same issue. Updated to the newest cryptoauthlib pkcs11 branch from a0007d2f to 2df0eb145. Before calling openSSL via curl, I get a segementation fault under certain circumstances.

flongo82, 2020-07-17:
After the merge of the pkcs11 branch into the master, everything seems to be working fine also with the HID interface. I'm closing the issue.

closed 2020-07-17 by flongo82