MicrochipTech/cryptoauthlib #148: calling atcab_read_zone consecutively fails arbitrarily

ramonaerne, 2020-04-15:
I am running an application which performs consecutive `atcab_read_zone` as well as `atcab_write_zone` calls to read out essentially most of the non secured address space.

However, running the application leads to individual reads (few percent of the total reads) to fail arbitrarily (running the application mutliple times causes different slots / blocks to be affected) with errors such as
```
    ATCA_PARSE_ERROR            = 0xD2, //!< response status byte indicates parsing error (status byte = 0x03)
    ATCA_RX_CRC_ERROR           = 0xE5, //!< CRC error in data received from device
    ATCA_RX_NO_RESPONSE         = 0xE7, //!< Not an error while the Command layer is polling for a command response.
    ATCA_RX_TIMEOUT             = 0xEB, //!< for Microchip PHY protocol, timeout on receipt waiting for master
    ATCA_COMM_FAIL              = 0xF0, //!< Communication with device failed. Same as in hardware dependent modules.
    ATCA_WAKE_SUCCESS           = 0xF3, //!< received proper wake token
```

In my opinion I2C (i am running the i2c linux userspace hal) should not be the issue since all other commands including all writes never fail, it is only the reads. This leads me to the following questions:
1. Could timing or buffer issues that only apply to reads explain this behavour
2. Is there any sequence of commands that guarantee a successful read (such as resetting the address counter, putting the device to sleep instead of idle, ...)
3. Any other configuration that might lead to this behavour

raerne, 2020-04-27:
Turns out this issue is essentialy the same as #149. We had another process running in the background using the cryptoki interface. Concurrent access must be handled appropriately e.g. using the locking mechanism of the pkcs11 implementation.

bryan-hunt, 2020-04-27:
That makes sense. For the moment you have the option of calling the pkcs11 locking functions manually from your application or only utilize the pkcs11 API. We are looking at bring the process safety features to a lower level in the library so it could be built to require all functions to be process safe. That however will not be available until the second half of this year.

closed 2020-06-12 by bryan-hunt