MicrochipTech/cryptoauthlib #149: Concurrent use of PKCS#11 corrupts communication

r-funke, 2020-04-22:
By following the hint in [issue #133](https://github.com/MicrochipTech/cryptoauthlib/issues/133#issuecomment-580425628) and putting cert templates and static config in **pkcs11_config.c** I was able to get reading and writing of certificates through PKCS#11 running, which is pretty cool.

In order to find out how well PKCS#11 lib is handling concurrent use by two separate processes, I did a load test invoking p11tool in an endless loop in one terminal and then invoking another run of p11tool in another terminal. So like this:
* terminal 1: `while true; do p11tool --provider /usr/lib/libcryptoauth.so --export "pkcs11:token=0123EE;object=device;type=cert"; done`
* terminal 2: `p11tool --provider /usr/lib/libcryptoauth.so --export "pkcs11:token=0123EE;object=signer;type=cert"`

The Atecc608a is attached to i2c bus.

## Observed behavior:

As long as only the loop in terminal 1 is running, all invokations succeed. When invoking the command in terminal 2, that one fails. But at the same time, one of the invokations in terminal 1 also fails. That means that one PKCS#11 instance was having an open session, when another PKCS#11 instance was interfering i2c communication by trying to open another session. As a result both sessions fail.

## Expected behavior:

A process that has a running PKCS#11 session should not be interfered by another process starting a PKCS#11 session. There should be some kind of locking in place to prevent that.

Ideally, concurrent use should be possible, i.e. two processes can use PKCS#11 concurrently in a way that one process can start a transaction once another process has finished its transaction.

We're launching a product that has a Linux board and an Atecc attached to it via i2c. Our use case is that we would like to run two separate applications in parallel, each one having its own set of keypair/cert in the Atecc. Do you think that the libcryptoauth could support that?

bryan-hunt, 2020-04-22:
This is a complex answer so I will attempt to point you in the right direction to see why you may be having issues.

a) The PKCS11 specification details the behavior a library must exhibit in many areas - including locking: [PKCS#11: C_Initialize](http://docs.oasis-open.org/pkcs11/pkcs11-base/v2.40/os/pkcs11-base-v2.40-os.html#_Toc416959740)

b) The PKCS11 implementation does its best to provide a robust locking system - if the application follows the PKCS11 recomendation and does not provide its own locking/unlocking functions during initialization. I have not looked at p11tool to see how they call C_Initialize and if they are providing their own functions

c) Our PKCS11 implementation defaults to using a mutex in shared memory (ATCA_USE_SHARED_MUTEX should be defined - it is by default in the CMake file) in order to ensure PROCESS safety rather than merely THREAD safety. If you wish to override the default PKCS11 compliant behavior and force it to use the internal implementation: see [pkcs11_init L178-L207](https://github.com/MicrochipTech/cryptoauthlib/blob/pkcs11/lib/pkcs11/pkcs11_init.c#L178..L207)

r-funke, 2020-04-23:
Thanks @bryan-hunt . It is exactly as you suspected. The p11tool provides its own locking callbacks and those ones probably don't provide any protection on process level. 

I repeated the load test, but this time with a tool that does not provide its own locking functions. This time I didn't see any issues. So I'm closing this issue.

closed 2020-04-23 by r-funke

GalaxyGorilla, 2021-03-17:
@bryan-hunt this is an old issue, but I have a follow up question out of curiosity: for what exactly is the locking mechanism here required? Is it because single API calls end up in a sequence of multiple I2C transactions which then get mixed up between processes?