MicrochipTech/cryptoauthlib #159: Improve encapsulation (reduce exposed symbols)

5frank, 2020-05-25:
Currently, including `cryptoauthlib.h` exposes a lot of macros and functions that I believe is not supposed to be part of the API. This has the following problems:
- Makes it hard to see intended usage.
- Short names without "namespace prefix" such as `PRINT` can cause name collision.
- If they are used outside the library, they are suddenly part of the API unintentionally and can not be changed without breaking things for others.

Examples:

lib/cryptoauthlib.h 

    #define BREAK(status, message) { break; }
    #define RETURN(status, message) { return status; }
    #define PRINT(message) { break; }
    #define DBGOUT(message) { break; }

lib/basic/atca_helpers.h (included by cryptoauthlib.h but only used from atca_helpers.c, can be static or possible removed)

    ATCA_STATUS packHex(const char* ascii_hex, size_t ascii_hex_len, char* packed_hex, size_t* packed_len);
    bool isDigit(char c);
    bool isWhiteSpace(char c);
    bool isAlpha(char c);
    bool isHexAlpha(char c);
    bool isHex(char c);
    bool isHexDigit(char c);
    bool isBase64(char c, const uint8_t * rules);

there is probably more.

bryan-hunt, 2020-05-25:
Actually there really isn't much that is inappropriately scoped. Generally when an API is released whatever decisions that were made will have to remain. The majority of the identified items are already resolved in the next major release

5frank, 2020-06-04:
Great to here this is addressed in next release! :)
Would also be nice if you considered removing some includes from `cryptoauthlib.h` or adding prefixes to all included symbols.  Example `atca_hal.h` - IMO it do not belong in that scope and `hal_i2c_init` causes conflicts in Mynewt (probably other places as well). It requires ugly hacks such as this:
```C
#define hal_i2c_init ATCA_hal_i2c_init 
#include "cryptoauthlib.h"
ATCA_STATUS hal_i2c_init(void *hal, ATCAIfaceCfg *cfg)
{
    // do stuff
    return ATCA_SUCCESS; 
}
#undef hal_i2c_init
```

bryan-hunt, 2020-06-15:
With the addition of methods to connect HALs to the library through API calls it is no longer necessary to use the same function names for HALs as the library originally required.

closed 2020-06-15 by bryan-hunt