MicrochipTech/cryptoauthlib #161: p11-kit: cryptoauthlib: module failed to initialize: An error occurred on the device

AlessioC31, 2020-06-09:
Hi everyone,
I'm on a Rpi 3 B and I'm trying to use a ATECC508a as a pkc11 slot. I can't manage to get it to a working state because when I execute `sudo p11tool --list-all` I get:
```
C_Initialize
IN: pInitArgs = NULL
C_Initialize = CKR_DEVICE_ERROR
p11-kit: cryptoauthlib: module failed to initialize: An error occurred on the device
pkcs11_init: PKCS #11 initialization error.
warning: no token URL was provided for this operation; the available tokens are:
```

That's my `lib/atca_config.h`:

```
/**
 * \file
 * \brief Cryptoauthlib Configuration Defines
 *
 * \copyright (c) 2015-2018 Microchip Technology Inc. and its subsidiaries.
 *
 * \page License
 *
 * Subject to your compliance with these terms, you may use Microchip software
 * and any derivatives exclusively with Microchip products. It is your
 * responsibility to comply with third party license terms applicable to your
 * use of third party software (including open source software) that may
 * accompany Microchip software.
 *
 * THIS SOFTWARE IS SUPPLIED BY MICROCHIP "AS IS". NO WARRANTIES, WHETHER
 * EXPRESS, IMPLIED OR STATUTORY, APPLY TO THIS SOFTWARE, INCLUDING ANY IMPLIED
 * WARRANTIES OF NON-INFRINGEMENT, MERCHANTABILITY, AND FITNESS FOR A
 * PARTICULAR PURPOSE. IN NO EVENT WILL MICROCHIP BE LIABLE FOR ANY INDIRECT,
 * SPECIAL, PUNITIVE, INCIDENTAL OR CONSEQUENTIAL LOSS, DAMAGE, COST OR EXPENSE
 * OF ANY KIND WHATSOEVER RELATED TO THE SOFTWARE, HOWEVER CAUSED, EVEN IF
 * MICROCHIP HAS BEEN ADVISED OF THE POSSIBILITY OR THE DAMAGES ARE
 * FORESEEABLE. TO THE FULLEST EXTENT ALLOWED BY LAW, MICROCHIP'S TOTAL
 * LIABILITY ON ALL CLAIMS IN ANY WAY RELATED TO THIS SOFTWARE WILL NOT EXCEED
 * THE AMOUNT OF FEES, IF ANY, THAT YOU HAVE PAID DIRECTLY TO MICROCHIP FOR
 * THIS SOFTWARE.
 */

#ifndef _ATCA_CONFIG_H
#define _ATCA_CONFIG_H

/** Use I2C */
#define ATCA_HAL_I2C

/** Use the following address for ECC devices */
#define ATCA_I2C_ECC_ADDRESS    0xB0

/** Define if cryptoauthlib is to use the maximum execution time method */
#define ATCA_NO_POLL

#define ATCA_ATECC508A_SUPPORT
/** Use RTOS timers (i.e. delays that yield) */
//#define ATCA_USE_RTOS_TIMER

#endif
```

`lib/pkcs11/pkcs11_config.h`:

```
/**
 * \file
 * \brief PKCS11 Library Configuration
 *
 * Copyright (c) 2017 Microchip Technology Inc. All rights reserved.
 *
 * \atmel_crypto_device_library_license_start
[... 170 lines cut ...]

# Slot 12 is the intermedate/signer certificate data
#object = certificate,signer,12

# Slot 15 is a public key
object = public,root,15
```

And lastly that's the output of `sudo i2cdetect -y 1`:

```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- 58 -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```

Can you please help me figuring out what I'm doing wrong? 

Thanks.

bryan-hunt, 2020-06-09:
You need to follow the wiki steps for either setting up p11-kit or provide the provider library to p11-tool in your commands

Remember that sudo changes the environment of execution as well. So that means if you set up p11-kit you will have to set it up properly to work in both. If you're using i2c you shouldn't have to be using sudo though.

AlessioC31, 2020-06-10:
I already set up p11-kit in order to use the cryptoauthlib module, I forgot to mention it, sorry.

`/usr/share/p11-kit/modules/cryptoauthlib.module`:

```
module: /usr/lib/libcryptoauth.so
critical: yes
trust-policy: yes
managed: yes
log-calls: yes
```

I executed p11tool without sudo before but the output doesn't change.
I also tried to execute p11tool providing the provider:
```
$ p11tool --provider=/usr/lib/libcryptoauth.so
p11-kit: (unknown): module failed to initialize: An error occurred on the device
pkcs11_add_provider: PKCS #11 error in device
```

Am I missing something more? Can you please link me the wiki you're referring to? So I can be sure I'm following the right thing, thanks

shearl, 2020-06-12:
Hi All,

I am trying to do almost the same thing as @AlessioC31. I am using a Rpi 3 B+ and I'm trying to use a ATECC608a connected via i2c as a pkc11 slot. I have followed the [wiki](https://github.com/MicrochipTech/cryptoauthlib/wiki/PKCS11-Linux-Setup) steps to set up `p11tool` and the suggestion here by @bryan-hunt to append ",1" for the i2c bus number in `/var/lib/cryptoauthlib/0.conf`.

When I execute the `--list-all` command I get the following output (regardless of what I type for the PIN).

```
$ p11tool --provider /usr/lib/libcryptoauth.so --list-all
Object 0:
	URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=62A6D7FF625F;token=0123EE;object=device;type=private
Token '0123EE' with URL 'pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=62A6D7FF625F;token=0123EE' requires user PIN
Enter PIN: 
	Type: Private key (EC/ECDSA)
	Label: device
	Flags: CKA_PRIVATE; CKA_SENSITIVE; 
	ID: 
$ 
```

When I try to execute the command in the [wiki](https://github.com/MicrochipTech/cryptoauthlib/wiki/PKCS11-Linux-Setup) to _Get the public key for a private key_ I get the error shown in the output below (again, regardless of what I type for the PIN).

```
pi@raspberrypi:~/projects/Microchip/cryptoauthlib $ p11tool --provider /usr/lib/libcryptoauth.so --export-pubkey "pkcs11:token=0123EE;object=device;type=private"
note: assuming --login for this operation.
warning: no --outfile was specified and the public key will be printed on screen.
Token '0123EE' with URL 'pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=62A6D7FF625F;token=0123EE' requires user PIN
Enter PIN: 
Error in pkcs11_export_pubkey:1391: The requested data were not available.
```
Below is the contents of my `lib/atca_config.h`, modified to follow the model of what @AlessioC31 had in his `lib/atca_config.h`. My modifications are as follows: I uncommented `#define ATCA_HAL_I2C`, changed the i2c address per the output of `i2cdetect` and added `#define ATCA_ATECC608A_SUPPORT`. (Note that uncommenting `#define ATCA_HAL_I2C` results in a lot of `warning: "ATCA_HAL_I2C" redefined` messages during compilation.)

```
#ifndef _ATCA_CONFIG_H
#define _ATCA_CONFIG_H

/** Use I2C */
#define ATCA_HAL_I2C

/** Use the following address for ECC devices */
#define ATCA_I2C_ECC_ADDRESS    0xC0

/** Define if cryptoauthlib is to use the maximum execution time method */
#define ATCA_NO_POLL

/** Use RTOS timers (i.e. delays that yield) */
//#define ATCA_USE_RTOS_TIMER

#define ATCA_ATECC608A_SUPPORT
```

Below are the contents of my `/var/lib/cryptoauthlib/0.conf`. The only thing I changed here is the i2c address and appended ",1" for the bus number per the suggestion from @bryan-hunt.

```
# Reserved Configuration for a device
# The objects in this file will be created and marked as undeletable
# These are processed in order. Configuration parameters must be comma
# delimited and may not contain spaces

interface = i2c,0xC0,1
freeslots = 1,2,3

# Slot 0 is the primary private key
object = private,device,0

# Slot 10 is the certificate data for the device's public key
#object = certificate,device,10

# Slot 12 is the intermedate/signer certificate data
#object = certificate,signer,12

# Slot 15 is a public key
object = public,root,15
```

Below is the output of `sudo i2cdetect -y 1`. I believe that the device address of 0x60 is 0xC0.

```
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: 60 -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --    
```

My guess is that the ATECC608A does not come pre-provisioned with the default keys and certificates as found in `/var/lib/cryptoauthlib/slot.conf.tmpl`? Is there a way under Linux to provision the ATECC608A? Also, is there a default PIN?

shearl, 2020-06-13:
@bryan-hunt My apologies for not digging deeper into the other issues to find the problem. I see from issues #105 and #95 that I missed the initialisation step. (It seems like it might be a good idea to explicitly add this to the [PKCS11 Linux Setup](https://github.com/MicrochipTech/cryptoauthlib/wiki/PKCS11-Linux-Setup) wiki page).

I get the same error mentioned in #105 and #95, but it does seem to actually initialise the token and I am able to export the public key.

```
$ p11tool --provider /usr/lib/libcryptoauth.so --initialize "pkcs11:serial=62A6D7FF625F" --label test
Enter Security Officer's PIN: 
Initializing token... 
Error in pkcs11_init:1439: PKCS #11 error.
```

Note: I'm surprised that I get the error because @raerne said [here](https://github.com/MicrochipTech/cryptoauthlib/issues/95#issuecomment-543718994) that changing `ATCA_I2C_ECC_ADDRESS` to `0xC0` in `lib/atca_config.h` fixed the problem. I have it set that way - see [above](https://github.com/MicrochipTech/cryptoauthlib/issues/161#issuecomment-643510869) - but I still get the error.

raerne, 2020-06-15:
I don't have an unprovisioned device available to reproduce. If I recall correctly with "fixed the problem" i meant that the function `pkcs11_token_init` succeeded which meant that the device itself was initialized. However, the p11tool cli call itself did not return with success but I was able to use all other commands like `p11tool ... --list-all` successfully.

shearl, 2020-06-19:
Hi @raerne, thanks for the message. Unfortunately it seems like my ATECC608 is still not working with the pkcs11 tools.

I started over with a brand new unprovisioned ATECC608 and did the following:

```
$ p11tool --initialize --label greengrass "pkcs11:serial=C2D1F027997E"
Enter Security Officer's PIN: 
Initializing token... 
Error in pkcs11_init:1439: PKCS #11 error.
```

Based on your comments and what @bryan-hunt has said, this shouldn't be a problem. However, when I ask for the list of tokens I get the following:

```
$ p11tool --list-tokens
Token 0:
	URL: pkcs11:model=p11-kit-trust;manufacturer=PKCS%2311%20Kit;serial=1;token=System%20Trust
	Label: System Trust
	Type: Trust module
	Flags: uPIN uninitialized
	Manufacturer: PKCS#11 Kit
	Model: p11-kit-trust
	Serial: 1
	Module: p11-kit-trust.so

Token 1:
	URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=C2D1F027997E;token=0123EE
	Label: 0123EE
	Type: Hardware token, Trust module
	Flags: RNG, Requires login, Uninitialized, uPIN uninitialized
	Manufacturer: Microchip Technology Inc
	Model: ATECC608A
	Serial: C2D1F027997E
	Module: /usr/lib/libcryptoauth.so
```

Note the this line `Flags: RNG, Requires login, Uninitialized, uPIN uninitialized` which seems to indicate that the token is still uninitialized.

When I ask for a list of all objects on the token it indeed shows two objects:

```
$ p11tool --list-all "pkcs11:serial=C2D1F027997E"
Object 0:
	URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=C2D1F027997E;token=0123EE;object=device;type=private
Token '0123EE' with URL 'pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=C2D1F027997E;token=0123EE' requires user PIN
Enter PIN: 
	Type: Private key (EC/ECDSA-SECP256R1)
	Label: device
	Flags: CKA_PRIVATE; CKA_SENSITIVE; 
	ID: 

Object 1:
	URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=C2D1F027997E;token=0123EE;object=device;type=public
	Type: Public key (EC/ECDSA-SECP256R1)
	Label: device
	ID: 
```

And it seems like I can successfully export the public key:

```
$ p11tool --export-pubkey "pkcs11:token=0123EE;object=device;type=private"
note: assuming --login for this operation.
warning: no --outfile was specified and the public key will be printed on screen.
Token '0123EE' with URL 'pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=C2D1F027997E;token=0123EE' requires user PIN
Enter PIN: 
-----BEGIN PUBLIC KEY-----
MFkwEwYHKoZIzj0CAQYIKoZIzj0DAQcDQgAE/jkBuiLcmZWeyI6T2l03oSjLsWBY
8TiI0U7pJc3wKGxGJCNiDzz9EwngosTQRTEtk2HQOpAsAa0Ir1KVJa2ASg==
-----END PUBLIC KEY-----
```

However, when I try to use the pkcs11 engine with OpenSSL to create a certificate signing request, I get more error messages about the token being uninitialized:

```
$ OPENSSL_CONF=myopenssl.conf openssl req -engine pkcs11 -key "pkcs11:token=0123EE;object=device;type=private" -keyform engine -new -out new_device.csr -subj "/CN=NEW CSR EXAMPLE"
engine "pkcs11" set.
Found uninitialized token
Specified object not found
Found uninitialized token
Specified object not found
PKCS11_get_private_key returned NULL
cannot load Private Key from engine
1995735056:error:80067065:pkcs11 engine:ctx_load_privkey:object not found:eng_back.c:974:
1995735056:error:26096080:engine routines:ENGINE_load_private_key:failed loading private key:../crypto/engine/eng_pkey.c:78:
unable to load Private Key
```

For reference, here are the relevant configuration files:
```
$ cat /etc/cryptoauthlib/cryptoauthlib.conf
# Cryptoauthlib Configuration File

filestore = /var/lib/cryptoauthlib

$ cat /var/lib/cryptoauthlib/0.conf 
# Reserved Configuration for a device
# The objects in this file will be created and marked as undeletable
# These are processed in order. Configuration parameters must be comma
# delimited and may not contain spaces

interface = i2c,0xC0,1
freeslots = 1,2,3

# Slot 0 is the primary private key
object = private,device,0

# Slot 10 is the certificate data for the device's public key
#object = certificate,device,10

# Slot 12 is the intermedate/signer certificate data
#object = certificate,signer,12

# Slot 15 is a public key
object = public,root,15

$ cat /etc/pkcs11/pkcs11.conf
# This setting controls whether to load user configuration from the
# ~/.config/pkcs11 directory. Possible values:
#    none: No user configuration
#    merge: Merge the user config over the system configuration (default)
#    only: Only user configuration, ignore system configuration
user-config: merge

$ cat ~/.config/pkcs11/modules/cryptoauthlib.module
module: /usr/lib/libcryptoauth.so
critical: yes
trust-policy: yes
managed: yes
log-calls: no

$ cat myopenssl.conf 
openssl_conf = openssl_init

[openssl_init]
engines=engine_section

[engine_section]
pkcs11 = pkcs11_section

[pkcs11_section]
engine_id = pkcs11
dynamic_path = /usr/lib/arm-linux-gnueabihf/engines-1.1/libpkcs11.so
MODULE_PATH = /usr/lib/libcryptoauth.so
init = 0
```

ander-galis-ikerlan, 2020-07-14:
I'm having the exact same issue, I'm able to export the key with p11tool but for OpenSSL the device is not initialized, where you able to solve it?

shearl, 2020-07-14:
Still not solved. I have an issue into Microchip tech support and they said
they are trying to recreate the issue.

ander-galis-ikerlan, 2020-07-15:
It looks that after the --initialize step the configuration memory is lock but not the data one. And we can't interact with it until it is closed. In order to do it what I did was to use cryptoauthtools, more specifically the config.py to close this memory. It clearly is a workaround but it worked for me and I was able to generate the CSR

vishalSpintly, 2022-01-27:
> Still not solved. I have an issue into Microchip tech support and they said they are trying to recreate the issue.
> […](#)
> On Tue, Jul 14, 2020 at 01:33 ander-galis-ikerlan ***@***.***> wrote: I'm having the exact same issue, I'm able to export the key with p11tool but for OpenSSL the device is not initialized, where you able to solve it?

Hello Shearl,

I am trying the same with a Rpi4 board with the ATECC608A connected over I2C lines as well.
I find myself in the same situation as you have reported here. I have tried figuring out the problem but I do not find any resources to resolve the issue.
see:

Can you let me know know if you resolved this problem and if yes, how.
Look forward to your response.

regards,
Vishal

dexterac, 2022-03-09:
Is this resolved?

bryan-hunt, 2022-03-09:
There is nothing to resolve. It's just improper configuration.

Start with the [Trust Platform Development Suite](https://www.microchip.com/en-us/products/security-ics/trust-platform#) and configure a TrustFlex part. 

Otherwise one needs to read the datasheet and understand the part before making changes to the part configuration.

vishalSpintly, 2022-03-09:
@bryan-hunt  Extremely disappointed  with the response.
You are not making any attempt at helping understand what the problem could be even on executing simple applications provided by Microchip themselves. This is unfortunate. 

On contacting the Microchip sales team I hear that the ATECC608A Chip has hardware issues and is not recommended for new designs.

bryan-hunt, 2022-03-09:
Like all semiconductor devices there are errata that are discovered. Given the worldwide semiconductor manufacturing situation it is recommended to use the latest revision of the part. The ATECC608B trustflex parts are available: https://www.microchipdirect.com/product/ATECC608B-TFLXTLSS-PROTO so that doesn't change the overall recommendation to use the trust platform rather than blank devices. These are complex devices that require attention to detail when performing configuration.

This is why the Trust Platform is the best way to get started with using these parts - the Trust Platform Development Suite comes with examples, and configuration tools.

When it comes to pkcs11 these instructions have been run numerous times by many people. The pkcs11 interface is used by hundreds of thousands of devices. Everything I have seen on this thread indicates the instructions having not been followed or a hardware issue for which I can't diagnose through this medium.

vishalSpintly, 2022-03-09:
>Everything I have seen on this thread indicates the instructions having not been followed 
If that is the case why haven't you pointed out the "instructions that have not been followed".
Everything that was done has been in compliance with the listed instructions.
It rather looks like you are not interested in helping solve customer reported issues.

dexterac, 2022-03-10:
Hi @bryan-hunt 

1. Worldwide semiconductor situation : This is rather recent few yrs issue. Chips and boards from 2019 and 2020 still in circulation and most of these are ATECC608A. Each issue reported here is about these. While we do understand new chips (ATECC608B) is recommended, it still doesn't address the issues with what we have and is readily available. 

2. Documentation is notoriously misleading and is scattered between github wiki, datasheet (not easily available unless NDA signed), microchip learning portal, community blogs and recommendations from community.

3. Github wiki created and maintained by you has many broken links.

4. A slightest error and the board/chip is locked and almost useless. Each board even with discounted price costs us $1-$2. 

5. @shearl, @vishalSpintly and I have faced same issue and we are asking a very discretely asking if there is a solution to the issue we are facing with the what is currently available. Suggesting to purchase new devices is not going to help. 

If you can, then share the solution and update the github wiki. We all would be appreciate a better documentation. Also, let someone else in the community respond to this issue when they find a right way to approach.

Thanks for your help.

ronnytittoto, 2022-03-10:
Microchip has a team of ESEs that can support customers in the field; additionally MCHP official support team, available under www.microchip.com/support, can support customers working on designs including this device. I would recommend getting in touch with a local ESE.
Hope this helps

amolchor25, 2022-12-08:
Hi all,

I have done the configuration as per the https://github.com/MicrochipTech/cryptoauthlib/wiki/PKCS11-Linux-Setup
but facing the issue as follows
root@am335x-evm:~ **p11tool --provider=/usr/lib/libcryptoauth.so --list-all**
**Object 0:
        URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=237E39B503D635EE;token=00ABC;object=device;type=private
        Type: Private key (EC/ECDSA)
        Label: device
        Flags: CKA_PRIVATE; CKA_SENSITIVE; 
        ID:** 

root@am335x-evm:~ **p11tool --provider=/usr/lib/libcryptoauth.so --initialize "pkcs11:serial=237E39B503D635EE" --label test
Enter Security Officer's PIN: 
Initializing token... done**

**Token was successfully initialized; use --initialize-pin and --initialize-so-pin to set or reset PINs**
root@am335x-evm:~ **p11tool --provider=/usr/lib/libcryptoauth.so --export-pubkey "pkcs11:token=00ABC;object=device;type=private"**
**warning: --login was not specified and it may be required for this operation.
warning: no --outfile was specified and the public key will be printed on screen.
Error in pkcs11_export_pubkey:1407: The requested data were not available.**
root@am335x-evm:~# **openssl req -engine pkcs11 -key "pkcs11:token=00ABC;object=device;type=private" -keyform engine -new -out new_device.csr -subj "/CN=NEW CSR EXAMPLE"**
**engine "pkcs11" set.
Segmentation fault (core dumped)**

is there any missing configuration then let me know?
I have tried figuring out the problem but I do not find any resources to resolve the issue.