MicrochipTech/cryptoauthlib #162: OpenVPN example

esguti, 2020-06-12:
I am trying to start a secure connection between a device with ATECC608A and an OpenVPN server.

I would like to store and use my signed certificate in the ATECC608A. I have managed to generate priv/pub keys with **cryptoauthlib** and a certificate request (following the tutorial [OpenSSL-Reference](https://github.com/MicrochipTech/cryptoauthlib/wiki/OpenSSL-Reference)).  In the server part, I have signed the csr with the root CA, obtaining the device certificate (crt). 

Now I need to store the device certificate (crt) in the ATECC608A, but I can not find any example. 
Lately, following the suggestions from the issue #135 , I managed to store the certificate in slot 13, but I cannot find the way to use it with openvpn configuration. Also, when I try to get the pkcs11-ids, there is no one:

```bash
openvpn --show-pkcs11-ids /usr/lib/libcryptoauth.so

The following objects are available for use.
Each object shown below may be used as parameter to
--pkcs11-id option please remember to use single quote mark.
```

¿Could you help me to find a way to properly store a certificate and use it?

bryan-hunt, 2020-06-15:
The PKCS11 interface for the library is to allow access to the a provisioned part via from a system that is able to access objects via PKCS11. The first step is to get your provisioning correct and then use the PKCS11 interface to access those objects. This means the first step has to be one of these two options:

a) Use the trust platform development suite (likely the trust flex devices). See [Microchip Trust Platform - Trust Flex](https://www.microchip.com/design-centers/security-ics/trust-platform/trustflex)
b) Work with your FAE to develop your use cases and provisioning

If you use the Trust Flex devices per the above recommendation then you can specify the trust flex device in the slot configuration which will pull in the correct configuration data. If you elect to go through the second option you once your certificate definitions are created you will need to provide a static object configuration to the library [app/pkcs11/example_pkcs11_config.c](https://github.com/MicrochipTech/cryptoauthlib/blob/master/app/pkcs11/example_pkcs11_config.c) and [app/pkcs11/example_cert_chain.c](https://github.com/MicrochipTech/cryptoauthlib/blob/master/app/pkcs11/example_cert_chain.c)

esguti, 2020-06-16:
It is a pity that there is no easy way to store my certificates. I see **Trust Flex** good for production, but not for development when a system/app (like OpenVPN) is being tested.

Anyway, I have a Microchip [DT100104](http://ww1.microchip.com/downloads/en/DeviceDoc/ATECC608A-Trust-Development-Board-Users-Guide-DS50002922A.pdf) with several ATECC available. I have tried the ATECC608A-TNGTLS (Trust&GO), which is supposed to have a certificate inside, but when I try to list the certificates, there is no one:

```bash
p11tool --list-all-certs 'pkcs11:manufacturer=Microchip%20Technology%20Inc'

3643:3643:C_Initialize:144:
3643:3643:pkcs11_config_parse_object:329:Certificate type: certificate3643:3643:C_Initialize:145:CKR_OK(0)
3643:3643:C_GetInfo:162:
3643:3643:C_GetInfo:163:CKR_OK(0)
3643:3643:C_GetSlotList:187:
3643:3643:C_GetSlotList:188:CKR_OK(0)
3643:3643:C_GetTokenInfo:205:
3643:3643:pkcs11_token_get_info:372:Token Info: 1
3643:3643:pkcs11_token_get_info:405:Token Locked
3643:3643:C_GetTokenInfo:206:CKR_OK(0)
3643:3643:C_GetSlotInfo:196:
3643:3643:C_GetSlotInfo:197:CKR_OK(0)
3643:3643:C_OpenSession:274:
3643:3643:C_OpenSession:275:CKR_OK(0)
3643:3643:C_CloseSession:283:
3643:3643:C_CloseSession:284:CKR_OK(0)
3643:3643:C_GetSlotList:187:
3643:3643:C_GetSlotList:188:CKR_OK(0)
3643:3643:C_GetTokenInfo:205:
3643:3643:pkcs11_token_get_info:372:Token Info: 1
3643:3643:pkcs11_token_get_info:405:Token Locked
3643:3643:C_GetTokenInfo:206:CKR_OK(0)
3643:3643:C_GetSlotInfo:196:
3643:3643:C_GetSlotInfo:197:CKR_OK(0)
3643:3643:C_OpenSession:274:
3643:3643:C_OpenSession:275:CKR_OK(0)
3643:3643:C_FindObjectsInit:437:
3643:3643:CKA_CLASS(0):4:CKO_CERTIFICATE(1)
3643:3643:CKA_CERTIFICATE_TYPE(80):4:0x7eeb7c90:
3643:3643:C_FindObjectsInit:438:CKR_OK(0)
3643:3643:C_FindObjects:446:
3643:3643:C_FindObjects:447:CKR_OK(0)
3643:3643:C_FindObjectsFinal:455:
3643:3643:C_FindObjectsFinal:456:CKR_OK(0)
3643:3643:C_CloseSession:283:
3643:3643:C_CloseSession:284:CKR_OK(0)
No matching objects found
```
But the module is recognized:

```bash
p11tool --list-tokens
3867:3867:C_Initialize:144:
3867:3867:C_Initialize:145:CKR_OK(0)
3867:3867:C_GetInfo:162:
3867:3867:C_GetInfo:163:CKR_OK(0)
Token 0:
	URL: pkcs11:model=p11-kit-trust;manufacturer=PKCS%2311%20Kit;serial=1;token=System%20Trust
	Label: System Trust
	Type: Trust module
	Flags: uPIN uninitialized
	Manufacturer: PKCS#11 Kit
	Model: p11-kit-trust
	Serial: 1
	Module: p11-kit-trust.so

3867:3867:C_GetSlotList:187:
3867:3867:C_GetSlotList:188:CKR_OK(0)
3867:3867:C_GetTokenInfo:205:
3867:3867:pkcs11_token_get_info:372:Token Info: 1
3867:3867:pkcs11_token_get_info:405:Token Locked
3867:3867:C_GetTokenInfo:206:CKR_OK(0)
3867:3867:C_GetSlotInfo:196:
3867:3867:C_GetSlotInfo:197:CKR_OK(0)
3867:3867:C_OpenSession:274:
3867:3867:C_OpenSession:275:CKR_OK(0)
3867:3867:C_CloseSession:283:
3867:3867:C_CloseSession:284:CKR_OK(0)
Token 1:
	URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=900CED1D691F;token=012301
	Label: 012301
3867:3867:C_GetSlotList:187:
3867:3867:C_GetSlotList:188:CKR_OK(0)
3867:3867:C_GetTokenInfo:205:
3867:3867:pkcs11_token_get_info:372:Token Info: 1
3867:3867:pkcs11_token_get_info:405:Token Locked
3867:3867:C_GetTokenInfo:206:CKR_OK(0)
3867:3867:C_GetSlotInfo:196:
3867:3867:C_GetSlotInfo:197:CKR_OK(0)
3867:3867:C_OpenSession:274:
3867:3867:C_OpenSession:275:CKR_OK(0)
3867:3867:C_CloseSession:283:
3867:3867:C_CloseSession:284:CKR_OK(0)
	Type: Hardware token, Trust module
	Flags: RNG, Requires login, uPIN uninitialized
	Manufacturer: Microchip Technology Inc
	Model: ATECC608A
	Serial: 900CED1D691F
3867:3867:C_GetSlotList:187:
3867:3867:C_GetSlotList:188:CKR_OK(0)
3867:3867:C_GetTokenInfo:205:
3867:3867:pkcs11_token_get_info:372:Token Info: 1
3867:3867:pkcs11_token_get_info:405:Token Locked
3867:3867:C_GetTokenInfo:206:CKR_OK(0)
3867:3867:C_GetSlotInfo:196:
3867:3867:C_GetSlotInfo:197:CKR_OK(0)
3867:3867:C_OpenSession:274:
3867:3867:C_OpenSession:275:CKR_OK(0)
3867:3867:C_CloseSession:283:
3867:3867:C_CloseSession:284:CKR_OK(0)
	Module: /usr/lib/libcryptoauth.so

3867:3867:C_GetSlotList:187:
3867:3867:C_GetSlotList:188:CKR_OK(0)
3867:3867:C_GetTokenInfo:205:
3867:3867:pkcs11_token_get_info:372:Token Info: 1
3867:3867:pkcs11_token_get_info:405:Token Locked
3867:3867:C_GetTokenInfo:206:CKR_OK(0)
3867:3867:C_GetSlotInfo:196:
3867:3867:C_GetSlotInfo:197:CKR_OK(0)
3867:3867:C_OpenSession:274:
3867:3867:C_OpenSession:275:CKR_OK(0)
3867:3867:C_CloseSession:283:
3867:3867:C_CloseSession:284:CKR_OK(0)
3867:3867:C_Finalize:153:
3867:3867:C_Finalize:154:CKR_OK(0)
```

Is there any way to use certificates with p11tool?

bryan-hunt, 2020-06-16:
Change the slot configuration file to one of these (note remove any object definitions)

For Trust & Go TLS:
```
interface = hid,i2c,0x6A
device = ATECC608A-TNGTLS
```

For Trust Flex - TLS:
```
interface = hid,i2c,0x6C
device = ATECC608A-TFLXTLS
```

esguti, 2020-06-17:
Still same output:

- with **TFLXTLS** (DIP Switch 2 in DT100104):

```bash
interface = i2c,0x6C
device = ATECC608A-TFLXTLS
```

The output is:

```bash
p11tool --list-all-certs 'pkcs11:manufacturer=Microchip%20Technology%20Inc'

7192:7192:C_Initialize:144:
7192:7192:C_Initialize:145:CKR_OK(0)
7192:7192:C_GetInfo:162:
7192:7192:C_GetInfo:163:CKR_OK(0)
7192:7192:C_GetSlotList:187:
7192:7192:C_GetSlotList:188:CKR_OK(0)
7192:7192:C_GetTokenInfo:205:
7192:7192:pkcs11_token_get_info:372:Token Info: 1
7192:7192:pkcs11_token_get_info:405:Token Locked
7192:7192:C_GetTokenInfo:206:CKR_OK(0)
7192:7192:C_GetSlotInfo:196:
7192:7192:C_GetSlotInfo:197:CKR_OK(0)
7192:7192:C_OpenSession:274:
7192:7192:C_OpenSession:275:CKR_OK(0)
7192:7192:C_CloseSession:283:
7192:7192:C_CloseSession:284:CKR_OK(0)
7192:7192:C_GetSlotList:187:
7192:7192:C_GetSlotList:188:CKR_OK(0)
7192:7192:C_GetTokenInfo:205:
7192:7192:pkcs11_token_get_info:372:Token Info: 1
7192:7192:pkcs11_token_get_info:405:Token Locked
7192:7192:C_GetTokenInfo:206:CKR_OK(0)
7192:7192:C_GetSlotInfo:196:
7192:7192:C_GetSlotInfo:197:CKR_OK(0)
7192:7192:C_OpenSession:274:
7192:7192:C_OpenSession:275:CKR_OK(0)
7192:7192:C_FindObjectsInit:437:
7192:7192:CKA_CLASS(0):4:CKO_CERTIFICATE(1)
7192:7192:CKA_CERTIFICATE_TYPE(80):4:0x7e943cc0:
7192:7192:C_FindObjectsInit:438:CKR_OK(0)
7192:7192:C_FindObjects:446:
7192:7192:C_FindObjects:447:CKR_OK(0)
7192:7192:C_FindObjectsFinal:455:
7192:7192:C_FindObjectsFinal:456:CKR_OK(0)
7192:7192:C_CloseSession:283:
7192:7192:C_CloseSession:284:CKR_OK(0)
No matching objects found
```

- with **TNGTLS** (DIP Switch 1 in DT100104):

```bash
interface = i2c,0x6A
device = ATECC608A-TNGTLS
```

The output is:

```bash
p11tool --list-all-certs 'pkcs11:manufacturer=Microchip%20Technology%20Inc'

7307:7307:C_Initialize:144:
7307:7307:C_Initialize:145:CKR_OK(0)
7307:7307:C_GetInfo:162:
7307:7307:C_GetInfo:163:CKR_OK(0)
7307:7307:C_GetSlotList:187:
7307:7307:C_GetSlotList:188:CKR_OK(0)
7307:7307:C_GetTokenInfo:205:
7307:7307:pkcs11_token_get_info:372:Token Info: 1
7307:7307:pkcs11_token_get_info:405:Token Locked
7307:7307:C_GetTokenInfo:206:CKR_OK(0)
7307:7307:C_GetSlotInfo:196:
7307:7307:C_GetSlotInfo:197:CKR_OK(0)
7307:7307:C_OpenSession:274:
7307:7307:C_OpenSession:275:CKR_OK(0)
7307:7307:C_CloseSession:283:
7307:7307:C_CloseSession:284:CKR_OK(0)
7307:7307:C_GetSlotList:187:
7307:7307:C_GetSlotList:188:CKR_OK(0)
7307:7307:C_GetTokenInfo:205:
7307:7307:pkcs11_token_get_info:372:Token Info: 1
7307:7307:pkcs11_token_get_info:405:Token Locked
7307:7307:C_GetTokenInfo:206:CKR_OK(0)
7307:7307:C_GetSlotInfo:196:
7307:7307:C_GetSlotInfo:197:CKR_OK(0)
7307:7307:C_OpenSession:274:
7307:7307:C_OpenSession:275:CKR_OK(0)
7307:7307:C_FindObjectsInit:437:
7307:7307:CKA_CLASS(0):4:CKO_CERTIFICATE(1)
7307:7307:CKA_CERTIFICATE_TYPE(80):4:0x7ec17cc0:
7307:7307:C_FindObjectsInit:438:CKR_OK(0)
7307:7307:C_FindObjects:446:
7307:7307:C_FindObjects:447:CKR_OK(0)
7307:7307:C_FindObjectsFinal:455:
7307:7307:C_FindObjectsFinal:456:CKR_OK(0)
7307:7307:C_CloseSession:283:
7307:7307:C_CloseSession:284:CKR_OK(0)
No matching objects found
```

bryan-hunt, 2020-06-17:
What version of the library are you using?

esguti, 2020-06-17:
I use the current HEAD of the branch pkcs11 (2df0eb145c7241263d07577394dd12f8b1e783f0). I modify 2 files:
- in **lib/pkcs11/pkcs11_config.h**, I set the line 68 **"#define PKCS11_DEBUG_ENABLE             1"** to enable DEBUG
- in  **lib/atca_cfgs.c**, I set the line 45 **"atcai2c.bus            = 1,"** to use bus 1

bryan-hunt, 2020-06-17:
PKCS11 has been mainlined. Use v3.2.0 and make sure your CMake configuration has PKCS11 & the desired Trust Platform device support included. You can specify them all if you are unsure.

esguti, 2020-06-17:
Thanks for your help @bryan-hunt, but still it does not recognize the certificates.

I have set:

```bash
set(ATCA_HAL_I2C ON)
set(ATCA_PRINTF ON)
set(ATCA_PKCS11 ON)
set(ATCA_OPENSSL ON)
set(ATCA_TNGTLS_SUPPORT ON)
set(ATCA_TFLEX_SUPPORT ON)
set(ATCA_ATECC608A_SUPPORT ON)
```
With **0.conf** (DIP Switch 1):

```bash
interface = i2c,0x6A,1
device = ATECC608A-TNGTLS
```

List tokens result:

```bash
p11tool --list-tokens

13553:13553:C_Initialize:140:
13553:13553:C_Initialize:141:CKR_OK(0)
13553:13553:C_GetInfo:158:
13553:13553:C_GetInfo:159:CKR_OK(0)
Token 0:
	URL: pkcs11:model=p11-kit-trust;manufacturer=PKCS%2311%20Kit;serial=1;token=System%20Trust
	Label: System Trust
	Type: Trust module
	Flags: uPIN uninitialized
	Manufacturer: PKCS#11 Kit
	Model: p11-kit-trust
	Serial: 1
	Module: p11-kit-trust.so

13553:13553:C_GetSlotList:183:
13553:13553:C_GetSlotList:184:CKR_OK(0)
13553:13553:C_GetTokenInfo:201:
13553:13553:pkcs11_token_get_info:426:Token Info: 1
13553:13553:pkcs11_token_get_info:459:Token Locked
13553:13553:C_GetTokenInfo:202:CKR_OK(0)
13553:13553:C_GetSlotInfo:192:
13553:13553:C_GetSlotInfo:193:CKR_OK(0)
13553:13553:C_OpenSession:270:
13553:13553:C_OpenSession:271:CKR_OK(0)
13553:13553:C_CloseSession:279:
13553:13553:C_CloseSession:280:CKR_OK(0)
Token 1:
	URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=900CED1D691F;token=00ABC
	Label: 00ABC
13553:13553:C_GetSlotList:183:
13553:13553:C_GetSlotList:184:CKR_OK(0)
13553:13553:C_GetTokenInfo:201:
13553:13553:pkcs11_token_get_info:426:Token Info: 1
13553:13553:pkcs11_token_get_info:459:Token Locked
13553:13553:C_GetTokenInfo:202:CKR_OK(0)
13553:13553:C_GetSlotInfo:192:
13553:13553:C_GetSlotInfo:193:CKR_OK(0)
13553:13553:C_OpenSession:270:
13553:13553:C_OpenSession:271:CKR_OK(0)
13553:13553:C_CloseSession:279:
13553:13553:C_CloseSession:280:CKR_OK(0)
	Type: Hardware token, Trust module
	Flags: RNG, uPIN uninitialized
	Manufacturer: Microchip Technology Inc
	Model: ATECC608A
[... 183 lines cut ...]
13789:13789:C_CloseSession:280:CKR_OK(0)
13789:13789:C_GetSlotList:183:
13789:13789:C_GetSlotList:184:CKR_OK(0)
13789:13789:C_GetTokenInfo:201:
13789:13789:pkcs11_token_get_info:426:Token Info: 1
13789:13789:pkcs11_token_get_info:459:Token Locked
13789:13789:C_GetTokenInfo:202:CKR_OK(0)
13789:13789:C_GetSlotInfo:192:
13789:13789:C_GetSlotInfo:193:CKR_OK(0)
13789:13789:C_OpenSession:270:
13789:13789:C_OpenSession:271:CKR_OK(0)
13789:13789:C_FindObjectsInit:433:
13789:13789:CKA_CLASS(0):4:CKO_CERTIFICATE(1)
13789:13789:CKA_CERTIFICATE_TYPE(80):4:00 00 00 00 :
13789:13789:C_FindObjectsInit:434:CKR_OK(0)
13789:13789:C_FindObjects:442:
13789:13789:C_FindObjects:443:CKR_OK(0)
13789:13789:C_FindObjectsFinal:451:
13789:13789:C_FindObjectsFinal:452:CKR_OK(0)
13789:13789:C_CloseSession:279:
13789:13789:C_CloseSession:280:CKR_OK(0)
No matching objects found
```

Could be certificates format?

bryan-hunt, 2020-06-18:
Go ahead and try to add in ATCA_TNG_LEGACY_SUPPORT as well to your cmake configuration as well. Make sure to dump the cmake cache and do a clean build.

esguti, 2020-06-19:
Thank you @bryan-hunt . Now p11tool recognize the certificates

I add CMakeLists.txt:

```bash
set(ATCA_HAL_I2C ON CACHE BOOL "ATECC")
set(ATCA_PRINTF ON CACHE BOOL "ATECC")
set(ATCA_PKCS11 ON CACHE BOOL "ATECC")
set(ATCA_OPENSSL ON CACHE BOOL "ATECC")
set(ATCA_TNGTLS_SUPPORT ON CACHE BOOL "ATECC")
set(ATCA_TFLEX_SUPPORT ON CACHE BOOL "ATECC")
set(ATCA_ATECC608A_SUPPORT ON CACHE BOOL "ATECC")
set(ATCA_TNG_LEGACY_SUPPORT ON CACHE BOOL "ATECC")
```

and the output is:

```bash
p11tool --list-all-certs 'pkcs11:manufacturer=Microchip%20Technology%20Inc'

Object 0:
	URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=8F2C14585FA8;token=00ABC;object=device;type=cert
	Type: X.509 Certificate (EC/ECDSA-SECP256R1)
	Expires: Fri Jan  1 00:23:23 2038
	Label: device
	ID: 

Object 1:
	URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=8F2C14585FA8;token=00ABC;object=signer;type=cert
	Type: X.509 Certificate (EC/ECDSA-SECP256R1)
	Expires: Fri Jan  1 00:23:23 2038
	Label: signer
	Flags: CKA_CERTIFICATE_CATEGORY=CA; CKA_TRUSTED; 
	ID: 
```

My problem now is that openvpn still does not recognize the certificates:

```bash
openvpn --show-pkcs11-ids /usr/lib/libcryptoauth.so

The following objects are available for use.
Each object shown below may be used as parameter to
--pkcs11-id option please remember to use single quote mark.
```

Could you help me to get the certificates ids? I need them to configure the OpenVPN or do you know other VPN software able to use ATECCX08?

bryan-hunt, 2020-06-19:
This is a question of OpenVPN for what certificate requirements they have and why it is not working. You'll have to submit this request to [Microchip Support](https://microchipsupport.force.com) if you wish for Microchip to determine what OpenVPN software is doing. It will take time.

[How to submit a request](https://microchipsupport.force.com/s/article/How-to-submit-a-case)

closed 2020-07-08 by esguti