MicrochipTech/cryptoauthlib #163: USB interface from Linux to Atecc508a through Samd21

Gingex, 2020-06-12:
Atecc508a is connected to samd21 and I intend to use it with pkcs11 from a linux machine. Do you have any suggestion on how to write a samd21 fw to implement the protocol to convert pkcs11 commands into i2c comands for the cryptochip?

bryan-hunt, 2020-06-12:
Submit a request for the SAMD21 cryptoauth kit bridge firmware in the Microchip Support Portal which is a project that provides a USB-HID interface that cryptoauthlib (ATCA_HAL_KIT_HID) already interfaces with.

[How to Submit a Case](https://microchipsupport.force.com/s/article/How-to-submit-a-case)

Gingex, 2020-06-12:
Is this fw opensource?

bryan-hunt, 2020-06-12:
Not it is not specifically published to the web. Something very similar has been published in the [aws-iot-zero-touch-secure-provisioning-kit](https://github.com/MicrochipTech/aws-iot-zero-touch-secure-provisioning-kit) which is based on a SAMG55 however.

You can request if you get permission to publish it in your system (retaining the Microchip License) or else base your application on what was already published above.

closed 2020-07-27 by bryan-hunt