MicrochipTech/cryptoauthlib #166: PKCS11 token name

vgottardi2, 2020-06-21:
According to [wiki](https://github.com/MicrochipTech/cryptoauthlib/wiki/PKCS11-Linux-Setup), the token should be "token=0123EE". That's how it worked on pkcs11 branch. On master, this changed to "token=00ABC" (which breaks backwards compatibility). 
What is token supposed to be? Neither 0123EE nor 00ABC are really useful. I'm guessing this should be some fixed, descriptive string such as "ATECC". If not fixed, can we have the token name configurable in the slot configuration?

bryan-hunt, 2020-06-21:
0123EE is actually the three components of the device serial number that are not serialized (as in the reserved field and manufacturer code). As custom part would have been 0123XX. This was never well documented and broke for next gen devices that do not have the same structure for device serial numbers. Also AWS has a desire that this field represent the “Thing ID”. I’m releasing a small update at the end of week (V3.2.1) that addresses some of these problems by doing exactly what you’re suggesting - adding the label to the slot configuration file.

vgottardi2, 2020-06-21:
That's great. Thank you!

bryan-hunt, 2020-06-29:
v3.2.1 has been released. The label can now be specified in the slot configuration file. See [app/pkcs11/slot.confg.tmpl](https://github.com/MicrochipTech/cryptoauthlib/blob/v3.2.1/app/pkcs11/slot.conf.tmpl)

closed 2020-07-27 by bryan-hunt