MicrochipTech/cryptoauthlib #167: device name mismatch with AT88CK590 and ATECC508A

scottwilson815, 2020-06-24:
I'm using the ATECC508A in the USB-based AT88CK590.  When calling atcab_init(), cfg->devtype is set to ATECC508A.  This worked with previous versions of cryptoauthlib, however with the latest it fails.

In kit_init(), "board:device(00)" is sent to the device and "ECC108 TWI 00(C0)" is returned.  dev_type is set to the first word and the first 6 characters of dev_type are compared against "ATECC508A", resulting in a mismatch.

(kit_init continues, finding the other devices in the AT88CK590: "SHA204 TWI 00(C8)" and "AES132 TWI 00(A0)".)

Changing cfg->devtype to ATECC108A allows atcab_init() to succeed but I'm concerned that the incorrect devtype  could cause problems elsewhere in cryptoauthlib.

bryan-hunt, 2020-06-24:
Yes the AT988CK590 reports the wrong device information and with changes in 20190830 this became a problem. A patch for this is not going to be available until July. If in the mean time you can obtain the [Trust Platform Kit](https://www.microchip.com/developmenttools/ProductDetails/DM320118) that in all would be a better starting point anyway as the ATECC508A is not recommended for new designs. The trust platform also has numerous use cases and examples

bryan-hunt, 2020-10-22:
CK590 support for ECC devices should have been corrected in v3.2.3

closed 2020-12-03 by bryan-hunt