MicrochipTech/cryptoauthlib #168: TLS certificate authentication with ATECC608 key and Python 'requests'

vgottardi2, 2020-06-24:
Is there a way to use TLS certificate authentication with Python 'requests'?

With the old ateccx08 engine, the (pseudo)key was stored in a file and you could pass it to Python. With the engine configured in openssl.cnf, this used to work:

    # Instruct OpenSSL to load openssl.cnf 
    import OpenSSL 
    from OpenSSL._util import (ffi as _ffi, lib as _lib)
    _lib.OPENSSL_config(_ffi.NULL)

    req = requests.get(url, verify='root-ca.pem', cert=('device.pem', 'device.key'))

With the new pkcs11 engine, the key is in "engine format" which Python doesn't support. Is there a workaround to get Python requests working with a key in ATECC608?

So far, my searches for Python requests with PKCS11 came back with negative answers
https://stackoverflow.com/questions/58398697/how-to-establish-tls-session-in-python-using-pkcs11
https://stackoverflow.com/questions/36233685/access-a-page-that-require-safenet-usb-token-from-urllib2-ot-httplib

bryan-hunt, 2020-06-24:
I believe the trick is that you have add a new adapter that inherits from HTTPAdapter which you will then connect to a requests.Session object using the `mount` method. I haven't looked at in too much detail. There was PEP to make pkcs11 support native to the python ssl implementation but I believe the author of that PEP joined the swift team so it has languished for at least two python versions.

I haven't attempted to cobble this together myself yet but what I've seen does seem to revolve around attaching your own adapter

vgottardi2, 2020-07-10:
I ended up patching the Python modules to use the PKCS11 engine if the certificate key starts with 'pkcs11:'. Might not be the most elegant solution, but worked for my case.

closed 2020-07-10 by vgottardi2

bryan-hunt, 2020-07-10:
@vgottardi2 Do you mean recompiling a specific python module or something that could be monkey patched by a user?

vgottardi2, 2020-07-10:
'cryptography' module requires recompilation. I had to add ENGINE_load_private_key to https://github.com/pyca/cryptography/blob/master/src/_cffi_src/openssl/engine.py
I'm using Yocto, so recompilation was not a big issue.

bryan-hunt, 2020-07-10:
Oh. I didn't realize that requests used the cryptography module for the backing TLS session rather than the built-in implementation - that would make it easier to get a change pushed. Thanks!

krishnan793, 2024-01-16:
> 'cryptography' module requires recompilation. I had to add ENGINE_load_private_key to https://github.com/pyca/cryptography/blob/master/src/_cffi_src/openssl/engine.py I'm using Yocto, so recompilation was not a big issue.

Do you have any reference of the patch?

vgottardi2, 2024-01-18:
[meta-ecc508.zip](https://github.com/MicrochipTech/cryptoauthlib/files/13981848/meta-ecc508.zip)
@krishnan793, here are the patches from my Yocto/Dunfel (partial) layer. I just extracted the Python part. Sorry, I don't have them in a friendlier format. Should still be useful as a starting point.

krishnan793, 2024-01-30:
@vgottardi2 
Thanks a lot. :+1: