MicrochipTech/cryptoauthlib #169: Question: pkcs11 branch

clogwog, 2020-06-25:
To integrate cryptoauthlib as a pkcs11 engine i've been following https://github.com/MicrochipTech/cryptoauthlib/wiki/PKCS11-Linux-Setup and am using the pkcs11 branch.

this branch was last updated 8 months ago.  I also read that master was merged back into pkcs11 (or the other way around?)
is this branch still needed to integrate the microchip secure token into openssl ? Or is master the right code to use ?

bryan-hunt, 2020-06-25:
I’m in the process of updating the wiki instructions. You can use the master branch (specifically the release v3.2.0) which has several improvements. The token label does change so be aware of that. There is a v3.2.1 release in process however so you’ll see some improvements in the pkcs11 functionality Monday.

clogwog, 2020-06-25:
Thanks Bryan ! I'll keep an eye out.
I've been having some trouble addressing the private key for my 608A TNGTLS using the pkcs11 engine through openssl, so if i'm able to address it a different way that will work then that would be awesome !!

`#  openssl req -engine pkcs11 -new -key "pkcs11:object=012301;type=private" -keyform engine -x509 -out ~/req.csr
engine "pkcs11" set.
Found slot without user PIN
PKCS11_get_private_key returned NULL
cannot load Private Key from engine
3069973872:error:26096080:engine routines:ENGINE_load_private_key:failed loading private key:eng_pkey.c:124:
unable to load Private Key
`

(also tried pkcs11:object=0123EE;type=private )

bryan-hunt, 2020-06-25:
p11tool can really help (see the p11tool reference in the wiki) primarily to list the objects in the token. The objects are specified in a slot configuration file. The slot in this context is a pkcs11 context and not device context. The configuration file is normally in /var/lib/crypotauthlib you’ll need to copy the existing template and name it 0.conf (I.e slot 0)

clogwog, 2020-06-25:
perhaps i should wait for the 3.2.0 official release, but i did see a 3.2.0 tag on master :

cmake . is not liking https://github.com/MicrochipTech/cryptoauthlib/commit/159d8b8d1ae021c68af499a2ca82ded2312a370a#diff-5b579ba0237255aa2c66982834e4fa5eR68

`cmake .
CMake Error at lib/CMakeLists.txt:68 (source_group):
  source_group Unknown argument
  "/home/toms/projects/scratch/cryptoauthlib-3.2.0/cryptoauthlib/lib".
  Perhaps the FILES keyword is missing.
`

are there going to be different instructions for calling cmake on Monday ? I can wait. just thought i'd ask

bryan-hunt, 2020-06-25:
Version 3.2.0 is an official release with release notes. You can find the release on the release tab. There will be a 3.2.1 with small improvements. The cmake instructions haven’t changed from the wiki. I can fill in a bit more color but really you have to specify the correct source location and it is always best to work with a separate build directory rather than using the source directory. The source directory you need is the repo root.

clogwog, 2020-06-25:
followed the wiki

downloaded the release from the tar file extracted it
```
# cd cryptoauthlib
# mkdir build
# cd build
# cmake -DATCA_ATECC608A_SUPPORT=ON -DATCA_TFLEX_SUPPORT=on -DATCA_OPENSSL=ON -DATCA_HAL_KIT_HID=ON ../
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /media/toms/projects/projects/builds/BUILD/build-mtdata/tmp/sysroots/x86_64-linux/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-gcc
-- Check for working C compiler: /media/toms/projects/projects/builds/BUILD/build-mtdata/tmp/sysroots/x86_64-linux/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /media/toms/projects/projects/builds/BUILD/build-mtdata/tmp/sysroots/x86_64-linux/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-g++
-- Check for working CXX compiler: /media/toms/projects/projects/builds/BUILD/build-mtdata/tmp/sysroots/x86_64-linux/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for malloc
-- Looking for malloc - found
-- Looking for free
-- Looking for free - found
CMake Error at lib/CMakeLists.txt:68 (source_group):
  source_group Unknown argument
  "/media/toms/projects/projects/scratch/cryptoauthlib-3.2.0/cryptoauthlib/lib".
  Perhaps the FILES keyword is missing.

CMake Error at lib/CMakeLists.txt:69 (source_group):
  source_group Unknown argument
  "/media/toms/projects/projects/scratch/cryptoauthlib-3.2.0/cryptoauthlib/lib".
  Perhaps the FILES keyword is missing.

```

clogwog, 2020-06-25:
i've downloaded the 3.2.0 tar.gz file, extracted it and followed the wiki:

`root@MTDToms:/media/toms/projects/projects/scratch/cryptoauthlib-3.2.0/cryptoauthlib/build# cmake -DATCA_ATECC608A_SUPPORT=ON -DATCA_TFLEX_SUPPORT=on -DATCA_OPENSSL=ON -DATCA_HAL_KIT_HID=ON ../
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /media/toms/projects/projects/builds/BUILD/build-mtdata/tmp/sysroots/x86_64-linux/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-gcc
-- Check for working C compiler: /media/toms/projects/projects/builds/BUILD/build-mtdata/tmp/sysroots/x86_64-linux/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /media/toms/projects/projects/builds/BUILD/build-mtdata/tmp/sysroots/x86_64-linux/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-g++
-- Check for working CXX compiler: /media/toms/projects/projects/builds/BUILD/build-mtdata/tmp/sysroots/x86_64-linux/usr/bin/arm-poky-linux-gnueabi/arm-poky-linux-gnueabi-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for malloc
-- Looking for malloc - found
-- Looking for free
-- Looking for free - found
CMake Error at lib/CMakeLists.txt:68 (source_group):
  source_group Unknown argument
  "/media/toms/projects/projects/scratch/cryptoauthlib-3.2.0/cryptoauthlib/lib".
  Perhaps the FILES keyword is missing.

CMake Error at lib/CMakeLists.txt:69 (source_group):
  source_group Unknown argument
  "/media/toms/projects/projects/scratch/cryptoauthlib-3.2.0/cryptoauthlib/lib".
  Perhaps the FILES keyword is missing.
`

i wonder if there is something wrong with my cmake (version 3.6.1)

clogwog, 2020-06-25:
yes.. **source_group(TREE**  has only been added since [3.8](https://cmake.org/cmake/help/latest/release/3.8.html)

so you might need to update 

`cmake_minimum_required(VERSION 3.1.0)`

or find a different way of doing the same ?

clogwog, 2020-06-27:
please ignore question above, i see it was opened as a separate question https://github.com/MicrochipTech/cryptoauthlib/issues/170#issuecomment-649649349

clogwog, 2020-06-29:
I'm still updating our Yocto build system to use the 3.2.0 release, so that may take a bit.

> p11tool can really help (see the p11tool reference in the wiki) primarily to list the objects in the token. The objects are specified in a slot configuration file. The slot in this context is a [...]

i've got a 0.conf already for the old pkcs11 branch and that seems to work, i'll update it once i have 3.2.0 repackaged.
 
> Also for a TNGTLS part you’ll definitely want to use the v3.2.0 branch as that makes the configuration quite simple. The configuration file just needs two lines then
> 
> Interface which may be i2c or hid depending on how you’re connecting it
> 
> And the device line:
> device = ATECC608A-TNGTLS
> 
> Comment out any object specifiers

this sounds much easier.

for the 
-key "pkcs11:object=012301;type=private"

what does a normally configured 60bA-TNGTLS pkcs11 'key-selector" look like ?
and is the 012301 the serial number of the chip ? 

the problem with the p11tool is that it is currently not building when i ask for the gnutls-tools package.. i think the p11-kit version is too old..(and for production we want to run with openssl only)  so i cannot check the output of the tool for the 'pkcs11 key selector' so any help would be appreciated by me.

clogwog, 2020-06-30:
found a way to get p11tool to run:
`./run ./p11tool --list-all
./p11tool: /usr/lib/libp11-kit.so.0: no version information available (required by ./p11tool)
C_Initialize
  IN: pInitArgs = NULL
C_Initialize = CKR_OK
C_GetInfo
 OUT: pInfo = {
	cryptokiVersion: 2.40
	manufacturerID: "Microchip Technology Inc"
	flags: 0
	libraryDescription: "Cryptoauthlib PKCS11 Interface"
	libraryVersion: 3.2
      }
C_GetInfo = CKR_OK
C_GetSlotList
  IN: tokenPresent = CK_TRUE
  IN: pulCount = 0xBE8276A4 = 48
 OUT: pSlotList = (1) [ SL0 ]
C_GetSlotList = CKR_OK
C_GetTokenInfo
  IN: slotID = SL0
 OUT: pInfo = {
	label: "012301"
	manufacturerID: "Microchip Technology Inc"
	model: "ATECC608A"
	serialNumber: "864EC2CF8196"
	flags: 1029 = CKF_RNG | CKF_LOGIN_REQUIRED | CKF_TOKEN_INITIALIZED
	ulMaxSessionCount: 1
	ulSessionCount: 0
	ulMaxRwSessionCount: 1
	ulRwSessionCount: 0
....
....
C_FindObjects = CKR_OK
C_FindObjectsFinal
  IN: hSession = S3068330960
C_FindObjectsFinal = CKR_OK
C_CloseSession
  IN: hSession = S3068330960
C_CloseSession = CKR_OK
Object 0:
	URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=864EC2CF8196;token=012301;object=device;object-type=private
	Type: Private key
	Label: device
	Flags: CKA_PRIVATE; CKA_SENSITIVE; 
	ID: 
Object 1:
	URL: pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=864EC2CF8196;token=012301;object=device;object-type=public
	Type: Public key
	Label: device
	ID: 
C_Finalize
  IN: pReserved = NULL
C_Finalize = CKR_OK`

but when i ask for a CSR:

`root@v11_9:~/p11# openssl req -engine pkcs11 -key "pkcs11:model=ATECC608A;manufacturer=Microchip%20Technology%20Inc;serial=864EC2CF8196;token=012301;object=device;object-type=private" -keyform engine -new -out new_device.csr -subj "/CN=NEW CSR EXAMPLE"
engine "pkcs11" set.
Found slot without user PIN
PKCS11_get_private_key returned NULL
cannot load Private Key from engine
3069928816:error:26096080:engine routines:ENGINE_load_private_key:failed loading private key:eng_pkey.c:124:
unable to load Private Key`

woelfman, 2020-06-30:
I can get a private key but not the CSR with `3.2.x` as well.

bryan-hunt, 2020-06-30:
The PKCS11 URIs have to exactly match and the environment has to be properly configured. Based on what I am seeing

a) If p11-kit is not being used then the openssl.cnf configuration is wrong and needs to be debugged
b) If p11-kit is being used then either the global or user configuration is wrong (or a user configuration is defined but you're running in sudo).

clogwog, 2020-07-01:
All good now !!  thank you  @bryan-hunt  !!

i have 
upgraded to cryptoauthlib 3.2.1 (including the update to the 2 line 0.conf)
updated my libp11 (from 0.4.0 to 0.4.10) 
and i configured my system for non p11-kit-proxy (configured it for both before !!)

i can now request the  CSR using openssl 
thank you for you patience !! That last comment triggered something to go back to the original instructions at 
https://github.com/MicrochipTech/cryptoauthlib/wiki/PKCS11-Linux-Setup
and re-read.

for anyone else:
libp11 >= 0.4.10
cryptoauthlib >= 3.2.0

/var/lib/cryptoauthlib/0.conf
`interface = i2c,0x6A
device = ATECC608A-TNGTLS
`

/usr/lib/ssl/openssl.cnf 
`openssl_conf = openssl_init`
at top

```
[openssl_init]                                                        
engines=engine_section                                                  
                                                                      
[engine_section]                                                           
pkcs11 = pkcs11_section                                                      
                                                                           
[pkcs11_section]                                                          
engine_id = pkcs11                                                    
dynamic_path = /usr/lib/engines/pkcs11.so                        
MODULE_PATH = /usr/lib/libcryptoauth.so                              
init = 0     
```                                                                                     
at bottom

closed 2020-07-01 by clogwog