MicrochipTech/cryptoauthlib #170: raise cmake_minimum_required to 3.8?

scottwilson815, 2020-06-25:
I happened to run cmake on a system where the version is 3.7.2 and got errors about the use of TREE in source_group.  Raising cmake_minimum_required to 3.8 (the version where I think TREE was introduced), or whatever is the appropriate version, would make the source of the problem more obvious.  I'm not a regular cmake user and maybe 3.7.2 is so ancient that no one else would ever run into this problem.

bryan-hunt, 2020-06-25:
No it is good form to have the proper versions installed or at least checked. I'll make the update.

bryan-hunt, 2020-06-29:
Can you confirm that the latest release (v.3.2.1) works for your environment?

scottwilson815, 2020-06-29:
Verified v3.2.1 builds with cmake version 3.7.2.

Thanks.

closed 2020-06-29 by bryan-hunt