MicrochipTech/cryptoauthlib #172: Extract device certificate

woelfman, 2020-06-30:
I'm trying to extract a device certificate from an ATECC508A. I see `app/pkcs11` but I don't understand how to apply it to my setup.

## 0.conf
```
label = MCHP
interface = i2c,0xC0,0
device = ATECC608A-TNGTLS
#device = ATECC508A
```

I'm not certain which device I need to specify for the device certificate to work, but `ATECC608A-TNGTLS` seems to work at least for some commands.

```sh
$ p11tool --export-chain "pkcs11:token=MCHP;object=device;type=cert"
Error in pkcs11_export_chain:610: The requested data were not available.
```

I have `device_def.c` and `signer_def.c`that I'm aware may need to be added to the build, but I'm not sure how to incorporate them. It doesn't seem to be enough to just edit `example_cert_chain.c`, but I don't see instructions on how to actually enable and use it with p11tool.

bryan-hunt, 2020-06-30:
If you have definitions for a custom part then you need to submit a support request to [Microchip Support](https://microchipsupport.force.com/s/) to help you produce the proper configuration for your part.

[How to submit a support request](https://microchipsupport.force.com/s/article/How-to-submit-a-case)

woelfman, 2020-06-30:
I'll give that a try. Thanks Bryan.

closed 2020-06-30 by woelfman