MicrochipTech/cryptoauthlib #174: Snapcraft: Don't require --no-cache option for pip3 install

WillNilges, 2020-07-02:
This is somewhat related to #136 because I experience the same symptoms. Basically, I'm trying to snap my python app which uses `cryptoauthlib` as a dependency. However, because of the bug mentioned in #136, I think I'm going to have to use the `--no-cache` option to make this app work. Unfortunately, it doesn't seem like Snapcraft supports passing this argument. I'm poking at the source code to see if I might be able to add support, but in the meantime, I need a way to correctly include this library in my snap. Any ideas for how I can do this? Why does `cryptoauthlib` need to be installed in this way?

bryan-hunt, 2020-07-02:
That’s only required if your environment isn’t set up properly and you had a failed install. The no-cache option for pip resets the build state in pip so it will start the build from scratch after you have resolved your environment problem.

WillNilges, 2020-07-02:
Okay, gotcha. I'll keep messing with it. If that's the case, I'm not sure what's wrong. Maybe the `pip` package is busted?

bryan-hunt, 2020-07-02:
If that were the case then no-cache would fail consistently since it will be downloading it from the package repository and starting from scratch.

WillNilges, 2020-07-02:
Fair point. I would like to point out, though, that the pip package is about 3 months out of date: https://pypi.org/project/cryptoauthlib/

bryan-hunt, 2020-07-02:
Yes there has been nothing meaningful changed in the python side - all external facing APIs have remained the same.

WillNilges, 2020-07-02:
:eyes: also a fair point :sweat_smile: 

Just nuked my cache and re-snapped, and the error is gone. Thanks for your time!

closed 2020-07-02 by WillNilges