MicrochipTech/cryptoauthlib #175: hal\hal_esp32_i2c wont compile

ASH247iT, 2020-07-14:
Hi - Trying to implement Cryptoauthlib for ATECC608 with ESP32 using i2c. I am not sure fully of the requirements but it seems the atca_hal needs something for the underlying functions and hal_esp32_i2c seemed an obvious choice.

When I try and compile I get a few trivial errors and the hal_i2c_receive method is called with 3 params but it needs 4. I also get a warning: missing initializer for field 'hal_data' which I dont understand. Can you help / fix please ? Heres the log (Im using ESP-IDF /Platformio)

components\cryptoauthlib-master\lib\hal\hal_esp32_i2c.c: In function 'hal_i2c_receive':
components\cryptoauthlib-master\lib\hal\hal_esp32_i2c.c:228:9: error: 'retries' undeclared (first use in this function); did you mean 'eSetBits'?
         retries = cfg->rx_retries;
         ^~~~~~~
         eSetBits
components\cryptoauthlib-master\lib\hal\hal_esp32_i2c.c:228:9: note: each undeclared identifier is reported only once for each function it appears in
components\cryptoauthlib-master\lib\hal\hal_esp32_i2c.c:229:33: error: 'status' undeclared (first use in this function); did you mean 'strtoul'?
         while (retries-- > 0 && status != ATCA_SUCCESS)
                                 ^~~~~~
                                 strtoul
components\cryptoauthlib-master\lib\hal\hal_esp32_i2c.c:293:27: error: 'rxdata_max_size' undeclared (first use in this function)
         if (read_length > rxdata_max_size)
                           ^~~~~~~~~~~~~~~
components\cryptoauthlib-master\lib\hal\hal_esp32_i2c.c:213:9: warning: unused variable 'low' [-Wunused-variable]
     int low = 0;
         ^~~
components\cryptoauthlib-master\lib\hal\hal_esp32_i2c.c:212:9: warning: unused variable 'high' [-Wunused-variable]
     int high = 0;
         ^~~~
components\cryptoauthlib-master\lib\hal\hal_esp32_i2c.c: In function 'hal_i2c_wake':
components\cryptoauthlib-master\lib\hal\hal_esp32_i2c.c:396:28: warning: passing argument 2 of 'hal_i2c_receive' makes integer from pointer without a cast [-Wint-conversion]
     hal_i2c_receive(iface, data, &rxlen);
                            ^~~~
components\cryptoauthlib-master\lib\hal\hal_esp32_i2c.c:207:54: note: expected 'uint8_t' {aka 'unsigned char'} but argument is of type 'uint8_t *' {aka 'unsigned char *'}
 ATCA_STATUS hal_i2c_receive(ATCAIface iface, uint8_t word_address, uint8_t *rxdata, uint16_t *rxlength)
                                              ~~~~~~~~^~~~~~~~~~~~
components\cryptoauthlib-master\lib\hal\hal_esp32_i2c.c:396:34: warning: passing argument 3 of 'hal_i2c_receive' from incompatible pointer type [-Wincompatible-pointer-types]
     hal_i2c_receive(iface, data, &rxlen);
                                  ^~~~~~
components\cryptoauthlib-master\lib\hal\hal_esp32_i2c.c:207:77: note: expected 'uint8_t *' {aka 'unsigned char *'} but argument is of type 'uint16_t *' {aka 'short unsigned int *'}
 ATCA_STATUS hal_i2c_receive(ATCAIface iface, uint8_t word_address, uint8_t *rxdata, uint16_t *rxlength)
                                                                    ~~~~~~~~~^~~~~~
components\cryptoauthlib-master\lib\hal\hal_esp32_i2c.c:396:5: error: too few arguments to function 'hal_i2c_receive'
     hal_i2c_receive(iface, data, &rxlen);
     ^~~~~~~~~~~~~~~
components\cryptoauthlib-master\lib\hal\hal_esp32_i2c.c:207:13: note: declared here
 ATCA_STATUS hal_i2c_receive(ATCAIface iface, uint8_t word_address, uint8_t *rxdata, uint16_t *rxlength)
             ^~~~~~~~~~~~~~~
components\cryptoauthlib-master\lib\hal\hal_esp32_i2c.c: In function 'hal_i2c_receive':
components\cryptoauthlib-master\lib\hal\hal_esp32_i2c.c:343:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
cc1.exe: some warnings being treated as errors
*** [.pio\build\esp-32-11c8f18d69\esp-idf\cryptoauthlib-master\lib\hal\hal_esp32_i2c.c.o] Error 1
components\cryptoauthlib-master\lib\hal\atca_hal.c:48:1: warning: missing initializer for field 'hal_data' of 'ATCAHAL_t' {aka 'struct <anonymous>'} [-Wmissing-field-initializers]
 };
 ^
In file included from components\cryptoauthlib-master\lib/cryptoauthlib.h:78,
                 from components\cryptoauthlib-master\lib\hal\atca_hal.c:35:
components\cryptoauthlib-master\lib/hal/atca_hal.h:67:11: note: 'hal_data' declared here
     void *hal_data;       // points to whatever the HAL implementation for this interface wants it to, HAL manages.
           ^~~~~~~~

Thanks

ASH247iT, 2020-07-15:
I looked a bit further and the error is on line 396 in the hal_i2c_wake function
It is missing a parameter Word Address - where could I get this info - Im thinking it should be something like - 0x03
I think the hal_i2c_receive was changed in the last release to have 4 params @bryan-hunt 
Thanks

ASH247iT, 2020-07-15:
I found this table - I think 0x03 seems right after a Wake - can you confirm?
[image]

kmwebnet, 2020-07-28:
Hello,
I also failed to build hal_esp32_i2c.c and rewrote it.
I put the code that worked in the following.
https://github.com/kmwebnet/ECC608-AzureDPS-Provision-idf4/blob/master/components/cryptoauthlib/hal_esp32_i2c.c
Just FYI.

ASH247iT, 2020-08-15:
Many Thanks Kmwebnet
That worked well once I figured out the slave address for the TNG version :-)

rohansingh, 2020-08-17:
Yeah folks, `hal_esp32_i2c.c` is completely broken in v3.2. I reverted to v3.1 and it compiles.

The commit that broke things seems to be 159d8b8d1ae021c68af499a2ca82ded2312a370a (/cc @bryan-hunt). It's a very large commit so I'm not sure what the provenance is of the specific changes to the ESP32 HAL.

AdityaHPatwardhan, 2021-04-05:
@kmwebnet Nice work 👏
@bryan-hunt What is the timeline for fixing the issue? can we expect a fix in `v3.3.1` ?

bryan-hunt, 2021-04-27:
This should be finally taken care of in v3.3.1

closed 2021-05-06 by bryan-hunt