MicrochipTech/cryptoauthlib #176: Cryptoauthlib unable to import in yocto linux.

hsnhrn, 2020-07-17:
I am using a yocto distribution and installed the meta-atmel layer which contains the cryptoauthlib recipe.

i am trying to run these examples https://github.com/MicrochipTech/cryptoauthtools/tree/master/python/examples
but i am not able to import cryptoauthlib and got this error below :
File "/usr/lib/python3.8/site-packages/cryptoauthlib/init.py". line 16. in load_cryptoauthlib()
File "/usr/lib/python3.8/site-packages/cryptoauthlib/library.py". line 83. in load_cryptoauthlib()
_CRYPTO_LIB =ctypes.cdll.LoadLibrary(os.path.join(curr_path. "libcryptoauth.so"))
File "/usr/lib/python3.8/ctypes/init.py". line 451. in LoadLibrary return self._dlltype(name)
File "/usr/lib/python3.8/ctypes/init.py". line 373. in init self._handle = _dlopen(self._name mode)
OSError : /usr/lib/python3.8/site-packages/cryptoauthlib/libcryptoauth.so: undefined symbol: hid_open

I have created an issue in the meta-atmel page too but it seems like lib problem
How can i fix this? thanks a lot!

bryan-hunt, 2020-07-17:
This would be a configuration issue. Are you going to be using a kit such as the [CryptoAuth Trust Platform Development Kit](https://www.microchip.com/developmenttools/ProductDetails/DM320118)? If you are then you'll have to include USB support (e.g. libudev) to your build. If you are not using a USB kit then you'll need to set the environment variable CRYPTOAUTHLIB_NOUSB for the python build.

hsnhrn, 2020-07-17:
I used Secure UDFN click with ATECC608A microchip with raspberry pi. Now porting it to STM32MP1 SBC having Yocto OS dunfell branch.  does this flag alse needed to be included if i am only testing it by typing import cryptoauthlib?

bryan-hunt, 2020-07-17:
The python library has a built extension. That extension if built with usb support needs to have the corresponding library available on the target or the extension can not be loaded. If you're not using USB you need to build the python extension without usb support (hence the above environment variable). The environment variable controls the build during build/install depending on which python setup/install process you are running, because of that I can't say when or where you need to have it set in your specific case.