MicrochipTech/cryptoauthlib #177: Fault/Crash when calling atcab_init_ext()

deangereaux, 2020-07-23:
Using version 3.2.1.  In atca_basic.c, function atca_init_ext(), crashes when calling atcab_read_bytes_zone() when the device type is ATECC608A because the global _gDevice isn't set.   Here's a quick fix I tested:

```#ifdef ATCA_ATECC608A_SUPPORT
        if (cfg->devtype == ATECC608A)
        {
            // FIX  BEG
            _gDevice = *device;
            // FIX END
            
            if ((status = atcab_read_bytes_zone(ATCA_ZONE_CONFIG, 0, ATCA_CHIPMODE_OFFSET, &(*device)->mCommands->clock_divider, 1)) != ATCA_SUCCESS)
            {
                return status;
            }
            (*device)->mCommands->clock_divider &= ATCA_CHIPMODE_CLOCK_DIV_MASK;
        }
#endif
```

bryan-hunt, 2020-07-23:
This is a misuse of the api. If you’re using atcab functions you need to use the atcab knit function which is the original api.

deangereaux, 2020-07-23:
What is the proper API to call when using the new calib_() funcs?  I've got an 608A and 508A I need to manage, the ability of the calib_() routines to use the device as an parameter is great.

I would suggest making atcab_init_ext() static as to not expose the API.

bryan-hunt, 2020-07-23:
If you’re using calib calls then yes you use the _ext function to initialize your device struct you provide to them. I do see what you mean now however for the 608 that the atcab call is being made during the init there. That will be fixed shortly. The fix would be to change that call to calib_read_bytes_zone(*device,...)

bryan-hunt, 2020-07-23:
As an FYI v3.2.3 is scheduled for friday

bryan-hunt, 2021-03-21:
This was resolved in v3.2.2

closed 2021-03-21 by bryan-hunt