MicrochipTech/cryptoauthlib #178: Feature Request:  Multiple ATCADevice instances statically allocated.

deangereaux, 2020-07-23:
Add the ability to use multiple ATCADevice instances without using any heap. Currently in atcab_init_ext(), if ATCA_NO_HEAP is defined then the global g_atca_device is reused, thus you can’t have to valid instances of ATCADevice.   For embedded environments ideally multiple ATCADevice strutcs would be allocated statically and initialized separately.  Something like this pseudo code.

```static ATCADevice  main_608A;
static ATCADevice  peripheral_608A.	

atca_init_instance(&main_608A, cfg_608a);
atca_init_instance(&peripheral_608A, cfg_periph);
```

bryan-hunt, 2021-03-21:
Do you mind checking out v3.3.0 and see if the changes to the ATCADevice and atca_iface_t structures satisfy your use case?

deangereaux, 2021-03-22:
Yes I'll verify.

deangereaux, 2021-03-23:
`initATCADevice()` satisfies this use case.   This function has been around for a while, I probably didn’t see it when I posted this enhancement request.   You can close this enhancement request.  Thanks,

closed 2021-11-05 by bryan-hunt