MicrochipTech/cryptoauthlib #179: Need to read twice the size of the certificate

lecocqg, 2020-07-27:
Hello,

I'm using cryptoauthlib v3.2.1 and I'm trying to read the device certificate of an ATECC608A-TFLXTLSS.

I'm using the certificate template from [tng folder](https://github.com/MicrochipTech/cryptoauthlib/tree/master/app/tng) (not sure if it's the correct one).

```
#include "cryptoauthlib.h"
#include "atcacert/atcacert_def.h"
#include "atcacert/atcacert_client.h"
#include "tflxtls_cert_def_4_device.h"

int main()
{
    ATCA_STATUS status;
    uint8_t serial_number[9];
    size_t max_cert_size = 0;
    size_t cert_size = 0;
    const atcacert_def_t cert_def = g_tflxtls_cert_def_4_device;

    cfg_ateccx08a_i2c_default.atcai2c.slave_address = 0x6C;
    cfg_ateccx08a_i2c_default.atcai2c.bus = 1;
    status = atcab_init(&cfg_ateccx08a_i2c_default);

    if (status != ATCA_SUCCESS)
    {
        printf("Error init %d %x\n", status, status);
        return status;
    }

    // Read cert size
    printf("CERT SIZE\n");
    status = atcacert_read_cert_size(&cert_def, &cert_size);
    printf("status: %02x\n", status);
    printf("cert_size: %d\n\n", cert_size);

    // Certificate in PEM
    printf("CERT IN PEM\n");
    cert_size *= 2; //TODO why ?
    char csr_pem_buffer[cert_size];
    size_t csr_pem_buffer_length = sizeof(csr_pem_buffer);
    status = atcacert_create_csr_pem(&cert_def, (char*)&csr_pem_buffer, &csr_pem_buffer_length);
    printf("status: %02x\n", status);
    printf("%s\n", csr_pem_buffer);

    return status;
}
```

When I'm using directly cert_size, I get an error ATCACERT_E_BUFFER_TOO_SMALL with atcacert_create_csr_pem(). I need to double it to have the complete certificate. Am I doing something wrong ?

bryan-hunt, 2020-07-27:
You would read the certificate using the following:

```
    ret = tng_atcacert_max_device_cert_size(&cert_size);
    TEST_ASSERT_EQUAL(ATCACERT_E_SUCCESS, ret);

   cert = malloc(cert_size);

    ret = tng_atcacert_read_device_cert(cert, &cert_size, NULL);
    TEST_ASSERT_EQUAL(ATCACERT_E_SUCCESS, ret);
```

atcacert_create_csr_pem is used to create a CSR in PEM format. A CSR is not the signed certificate.

Have you started with the [Trust Platform Design Suite](https://microchipdeveloper.com/authentication:trust-platform)?

lecocqg, 2020-07-28:
I originally started from the schematic named : AWZ & Azure C

I only found an examples in : 
cryptoauth_trustplatform_designsuite/TrustnGO/05_cloud_connect/firmware/src/common/cloud_wifi_ecc_process.c

I'm now trying your method, but I'm getting an ATCACERT_E_WRONG_CERT_DEF on tng_atcacert_read_device_cert(). After an investigation, I found that there's an empty string returned from the read in the OTP zone (app/tng/tng_atca.c line 91).
```
int ret;
size_t device_cert_size = 0;
uint8_t *device_cert = NULL;

ret = tng_atcacert_max_device_cert_size(&device_cert_size);
printf("status: %02x cert_size: %d\n", ret, device_cert_size);

device_cert = malloc(device_cert_size);

ret = tng_atcacert_read_device_cert(device_cert, &device_cert_size, NULL);
printf("status: %02x \n", status);

return ret;
```

bryan-hunt, 2020-07-28:
That is an unusual circumstance. Please open a support request with [Microchip Support](https://www.microchip.com/support) and describe your issue and also report where you obtained your part(s) and their serial number(s).

[How to submit a technical support request](https://microchipsupport.force.com/s/article/How-to-submit-a-case)

het-masteen, 2020-07-28:
All right, thanks for your help. I will close this ticket, since I opened an other one in Microchip Support.

I will update the current ticket when I get some news from the support.

closed 2020-07-28 by het-masteen

het-masteen, 2020-07-29:
I found a workaround by forcing the cert_def and reading the 3 certificates (device, root, signer) with `atcacert_read_cert()` and `tng_atcacert_root_cert()`.