MicrochipTech/cryptoauthlib #182: ATECC608A SlotConfig / KeyConfig for storing public P256 key to use with Verify(Stored) command

DonV83, 2020-08-24:
I have an ECC P256 Public Key that I want to store in slot 09 of ATECC608A. I use the matching private key to generate a signature of a SHA256 hash. I already have the R and S components of this signature.

I want to use the ATECC608A to verify the hash given the ECC Public Key in slot 09, the 32 bytes message hash in the TempKey buffer and the R and S components in the Verify Command bytes (Data1 and Data2).

I have configured slot 09 as follows: SlotConfig: 0x0080; KeyConfig: 0x3000.
I have configured slot 0A as follows: SlotConfig: 0x0020; KeyConfig: 0x3000.
I have configured slot 0B as follows: SlotConfig: 0x8080; KeyConfig: 0x3000.
I have configured slot 0C as follows: SlotConfig: 0x8020; KeyConfig: 0x3000.

When I use the Nonce command to store the 32 bytes hash in TempKey (I use the ACES v6 software to check that this actually happens correctly) and then run the **Verify(Stored)** command using either slot 09, 0A, 0B or 0C, I always get an Execution Error (0x0F).

However, when I use the Nonce command to store the 32 bytes hash in TempKey (I use the ACES v6 software to check that this actually happens correctly) and then run the **Verify(External)** command, providing the X and Y of the ECC Public Key in the command, it always works and it returns 0x00, Command completed succesfully. This proves that my R, S, X and Y match my 32 bytes SHA256 hash.

I have formatted the Public Key in slots 09, 0A, 0B and 0C according to the manual, like: 0x00, 0x00, 0x00, 0x00, X, 0x00, 0x00, 0x00, 0x00, Y (so we fill up exactly 72 bytes).

I think that my KeyConfig and/or SlotConfig is improperly configured, but I have not been able to figure out a correct configuration to accomplish my goal. I hope someone can point me in the right direction.

Kind regards,
Don

bryan-hunt, 2020-08-27:
This does not appear to be a library related issue. 

For product support questions you should submit a product support request through the [Microchip Support portal](https://microchipsupport.force.com/s/). We are unable to provide the same level of product usage support via github as we do through the support portal.

For details on the Support Portal or how to use it please see [How to Submit a Case](https://microchipsupport.force.com/s/article/How-to-submit-a-case)

A couple of important items:

a) You need to provide the complete config zone data - from the above I can't tell if the part is locked or not. If the part is unlocked you can't perform cryptographic operations with stored keys.

b) cryptoauthlib has many helper functions to ensure data is properly written to the device. I would suggest using atcab_write_pubkey to ensure the key is properly written

DonV83, 2020-08-28:
Hi Bryan,

Thanks for your quick reply. I looked into the details regarding the atcab_write_pubkey function and I think that my public key was not properly formatted. We can close this issue for now.

Kind regards,
Don

closed 2020-08-28 by bryan-hunt