MicrochipTech/cryptoauthlib #184: How do I implement ECDH using cryptoauthlib?

T-Akasaki, 2020-09-04:
I have a Socket server up in Python and the clients are connected using SAME54 (+ ECC608A).

We recognize that there are five tasks to be done with ECDH
We are referring to the TLS1.2 handshake.
1. ECC key exchange
2. random value exchange
3. Creating a Pre-Master Secret
4. Create Master Secret
5. Key Block Creation

1 the ECC key exchange is done.
```
xResult = atcab_ecdh_tempkey(&ucPublicKeyDer[ASN1_HEADER_SIZE], &ucPreMasterSecret);
```
2 can be replaced using the current fixed value.
3 has been created.
4 and 5 have not been created.

In order to create Master Secret and Key Block, which function should I call, and in what order?

bryan-hunt, 2020-09-21:
Detailed TLS specification support is really outside the scope of anything we can provide.

bryan-hunt, 2020-09-21:
WolfSSL is included in Harmony 3 and comes with a license for use on Microchip Products. Given you're using a SAME54 I would recommend using the onboard ECDH capabilities for ephemeral TLS keys anyway. Regardless WolfSSL has ATECC608A support built in with a tested TLS implementation.

For product support questions you should submit a product support request through the [Microchip Support portal](https://microchipsupport.force.com/s/). For details on the Support Portal or how to use it please see [How to Submit a Case](https://microchipsupport.force.com/s/article/How-to-submit-a-case)

closed 2020-10-22 by bryan-hunt