MicrochipTech/cryptoauthlib #187: Question: How to build on windows?

minghuadev, 2020-09-30:
Hi, I want to add an I2C interface through a buspirate dongle. On Linux, the library can be built using "python setup.py install" to build and install, or using "python setup.py build_ext" to build only. How that can be done on a windows machine? 

Thanks.

bryan-hunt, 2020-10-05:
The same way. You'll need something that cmake can detect however without passing in a bunch of arguments - generally that means having MSVC installed.

minghuadev, 2020-10-22:
Thanks. I'm wishing that a gcc can do the work.

closed 2020-10-22 by minghuadev

bryan-hunt, 2020-10-22:
@minghuadev you'll have to identify the toolchain information to CMake. If you're using the gui it'll have a prompt when you run configure for what generator to use. If you running from the command line you'll have to manually specify the generator if CMake can't identify a toolchain. For example to use gcc as part of mingw (the common way) you can run cmake from mingw or specify `-G "MinGW Makefiles"` with your CMake call.