MicrochipTech/cryptoauthlib #188: Question: how to deal with status WAKE_SUCCESS 0x11?

minghuadev, 2020-09-30:
Hi, When running the "config.py" command from the cryptoauthtools repo, I'm seeing status 0x11 for write commands. This happens only on address 8, and address 24, but not other addresses in the configuration area. 

The code is modified to debug the problem, and to change the status 0x11 to 0 so the program will continue: 
```
--- a/lib/basic/atca_basic_write.c
+++ b/lib/basic/atca_basic_write.c
@@ -95,11 +95,15 @@ ATCA_STATUS atcab_write(uint8_t zone, uint16_t address, const uint8_t *value, co

         if ((status = atca_execute_command(&packet, _gDevice)) != ATCA_SUCCESS)
         {
+            printf("  %s %d status %d ", __func__, __LINE__, status);
+            if ( status == ATCA_WAKE_SUCCESS ) { printf("wake-success"); status = 0; }
+            printf("\n");
             break;
         }

     }
     while (0);
+    printf("  %s %d status %d zone %u address %u\n", __func__, __LINE__, status, zone, address);

     return status;
 }
```

The screen output will see: 
```
    Programming ATECC608A Configuration
  atcab_write 106 status 0 zone 0 address 4
  atcab_write 106 status 0 zone 0 address 5
  atcab_write 106 status 0 zone 0 address 6
  atcab_write 106 status 0 zone 0 address 7
  atcab_write 98 status 243 wake-success
  atcab_write 106 status 0 zone 128 address 8
  atcab_write 106 status 0 zone 0 address 16
  atcab_write 106 status 0 zone 0 address 17
  atcab_write 106 status 0 zone 0 address 18
  atcab_write 106 status 0 zone 0 address 19
  atcab_write 106 status 0 zone 0 address 20
  atcab_write 106 status 0 zone 0 address 22
  atcab_write 106 status 0 zone 0 address 23
  atcab_write 98 status 243 wake-success
  atcab_write 106 status 0 zone 128 address 24
        Success
```

Though this change does not fix the problem. The verification step will fail, that means data were not written properly. 

To debug further, I added retries of the writes those are getting status 0x11. This seems to pass the verification step: 
```
diff --git a/lib/atca_execution.c b/lib/atca_execution.c
index bc96e1a..1704251 100644
--- a/lib/atca_execution.c
+++ b/lib/atca_execution.c
@@ -323,6 +323,16 @@ ATCA_STATUS atca_execute_command(ATCAPacket* packet, ATCADevice device)
             break;
         }

+        #define HAL_HACK_RETRY (1) /* 0 to disable */
+        #if defined(HAL_HACK_RETRY) && (HAL_HACK_RETRY)
+        int retrycnt = 0;
+        unsigned int pktsz = packet->txsize;
+        unsigned char pktbuf[256];
+        if (pktsz > 255) printf("Error: size over bufer limit\n");
+        memcpy(pktbuf, (uint8_t*)packet, pktsz+1);
+      retry: retrycnt ++;
+        memcpy( (uint8_t*)packet, pktbuf, pktsz+1);
+        #endif
         // send the command
         if ((status = atsend(device->mIface, (uint8_t*)packet, packet->txsize)) != ATCA_SUCCESS)
         {
@@ -374,6 +384,14 @@ ATCA_STATUS atca_execute_command(ATCAPacket* packet, ATCADevice device)

         if ((status = isATCAError(packet->data)) != ATCA_SUCCESS)
         {
+        #if defined(HAL_HACK_RETRY) && (HAL_HACK_RETRY)
+            if (status == ATCA_WAKE_SUCCESS && retrycnt <= 3) {
+                printf("  %s %d status %d retry %d\n", __func__, __LINE__, status, retrycnt);
+                goto retry;
+            } else {
+                printf("  %s %d status %d\n", __func__, __LINE__, status);
+            }
+        #endif
             break;
         }
     }
```

Questions: 

- Is this a proper fix or a workaround? 
- What's the reason for receiving this status? 

The datasheets seem to tell little about why this is happening and how to deal with it.

minghuadev, 2020-10-05:
The fix does not work. It failed ecdh.py command due to the same error despite the retry.

bryan-hunt, 2020-10-05:
This appears to be a HAL related problem if not a usage problem. Either way you'll have to work with Microchip Support FAEs to help you with your specific implementation. You should submit a product support request through the [Microchip Support portal](https://microchipsupport.force.com/s/). For details on the Support Portal or how to use it please see [How to Submit a Case](https://microchipsupport.force.com/s/article/How-to-submit-a-case)

minghuadev, 2020-10-22:
I captured the I2C transactions when it is working and not working. A few interesting findings: 

[1] If you send the next command too soon, the device may respond with a NACK. The datasheet does not say how to deal with this. 
[2] Afer a "wake" command, the status is "wake-success". When it fails as above, when you send the next command, it gets "wake-success". That looks like the chip is somehow confused, not responding to the new command, but to the previous "wake" command. 

We found a Linux platform that works ok on /dev/i2c-n node. Am also wondering whether the new ATECC608B I2C fixes will work with this issue.  

Will work with Microchip support. Thanks for pointing to the portal and the case instructions.

closed 2020-10-22 by minghuadev