MicrochipTech/cryptoauthlib #189: Cannot build python package

fstephany, 2020-10-22:
I'm trying to build the python bindings (Fedora 33, Python 3.9, cmake 3.18.3). I followed the steps described in the `.travis.yml` to prepare the build:

~~~
$ cd python/
$ git describe --exclude v* --tags --abbrev=0 >VERSION
$ cp -r ../lib .
$ cp -r ../third_party .
$ cp ../test/atca_utils_sizes.c .
$ mkdir app
$ cp -r ../app/tng app
$ python setup.py install
~~~

The build failed because of references to `TA_SELF_TEST_FAILURE` in the linux_spi HAL:

~~~
/home/user/Code/cryptoauthlib/python/lib/hal/hal_linux_spi_userspace.c:280:30: error: ‘TA_SELF_TEST_FAILURE’ undeclared (first use in this function)
  280 |                     status = TA_SELF_TEST_FAILURE;
~~~

As I'm not interested in the SPI interface, I've replaced the `TA_SELF_TEST_FAILURE` by `ATCA_ASSERT_FAILURE` just to get it built. 

I then run into another error at link time:

~~~
RuntimeError: Scanning dependencies of target cryptoauth
[  1%] Building C object CMakeFiles/cryptoauth.dir/hal/hal_linux_spi_userspace.c.o
[  2%] Building C object CMakeFiles/cryptoauth.dir/home/fmos/Code/cryptoauthlib/python/atca_utils_sizes.c.o
[  4%] Linking C shared library /home/user/Code/cryptoauthlib/python/build/lib.linux-x86_64-3.9/cryptoauthlib/libcryptoauth.so
/usr/bin/ld: CMakeFiles/cryptoauth.dir/atca_cfgs.c.o:(.bss+0x0): multiple definition of `atcab_b64rules_default'; CMakeFiles/cryptoauth.dir/atca_basic.c.o:(.bss+0x0): first defined here
/usr/bin/ld: CMakeFiles/cryptoauth.dir/atca_cfgs.c.o:(.bss+0x4): multiple definition of `atcab_b64rules_mime'; CMakeFiles/cryptoauth.dir/atca_basic.c.o:(.bss+0x4): first defined here
/usr/bin/ld: CMakeFiles/cryptoauth.dir/atca_cfgs.c.o:(.bss+0x8): multiple definition of `atcab_b64rules_urlsafe'; CMakeFiles/cryptoauth.dir/atca_basic.c.o:(.bss+0x8): first defined here
/usr/bin/ld: CMakeFiles/cryptoauth.dir/atca_cfgs.c.o:(.bss+0x10): multiple definition of `_gDevice'; CMakeFiles/cryptoauth.dir/atca_basic.c.o:(.bss+0x10): first defined here
/usr/bin/ld: CMakeFiles/cryptoauth.dir/atca_command.c.o:(.bss+0x0): multiple definition of `atcab_b64rules_default'; CMakeFiles/cryptoauth.dir/atca_basic.c.o:(.bss+0x0): first defined here
/usr/bin/ld: CMakeFiles/cryptoauth.dir/atca_command.c.o:(.bss+0x4): multiple definition of `atcab_b64rules_mime'; CMakeFiles/cryptoauth.dir/atca_basic.c.o:(.bss+0x4): first defined here
/usr/bin/ld: CMakeFiles/cryptoauth.dir/atca_command.c.o:(.bss+0x8): multiple definition of `atcab_b64rules_urlsafe'; CMakeFiles/cryptoauth.dir/atca_basic.c.o:(.bss+0x8): first defined here
/usr/bin/ld: CMakeFiles/cryptoauth.dir/atca_command.c.o:(.bss+0x10): multiple definition of `_gDevice'; CMakeFiles/cryptoauth.dir/atca_basic.c.o:(.bss+0x10): first defined here

...repeated a number of times for different object files with always the same four definitions
~~~

The arguments used to configure cmake  in the `def build_extension` method:

~~~
['-DATCA_HAL_CUSTOM=ON', '-DATCA_TNGTLS_SUPPORT=ON', '-DATCA_TNGLORA_SUPPORT=ON', '-DATCA_TFLEX_SUPPORT=ON', '-DATCA_TNG_LEGACY_SUPPORT=ON', '-DATCA_USE_ATCAB_FUNCTIONS=ON', '-DATCA_HAL_KIT_HID=ON', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/home/user/Code/cryptoauthlib/python/build/lib.linux-x86_64-3.9/cryptoauthlib', '-DATCA_HAL_I2C=ON', '-DATCA_HAL_SPI=ON', '-DATCACERT_DEF_SRC=/home/user/Code/cryptoauthlib/python/atca_utils_sizes.c']
~~~

I'm far from a cmake user. Any idea what I might be doing wrong?

bryan-hunt, 2020-10-22:
This is a gcc10 related issue with the option -fno-common per #185. There is a release tomorrow that will address it.

fstephany, 2020-10-22:
I didn't think it was related, good to know, thanks!

bryan-hunt, 2020-10-22:
@fstephany The linking errors should be resolved in v3.2.4 which was just published

fstephany, 2020-10-26:
They are, the module now compiles fine.
Thanks!

closed 2020-10-26 by fstephany