MicrochipTech/cryptoauthlib #191: atcah_check_mac returing success even with false  response data

lijintv07, 2020-10-22:
I use SAMD21 target platform with ECC chip. I was trying to write some testcases using  ATCAH_* functions on linux platform to generate  a nonce , mac and checkmac.  

Brief Explanation of what i was trying :  Generate a Nonce , pass it as a challenge for Mac generation and check the generated mac using checkmac

Expected Result : 
A 32 byte random nonce at **rand_out** :  Generated Mac at **host_mac** :  Mac verification

Problems Faced :  
1. **atcah_nonce** did return Success status but Nonce was not generated 
2.  **atcah_check_mac** is returning success even if i change the  first byte of **host_mac** while assigning to **checkmac_params.client_resp**.

```
#include <stdio.h>
#include <stdlib.h>
#include "atca_host.h"
#include "atca_status.h"

int main()
{

	int ret = 0;
	ATCA_STATUS status;
	uint8_t sn[ATCA_SERIAL_NUM_SIZE]= {0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff,0xff};
	uint8_t num_in[NONCE_NUMIN_SIZE]= {0x01,0x02,0x03,0x04,0x05,0x06,0x07,0x08,0x09,0x0a,
                                                                       0x0b,0x0c,0x0d,0x0e,0x0f,0x10,0x11,0x12,0x13,0x14};
	uint8_t rand_out[RANDOM_NUM_SIZE];
	uint8_t host_mac[MAC_SIZE];
	uint8_t other_data[CHECKMAC_OTHER_DATA_SIZE];
	const uint8_t g_slot4_key[] = { 0x37, 0x80, 0xe6, 0x3d, 0x49, 0x68, 0xad, 0xe5,
                                                       0xd8, 0x22, 0xc0, 0x13, 0xfc, 0xc3, 0x23, 0x84,
                                                       0x5d, 0x1b, 0x56, 0x9f, 0xe7, 0x05, 0xb6, 0x00,
                                                       0x06, 0xfe, 0xec, 0x14, 0x5a, 0x0d, 0xb1, 0xe3};
	
	
	atca_temp_key_t temp_key;
	atca_nonce_in_out_t nonce_params;
	atca_mac_in_out_t mac_params;
	atca_check_mac_in_out_t checkmac_params;

	// Setup nonce command
	memset(&temp_key, 0, sizeof(temp_key));
	memset(&nonce_params, 0, sizeof(nonce_params));
	nonce_params.mode = NONCE_MODE_SEED_UPDATE;
	nonce_params.zero = 0;
	nonce_params.num_in = num_in;
	nonce_params.rand_out = rand_out;
	nonce_params.temp_key = &temp_key;
	
	
	// Calculate nonce in host
	if ((status = atcah_nonce(&nonce_params)) != ATCA_SUCCESS)
	{
		printf("atcah_nonce_ERROR : %d", status);
		return status;
	}
	
	printf("\nNONCE- RandOut :\n");
	for (int i = 0; i< RANDOM_NUM_SIZE;printf("%02x ",nonce_params.rand_out[i++]));

	// Setup MAC command
	memset(&mac_params, 0, sizeof(mac_params));
	mac_params.mode = MAC_MODE_CHALLENGE;
	mac_params.key_id = 5;
	mac_params.challenge =rand_out;
	mac_params.key = g_slot4_key;
	mac_params.otp = NULL;
	mac_params.sn = sn;
	mac_params.response = host_mac;
	mac_params.temp_key = &temp_key;

	// Calculate the MAC on the host side
	if ((status = atcah_mac(&mac_params)) != ATCA_SUCCESS)
	{
		printf("\natcah_mac_ERROR : %d", status);
		return status;
	}

	printf("\nMAC :\n");
	for (int i = 0; i< MAC_SIZE;printf(" %02x",host_mac[i++]));
		
	//Setup Check MAC command
	memset(&checkmac_params, 0, sizeof(checkmac_params));
	checkmac_params.mode = CHECKMAC_MODE_CHALLENGE ;
	checkmac_params.key_id = 5;
	checkmac_params.client_chal =rand_out;
	checkmac_params.client_resp =host_mac;
	checkmac_params.other_data =other_data;
	checkmac_params.sn = sn;
	checkmac_params.otp = NULL;
	checkmac_params.slot_key = g_slot4_key;
	checkmac_params.target_key = NULL;
	checkmac_params.temp_key = &temp_key;

	if ((status = atcah_check_mac(&checkmac_params)) != ATCA_SUCCESS)
	{
		printf("\natcah_check_mac_ERROR : %d", status);
		return status;
	}

	return ret;

}
```

bryan-hunt, 2020-10-22:
Product usage and support questions are answered through the Microchip Support portal where FAEs can work on your use case.

You should submit a product support request through the [Microchip Support portal](https://microchipsupport.force.com/s/). For details on the Support Portal or how to use it please see [How to Submit a Case](https://microchipsupport.force.com/s/article/How-to-submit-a-case)

closed 2020-10-22 by bryan-hunt

lijintv07, 2020-10-22:
Okay , but i'm having issue using the crypauthlib library

bryan-hunt, 2020-10-22:
@lijintv07 [atcah_check_mac](https://microchiptech.github.io/cryptoauthlib/a08587.html#ga5b457cd0ccba73864eee4030b6079ba7) does not check a mac per the documentation (linked). It generates the expected mac. The application is expected to compare the result.