MicrochipTech/cryptoauthlib #192: Make  failed while enabling mbedtls ATCA_MBEDTLS:BOOL=ON

lijintv07, 2020-10-23:
/usr/bin/ld: libmbedtls.a(x509_crt.c.o): relocation R_X86_64_PC32 against symbol `mbedtls_x509_crt_parse' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: Bad value
collect2: error: ld returned 1 exit status
lib/CMakeFiles/cryptoauth.dir/build.make:1811: recipe for target 'lib/libcryptoauth.so' failed
make[2]: *** [lib/libcryptoauth.so] Error 1
CMakeFiles/Makefile2:87: recipe for target 'lib/CMakeFiles/cryptoauth.dir/all' failed
make[1]: *** [lib/CMakeFiles/cryptoauth.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2

bryan-hunt, 2020-10-23:
Can you also provide the list of other CMake option values?

lijintv07, 2020-10-24:
@bryan-hunt  The other CMake options are mentioned below

// Include support for ATECC108A device
ATCA_ATECC108A_SUPPORT:BOOL=OFF

// Include support for ATECC508A device
ATCA_ATECC508A_SUPPORT:BOOL=OFF

// Include support for ATECC608 device
ATCA_ATECC608_SUPPORT:BOOL=ON

// Include support for ATSHA204A device
ATCA_ATSHA204A_SUPPORT:BOOL=OFF

// Include support for ATSHA206A device
ATCA_ATSHA206A_SUPPORT:BOOL=OFF

// Build CryptoAuthLib as shared library
ATCA_BUILD_SHARED_LIBS:BOOL=ON

// Include support for Custom/Plug-in Hal Driver
ATCA_HAL_CUSTOM:BOOL=OFF

// Include the I2C Hal Driver - Linux & MCU only
ATCA_HAL_I2C:BOOL=OFF

// Include the CDC HAL Driver (Legacy)
ATCA_HAL_KIT_CDC:BOOL=OFF

// Include the HID HAL Driver
ATCA_HAL_KIT_HID:BOOL=OFF

// Include the SPI HAL Driver - Linux & MCU only
ATCA_HAL_SPI:BOOL=OFF

// Integrate with mbedtls
ATCA_MBEDTLS:BOOL=ON

// Do not use dynamic (heap) allocation functions
ATCA_NO_HEAP:BOOL=OFF

// Integration with OpenSSL
ATCA_OPENSSL:BOOL=OFF

// Build PKCS11 Library
ATCA_PKCS11:BOOL=OFF

// 
ATCA_PLATFORM_FREE:STRING=free

// 
ATCA_PLATFORM_MALLOC:STRING=malloc

// Enable Debug print statements in library
ATCA_PRINTF:BOOL=OFF

// Include support for TA100 device
ATCA_TA100_SUPPORT:BOOL=OFF

// Include Trust Flex Certificates
ATCA_TFLEX_SUPPORT:BOOL=OFF

// Include Trust & Go LORA Certificates
ATCA_TNGLORA_SUPPORT:BOOL=OFF

// Include Trust & Go TLS Certificates
ATCA_TNGTLS_SUPPORT:BOOL=OFF

// Include previous version of Trust & Go Certificates
ATCA_TNG_LEGACY_SUPPORT:BOOL=OFF

// Build the atcab_ api functions rather than using macros
ATCA_USE_ATCAB_FUNCTIONS:BOOL=OFF

// Integrate with WolfSSL
ATCA_WOLFSSL:BOOL=OFF

// Create Test Application with library
BUILD_TESTS:BOOL=OFF

// Default build type
CMAKE_BUILD_TYPE:STRING=RelWithDebInfo

// The default file for library configuration
DEFAULT_CONF_FILE_NAME:STRING=cryptoauthlib.conf

// The default location of cryptoauthlib.conf
DEFAULT_CONF_PATH:STRING=/etc/cryptoauthlib

// The default absolute library path
DEFAULT_LIB_PATH:STRING=/usr/lib

// The default location of the filestore directory
DEFAULT_STORE_PATH:STRING=/var/lib/cryptoauthlib

> 
> 
> Can you also provide the list of other CMake option values?

lijintv07, 2020-10-26:
@bryan-hunt  The list of other CMake options are mentioned below

// Include support for ATECC108A device
ATCA_ATECC108A_SUPPORT:BOOL=OFF

// Include support for ATECC508A device
ATCA_ATECC508A_SUPPORT:BOOL=OFF

// Include support for ATECC608 device
ATCA_ATECC608_SUPPORT:BOOL=ON

// Include support for ATSHA204A device
ATCA_ATSHA204A_SUPPORT:BOOL=OFF

// Include support for ATSHA206A device
ATCA_ATSHA206A_SUPPORT:BOOL=OFF

// Build CryptoAuthLib as shared library
ATCA_BUILD_SHARED_LIBS:BOOL=ON

// Include support for Custom/Plug-in Hal Driver
ATCA_HAL_CUSTOM:BOOL=OFF

// Include the I2C Hal Driver - Linux & MCU only
ATCA_HAL_I2C:BOOL=OFF

// Include the CDC HAL Driver (Legacy)
ATCA_HAL_KIT_CDC:BOOL=OFF

// Include the HID HAL Driver
ATCA_HAL_KIT_HID:BOOL=OFF

// Include the SPI HAL Driver - Linux & MCU only
ATCA_HAL_SPI:BOOL=OFF

// Integrate with mbedtls
ATCA_MBEDTLS:BOOL=ON

// Do not use dynamic (heap) allocation functions
ATCA_NO_HEAP:BOOL=OFF

// Integration with OpenSSL
ATCA_OPENSSL:BOOL=OFF

// Build PKCS11 Library
ATCA_PKCS11:BOOL=OFF

//
ATCA_PLATFORM_FREE:STRING=free

//
ATCA_PLATFORM_MALLOC:STRING=malloc

// Enable Debug print statements in library
ATCA_PRINTF:BOOL=OFF

// Include support for TA100 device
ATCA_TA100_SUPPORT:BOOL=OFF

// Include Trust Flex Certificates
ATCA_TFLEX_SUPPORT:BOOL=OFF

// Include Trust & Go LORA Certificates
ATCA_TNGLORA_SUPPORT:BOOL=OFF

// Include Trust & Go TLS Certificates
ATCA_TNGTLS_SUPPORT:BOOL=OFF

// Include previous version of Trust & Go Certificates
ATCA_TNG_LEGACY_SUPPORT:BOOL=OFF

// Build the atcab_ api functions rather than using macros
ATCA_USE_ATCAB_FUNCTIONS:BOOL=OFF

// Integrate with WolfSSL
ATCA_WOLFSSL:BOOL=OFF

// Create Test Application with library
BUILD_TESTS:BOOL=OFF

// Default build type
CMAKE_BUILD_TYPE:STRING=RelWithDebInfo

// The default file for library configuration
DEFAULT_CONF_FILE_NAME:STRING=cryptoauthlib.conf

// The default location of cryptoauthlib.conf
DEFAULT_CONF_PATH:STRING=/etc/cryptoauthlib

// The default absolute library path
DEFAULT_LIB_PATH:STRING=/usr/lib

// The default location of the filestore directory
DEFAULT_STORE_PATH:STRING=/var/lib/cryptoauthlib

reneromanus, 2020-11-09:
Briefly, the error means that you can't use a static library to be linked w/ cryptoauthlib as a dynamic one.
If you choose to remain mbedtls as static library, you should compile cryptoauthlib static as well:

`ATCA_BUILD_SHARED_LIBS:BOOL=OFF` may resolve your configuration.

bryan-hunt, 2020-12-02:
@lijintv07 @reneromanus  This should be resolved in v3.2.5

closed 2020-12-07 by bryan-hunt