MicrochipTech/cryptoauthlib #193: consider supporting use of more inclusive language throughout codebase

nilslice, 2020-10-23:
Hi, first, thanks for all the work put into this library! 

I'm going through some work to purge language from our codebases that may make folks uncomfortable or shy away from contributing to projects, and came across a field member of the `ATCAIfaceCfg` struct, `.atcai2c.slave_address`. I was wondering if there was a chance you'd consider using an alternative term. It's possible that this and similar terms exist elsewhere, but this was the first occurrence that caught my eye. I would be happy to search through the code and find others if it is indeed something you would consider updating. 

I fully understand the issues and difficulty imposed with regard to renaming fields and preserving API stability, so I wouldn't expect that this is considered until a new release could be cut that would indicate a breaking change. However, the feedback from the software community at large has been in favor of supporting these changes - so I don't think most teams would mind :)

bryan-hunt, 2020-10-23:
You will notice that default branch for this repo has been converted to main already (however some systems still require the master branch to exist so it is being kept in sync) and wherever else possible but yes API compatibility does limit the ability make modifications so it is very unlikely that this library will be able to change those ever.

bryan-hunt, 2020-10-23:
To clarify it is easy to add an alternative definition that can be guarded with #defines so customers may use the API of their choosing - it is significantly more difficult to remove that language completely from the library without impacting many that are unable to change.

nilslice, 2020-10-23:
@bryan-hunt -- thanks for the quick reply. Makes sense, and thank you for addressing the change to `main` over `master` as the default 👍  

> To clarify it is easy to add an alternative definition that can be guarded with #defines so customers may use the API of their choosing

That sounds like a great alternative -- though, excuse my naivety here -- is this something our lib (which uses `cryptoauthlib`) would define? or something set here in this source code that we as lib users could opt-in to?

bryan-hunt, 2020-10-23:
> That sounds like a great alternative -- though, excuse my naivety here -- is this something our lib (which uses `cryptoauthlib`) would define? or something set here in this source code that we as lib users could opt-in to?

I can't absolutely say how it will go - most likely it will be an opt in to use deprecated features (e.g. ATCA_ENABLE_DEPRECATED)

nilslice, 2020-10-23:
Ok, thank you. Feel free to ping me if you end up going through with this. Happy to help however I can or see if someone on my team can.

bryan-hunt, 2021-11-04:
I think generally this was addressed - I haven't seen any additional comments anyway so I'm going to close it as complete.

closed 2021-11-04 by bryan-hunt