MicrochipTech/cryptoauthlib #194: Failing to compile for Python

rz259, 2020-10-27:
I cloned the repository and tried to compile everything for the python examples. I did this by executing the command
pip3 install -r requirements.txt.

As as result I got the following message:

```
 Traceback (most recent call last):
    File "/tmp/pip-install-x5ce_76p/cryptoauthlib/setup.py", line 180, in build_extension
      subprocess.check_output(['cmake', '--build', '.'] + build_args, cwd=os.path.abspath(self.build_temp), stderr=subprocess.STDOUT)
    File "/usr/lib/python3.7/subprocess.py", line 395, in check_output
      **kwargs).stdout
    File "/usr/lib/python3.7/subprocess.py", line 487, in run
      output=stdout, stderr=stderr)
  subprocess.CalledProcessError: Command '['cmake', '--build', '.']' returned non-zero exit status 2.

  During handling of the above exception, another exception occurred:

  Traceback (most recent call last):
    File "<string>", line 1, in <module>
    File "/tmp/pip-install-x5ce_76p/cryptoauthlib/setup.py", line 229, in <module>
      zip_safe=False
    File "/usr/lib/python3/dist-packages/setuptools/__init__.py", line 145, in setup
      return distutils.core.setup(**attrs)
    File "/usr/lib/python3.7/distutils/core.py", line 148, in setup
      dist.run_commands()
    File "/usr/lib/python3.7/distutils/dist.py", line 966, in run_commands
      self.run_command(cmd)
    File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
      cmd_obj.run()
    File "/usr/lib/python3/dist-packages/wheel/bdist_wheel.py", line 188, in run
      self.run_command('build')
    File "/usr/lib/python3.7/distutils/cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
      cmd_obj.run()
    File "/usr/lib/python3.7/distutils/command/build.py", line 135, in run
      self.run_command(cmd_name)
    File "/usr/lib/python3.7/distutils/cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "/usr/lib/python3.7/distutils/dist.py", line 985, in run_command
      cmd_obj.run()
    File "/usr/lib/python3/dist-packages/setuptools/command/build_ext.py", line 78, in run
      _build_ext.run(self)
    File "/usr/lib/python3.7/distutils/command/build_ext.py", line 340, in run
      self.build_extensions()
    File "/usr/lib/python3.7/distutils/command/build_ext.py", line 449, in build_extensions
      self._build_extensions_serial()
    File "/usr/lib/python3.7/distutils/command/build_ext.py", line 474, in _build_extensions_serial
      self.build_extension(ext)
    File "/tmp/pip-install-x5ce_76p/cryptoauthlib/setup.py", line 185, in build_extension
      raise RuntimeError(e.output.decode('utf-8'))
  RuntimeError: Scanning dependencies of target cryptoauth
  [  1%] Building C object CMakeFiles/cryptoauth.dir/atca_basic.c.o
  In file included from /tmp/pip-install-x5ce_76p/cryptoauthlib/lib/atca_basic.h:34,
                   from /tmp/pip-install-x5ce_76p/cryptoauthlib/lib/atca_basic.c:29:
  /tmp/pip-install-x5ce_76p/cryptoauthlib/lib/cryptoauthlib.h:116:10: fatal error: talib/talib_status.h: Datei oder Verzeichnis nicht gefunden
   #include "talib/talib_status.h"
            ^~~~~~~~~~~~~~~~~~~~~~
  compilation terminated.
  make[2]: *** [CMakeFiles/cryptoauth.dir/build.make:63: CMakeFiles/cryptoauth.dir/atca_basic.c.o] Fehler 1
  make[1]: *** [CMakeFiles/Makefile2:73: CMakeFiles/cryptoauth.dir/all] Fehler 2
  make: *** [Makefile:84: all] Fehler 2
```

So the problem is the directory talib and the file talib_status.h is missing. Could you please add the missing files and verify that everything compiles fine?

Thanks a lot,

Rudi

bryan-hunt, 2020-10-27:
Unless you're using yocto the proper method of installing python packages is to use pip `pip install cryptoauthlib`

Regardless this error is an indication that your directory is not clean - do a fresh clone or clean and checkout v3.2.4 or 20200912

rz259, 2020-10-27:
Thanks for your help. Unfortunately I still get the same error. I did a new clone for cryptoauthlib and switched to branch 20200912. When I do a pip install cryptoauthlib (I'am using pip3 for python 3) I still get the same error (the file "talib/talib_status.h" can't be found)

bryan-hunt, 2020-10-27:
I assume your build cache is also incorrect. You have to clean out the whole system - all the artifacts from your previous builds as well. For python you can force it to start from scratch by `pip install --no-cache cryptoauthlib`

rz259, 2020-10-27:
I don't get it right, sorry for being stupid;-(

I deleted the git cache ("git rm -r --cached . "), I also deleted the directory for cryptoauthlib and did a new clone for cryptoauthlib.
I then switched to 20200912 (also tried  v3.2.4) and tried "pip install --no-cache cryptoauthlib" - but still the same result.

getlarge, 2020-11-06:
@rz259 I went through the same problem and managed to build `cryptoauthlib` with `pip install --no-cache cryptoauthlib==20200208`. Hope it will help you.

bryan-hunt, 2020-12-02:
@rz259 @getlarge This issue should be resolved in v3.2.5 (20201130) which is available on pypi now

closed 2020-12-03 by bryan-hunt

getlarge, 2020-12-03:
@bryan-hunt Thanks for letting us know.