MicrochipTech/cryptoauthlib #195: openssl-1.1.0: EVP_CIPHER_CTX_cleanup removed, EVP_CIPHER_CTX_reset may be used instead

reneromanus, 2020-10-30:
In atca_openssl_interface function atcac_aes_gcm_decrypt_finish, you use deprecated openssl function "EVP_CIPHER_CTX_cleanup" (removed in version 1.1.0). Instead, you might use still available function "EVP_CIPHER_CTX_reset".

Relevant CMake option values (as excerpt) are:
ATCA_OPENSSL=ON
ATCA_PKCS11=ON
ATCA_BUILD_SHARED_LIBS=OFF

bryan-hunt, 2020-10-30:
It looks like _init and _cleanup are both aliased to _reset when OPENSSL_NO_DEPRECATED_1_1_0 is not defined which seems to be a strict option (I hadn't encountered it being set in the systems the library is tested with). It's an easy enough fix to include - in the meantime not setting that option in your openssl configuration for your system should fix the cryptoauthlib build.

bryan-hunt, 2020-12-03:
This should be resolved in v3.2.5

closed 2020-12-07 by bryan-hunt