MicrochipTech/cryptoauthlib #196: A brief tutorial / walkthrough for those starting with cryptoauthlib

krish2487, 2020-11-01:
Hello,

I apologize in advance if this question seems stupid. We are trying to integrate the ATECC608A into one of the projects we are working on. However, the cryptoauthlib is rather daunting and intimidating for someone just starting out with this lib / chip combo. For information, we are using a custom board with an STM32F1 processor and the libopencm3 library.

There is a plethora of documentation available, but it is rather scattered across multiple app notes and not nearly intuitive enough to start using it without more expert help. 

I was wondering if anyone can take the time to give a step by step crash course. Nothing too complex, say if I wanted to read the UUID of the cryptochip using the ATCAB_ functions. I am looking for some help in doing so with the STM32 processor. I am looking for something like so :- 

1. Include so and so files.
2. Declare so and so structures.
3. Define so and so structure members to point to so and so libopencm3 or HAL  read/write/delay functions.
4. Now call so and so init functions.
5. Now call so and so functions to read the UUID.

Any help / suggestions are appreciated and welcome. I do feel this might help others to get started with the cryptoauthlib as well, and I apologize again if this looks like a stupid question.

Thanks in advance! :-)

bryan-hunt, 2020-11-01:
The best place to get started is the [Trust Platform Design Suite](https://www.microchip.com/design-centers/security-ics/trust-platform) which will walk you through use cases and the corresponding firmware development.

The fact that libopencm3 is licensed under GPL/LGPL means we simply do not see any commercial customers using it so we don't have any plans to create a HAL using that framework.

krish2487, 2020-11-01:
@bryan-hunt  
Thank you for taking the time to respond back. I understand about the libopencm3 part and I am not looking for microchip to work on it. I was merely asking if there was a easier way to communicate with the chip and get started. I will explore the link you have suggested and post back if I do have other queries. 

Thank you again. :-)

krish2487, 2020-11-01:
@bryan-hunt 
I think the macos installed for the trust platform is broken. I have tried installing the trust platform on Mac OS Catalina 10.15.7
And it fails with an error. I think it is because the installer tries to work with the wrong binaries for anaconda. 
If I look under the trust_platform folder in my user directory, I see conda.exe excutable, not the macos version.

Can you please confirm if you see the same issue?

bryan-hunt, 2020-11-01:
That question would be best answered through the Microchip Support portal where FAEs can work on your use case.
 
You should submit a product support request through the [Microchip Support portal](https://microchipsupport.force.com/s/). For details on the Support Portal or how to use it please see [How to Submit a Case](https://microchipsupport.force.com/s/article/How-to-submit-a-case)

closed 2020-12-02 by bryan-hunt