MicrochipTech/cryptoauthlib #199: Python package always attempts to build shared library

cquartier, 2020-11-11:
This section (https://github.com/MicrochipTech/cryptoauthlib/blob/main/python/setup.py#L57-L65) seems to be the culprit:

```python
# See if the library is already installed
try:
    lib = cdll.LoadLibrary('libcryptoauth.so')
    # Test to ensure it has the required features to support the
    # python wrapper. It may change later to a version check
    assert 0 != lib.ATCAIfacecfg_size
    _EXTENSIONS = None
except:
    _EXTENSIONS = [Extension('cryptoauthlib', sources=[])]
```

in particular this line:

`assert 0 != lib.ATCAIfacecfg_size`

I have a system where I've compiled `libcryptoauth.so` with custom modifications and want to `pip install cryptoauthlib` to get the python module but this always fails because this check doesn't pass and thinks the library is not already on the system. I'm able to `cdll.LoadLibrary('libcryptoauth.so')` from a python repl on my system by hand, and run commands that way, so I know python is able to load it on my system.

I've been testing this with `python3.7`, am I missing something in `libcryptoauth.so`? Has this check ever worked?

bryan-hunt, 2020-11-11:
It's because the symbol ATCAIfacecfg_size doesn't exist in the library you build. Without information on structure sizes the python module can't function (it includes numerous checks for proper structure creation). Include [test/atca_utils_sizes.c](https://github.com/MicrochipTech/cryptoauthlib/blob/main/test/atca_utils_sizes.c) in your library build.

cquartier, 2020-11-13:
Is this the line that ensures this file gets included in the cmake build in `setup.py`?

```python
cmake_args += ['-DATCACERT_DEF_SRC={}atca_utils_sizes.c'.format(setupdir.replace('\\','/') if _sdist_build else '../test/')]
```

if so, it seems I just need to add `-DATCACERT_DEF_SRC=../test/atca_utils_sizes.c` to my build, yes?

Last question: why not always include this file in the build? Or at least provide a more obvious cmake variable to enable including this file?

cquartier, 2020-11-13:
I built the library with `-DATCACERT_DEF_SRC=../test/atca_utils_sizes.c`. The `setup.py` file still has a bug by the way:

`ATCAIfacecfg_size` is the wrong name. When I run strings on `libcryptoauth.so` I see `ATCAIfaceCfg_size`

cquartier, 2020-11-20:
@bryan-hunt should I make a new issue about the bug in my above comment?

bryan-hunt, 2020-11-20:
No it's already been noted internally and will be addressed in the next release which is forthcoming.

cquartier, 2020-11-23:
Glad to hear that, thanks!

bryan-hunt, 2020-12-03:
@cquartier This should be resolved in version 20201203 if you were to download it via pypi

closed 2020-12-07 by bryan-hunt