MicrochipTech/cryptoauthlib #200: 20201017 release on PyPi

fstephany, 2020-11-13:
It seems that the [latest version on PyPi](https://pypi.org/project/cryptoauthlib/#history) is the 20200912. 
Is there any issue to push the latest release on PyPi? Or is there another mechanism to retrieve the package?

bryan-hunt, 2020-11-13:
There is an issue in the appveyor builds I’m trying to resolve so it is delayed until that is working again (otherwise the pypi release ends up with only some of wheels). From a CI perspective with the major shift in the way Travis-CI is as well I believe I’m going to be transitioning everything over to GitHub actions next week.

bryan-hunt, 2020-12-02:
@fstephany This should now be fixed with the release of v3.2.5 (20201130)

closed 2020-12-02 by bryan-hunt