MicrochipTech/cryptoauthlib #201: Static analysis

Harvie, 2020-12-05:
Hello,
have you considered cleaning up the code to show no warnings in static analysis tools? It can make the code more robust. Especially cppcheck and scan-build:

http://cppcheck.sourceforge.net/
https://clang-analyzer.llvm.org/scan-build.html

Eg. cppcheck detects some errors (some of them might be false-positive):

```
$ time cppcheck . >/dev/null
lib/crypto/atca_crypto_sw.h:49:0: error: No header in #include [preprocessorErrorDirective]
#include MBEDTLS_CONFIG_FILE
^
lib/mbedtls/atca_mbedtls_ecdh.c:33:0: error: No header in #include [preprocessorErrorDirective]
#include MBEDTLS_CONFIG_FILE
^
lib/mbedtls/atca_mbedtls_ecdsa.c:33:0: error: No header in #include [preprocessorErrorDirective]
#include MBEDTLS_CONFIG_FILE
^
lib/mbedtls/atca_mbedtls_wrap.c:34:0: error: No header in #include [preprocessorErrorDirective]
#include MBEDTLS_CONFIG_FILE
^
lib/pkcs11/pkcs11_token.c:164:17: error: Uninitialized variable: buf [uninitvar]
        switch (buf[2])
                ^
third_party/hidapi/mac/hid.c:802:9: warning: Possible null pointer dereference: data [nullPointer]
 memcpy(data, rpt->data, len);
        ^
third_party/hidapi/mac/hid.c:592:21: note: Calling function 'return_data', 2nd argument 'NULL' value is 0
   return_data(dev, NULL, 0);
                    ^
third_party/hidapi/mac/hid.c:802:9: note: Null pointer dereference
 memcpy(data, rpt->data, len);
        ^
third_party/hidapi/windows/hid.c:187:8: error: Uninitialized variable: msg [uninitvar]
 ptr = msg;
       ^
third_party/hidapi/windows/hid.c:188:10: error: Uninitialized variable: ptr [uninitvar]
 while (*ptr) {
         ^
third_party/hidapi/windows/hid.c:187:8: note: Assignment 'ptr=msg', assigned value is <Uninit>
 ptr = msg;
       ^
third_party/hidapi/windows/hid.c:188:10: note: Uninitialized variable: ptr
 while (*ptr) {
         ^
third_party/hidapi/windows/hid.c:189:8: error: Uninitialized variable: ptr [uninitvar]
  if (*ptr == '\r') {
       ^
third_party/hidapi/windows/hid.c:187:8: note: Assignment 'ptr=msg', assigned value is <Uninit>
 ptr = msg;
       ^
third_party/hidapi/windows/hid.c:189:8: note: Uninitialized variable: ptr
  if (*ptr == '\r') {
       ^
third_party/hidapi/windows/hid.c:190:5: error: Uninitialized variable: ptr [uninitvar]
   *ptr = 0x0000;
    ^
third_party/hidapi/windows/hid.c:187:8: note: Assignment 'ptr=msg', assigned value is <Uninit>
 ptr = msg;
       ^
third_party/hidapi/windows/hid.c:190:5: note: Uninitialized variable: ptr
   *ptr = 0x0000;
    ^
third_party/hidapi/windows/hid.c:199:27: error: Uninitialized variable: msg [uninitvar]
 device->last_error_str = msg;
                          ^
third_party/unity/unity_memory.c:104:5: error: Memory leak: guard [memleak]
    return (void*)mem;
    ^
third_party/hidapi/mac/hid.c:802:9: error: Null pointer dereference: data [ctunullpointer]
 memcpy(data, rpt->data, len);
        ^
third_party/hidapi/mac/hid.c:592:15: note: Calling function return_data, 2nd argument is null
   return_data(dev, NULL, 0);
              ^
third_party/hidapi/mac/hid.c:802:9: note: Dereferencing argument data that is null
 memcpy(data, rpt->data, len);
        ^
nofile:0:0: information: Too many #ifdef configurations - cppcheck only checks 12 configurations. Use --force to check all configurations. For more details, use --enable=information. [toomanyconfigs]

real	0m51,802s
user	0m49,973s
sys	0m0,155s
```

bryan-hunt, 2020-12-05:
Yes the library runs through PCLint for MISRA 2012 compliance and Coverity for Cert-C. I only see the one warning for the uninitialized buffer.  It was missed by our checkers because we don't have an active configuration where PKCS11_TOKEN_INIT_SUPPORT is enabled.

* Hidapi is for use with kits and thus should never be used in a production environment - it is also a third party library. Third party libraries are obviously not checked.
* PKCS11_TOKEN_INIT_SUPPORT by definition is unsafe and should never be enabled in a production system. 
* The MBEDTLS_CONFIG_FILE warning is due to an environment misconfiguration (and obviously would not compile) but is a weird construct necessitated by the way mbedtls handles their configuration (in order to support mbedtls properly it is required)

bryan-hunt, 2021-01-27:
This should be resolved in 20210121 and v3.3.0

closed 2021-01-27 by bryan-hunt